genodelabs/genode #5096: dde_linux: remove static constructors for `module_init` calls and friends

ssumpf, 2024-01-19:
A call to `module_init` currently registers with `__attribute__((constructor))` leading to issues with libc components (e.g., wifi) where `Lk_kit::initialize` needs to be called before `exec_static_constructors`. Therefore, we want register these initcalls by calling a lx_kit specific (generated) function.

ssumpf, 2024-01-22:
The commits above implement `module_init` calls and friends by generating _initcall_table.c_ which contains a `lx_emul_register_initcalls` function that is called in `lx_emul_start_kernel`. The init calls are now global symbols that point to the functions that where formally registered as constructors. The function pointers are collected using `nm` during creation of `lx_emul_register_initcalls` and executed in this function.

[984f4b3](https://github.com/ssumpf/genode-imx/commit/984f4b3961319525be0b366f0c9291e0a01bcae2) adds the code generation rules for the MNT Reform.

[df69798](https://github.com/ssumpf/genode/commit/df69798a3acda5a0e26a8c559ee5ce56450f20c0) and [16bba66](https://github.com/ssumpf/genode/commit/16bba667288d95a02d950a3509c79b1a225d76c1) remove `Component::construct` which is not needed anymore and the left over static constructors from our Wifi driver, which is a shared library.

skalk, 2024-01-23:
I'm very fine with the commit series. Thanks for the good solution @ssumpf.
One line grabbed my attention, which is the usage of `awk` that we did not used till now. But as this tool is used probably already implicitely when we call Linux build system to generate Linux headers, I think it shouldn't be an issue.

chelmuth, 2024-01-24:
Merged to staging.