genodelabs/genode #5102: ci: false positives due to error code overwrite

brfc, 2024-01-25:
On our CI pipeline we occasionally observed false negatives, where successful tests got reported as failed. After some debugging, we were able to pin down the problem to the fact, that the command used to power off the target failed in these cases.

To prevent the problem we catch errors from `run_power_off`.

Issue gapfruit#1953

chelmuth, 2024-01-30:
In your commit, you note *The solution is to catch possible errors from power-off within the exit procedure.* Could you please try to explain in more detail why this helps to prevent *a failing run_power_off lead to false-positives*?

chelmuth, 2024-04-26:
> In your commit, you note _The solution is to catch possible errors from power-off within the exit procedure._ Could you please try to explain in more detail why this helps to prevent _a failing run_power_off lead to false-positives_?

@brfc I'm still struggling to grasp the issue you fix. Without further information I tend to close this issue as wontfix.

brfc, 2024-04-26:
@chelmuth, this happened some time ago and I can't recall all the details. However, from what I remember, the situation was as follows:

 - A test failed and exited with code 1, which was necessary to indicate test failure in the pipeline.
 - The new exit procedure was executed.
       - During execution, the run_power_off function was called.
       - However, run_power_off failed and exited with code 0.
  - The execution of the new exit procedure was interrupted by the exit call from run_power_off.

As a result, a final exit code of value 0 was produced, even though the original error originated from a failed test, which should have produced an exit code of 1. Consequently, gapfruit pipelines, as all pipelines affected in this case, would receive a return value of 0, and produce the according green pass. In fact, what matters is the first exit code (1), and not the latter, which is produced during a tear down event. 

The catch is a sound way of addressing the issue. Since is logically correct. 

For more information, or issue continuation, please tag @trimpim 

Best regards, 
Thank you for your attention

closed 2024-05-02 by chelmuth