genodelabs/genode #5103: tool: add `backtrace`

cproc, 2024-01-25:
Add a tool which decodes the output of `Genode::backtrace()` and other memory addresses to the corresponding source locations.

cproc, 2024-01-25:
Commit edb8709 implements the tool. It is inspired by the Fiasco.OC backtrace tool and used in a similar way, but implemented in Tcl and tailored to Genode.

```
Usage (in Genode debug directory):

[KERNEL=...] backtrace [binary]

then paste shared library info and backtrace addresses.

If the KERNEL environment variable is set, the file name 
'ld.lib.so' will be replaced by 'ld-KERNEL.lib.so'
```

Example:

```
KERNEL=linux ../../../tool/backtrace test-log
```

```
Scanned image test-log
[init -> test-log]   0x1000000 .. 0x10ffffff: linker area                                                                                                    
[init -> test-log]   0x40000000 .. 0x4fffffff: stack area                                                                                                    
[init -> test-log]   0x50000000 .. 0x601b3fff: ld.lib.so                                                                                                     
[init -> test-log] backtrace "ep"                                                                                                                            
[init -> test-log]   401ff808   1000643                                                                                                                      
[init -> test-log]   401ff898  500b9581                                                                                                                      
[init -> test-log]   401ffac8  5006af80                                                                                                                      
[init -> test-log]   401ffb38  50053cba                                                                                                                      
[init -> test-log]   401ffde8  5008d54a                                                                                                                      
[init -> test-log]   401fff58  500a4a57                                                                                                                      
                                                                                                                                                             
                                                                                                                                                             
                                          
Scanned image ld-linux.lib.so

test_func()

    * 0x1000643: test-log:0x1000643 T
    * /.../repos/base/src/test/log/main.cc:24

Component::construct(Genode::Env&)

    * 0x500b9581: ld-linux.lib.so:0x500b9581 t
    * /.../repos/base/src/lib/ldso/main.cc:432

Genode::Rpc_object<(anonymous namespace)::Constructor, (anonymous namespace)::Constructor_component>::dispatch(Genode::Rpc_opcode, Genode::Ipc_unmarshaller&, Genode::Msgbuf_base&)

    * 0x5006af80: ld-linux.lib.so:0x5006af80 t
    * /.../repos/base/include/base/ipc_msgbuf.h:173

Genode::Rpc_entrypoint::entry()

    * 0x50053cba: ld-linux.lib.so:0x50053cba T
    * /.../repos/base/src/lib/base/rpc_dispatch_loop.cc:71

Genode::Thread::_thread_start()

    * 0x5008d54a: ld-linux.lib.so:0x5008d54a t
    * /.../repos/base/include/base/blockade.h:43

thread_start

    * 0x500a4a57: ld-linux.lib.so:0x500a4a57 t
    * /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:64
```

ssumpf, 2024-01-26:
@cproc: Very nice!

chelmuth, 2024-01-26:
> @cproc: Very nice!

Huge understatement - it's marvellous! ;-)

alex-ab, 2024-01-26:
@cproc: kudos

chelmuth, 2024-01-26:
Using the tool for *x86_32* does not work.

```
build/x86_32/debug> ~/plain/src/genode/genode_staging.git/tool/backtrace test-log
DEBUG file_output: test-log: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter ld.lib.so, stripped
Error: could not obtain architecture from image file
```

How about using *readelf*, which would also be available on Sculpt with our tool chain and works for all archs.

```sh
for d in arm_v6 arm_v7a arm_v8a riscv x86_32 x86_64; do
  genode-arm-readelf -h $d/debug/test-log | grep 'Machine:'
done

  Machine:                           ARM
  Machine:                           ARM
  Machine:                           AArch64
  Machine:                           RISC-V
  Machine:                           Intel 80386
  Machine:                           Advanced Micro Devices X86-64
```

cproc, 2024-01-26:
> How about using _readelf_, which would also be available on Sculpt with our tool chain and works for all archs.

So, just `readelf` and on Sculpt have a `readelf` binary accessible via `PATH` (for example `/usr/local/genode/tool/23.05/x86_64-pc-elf/bin/readelf` on x86_64 Sculpt)?

chelmuth, 2024-01-26:
> > How about using _readelf_, which would also be available on Sculpt with our tool chain and works for all archs.
> 
> So, just `readelf` and on Sculpt have a `readelf` binary accessible via `PATH` (for example `/usr/local/genode/tool/23.05/x86_64-pc-elf/bin/readelf` on x86_64 Sculpt)?

That's a question I have no answer to and why I used *genode-arm-readelf* in my example. May we expect *genode-x86-readelf* is always available on Sculpt for now?

cproc, 2024-01-29:
Fixup commit 4aaddd9 uses `genode-x86-readelf` instead of `file` to detect the architecture.

skalk, 2024-01-29:
@cproc: thanks a lot!

commit d893edd5 (2024-01-26): tool: add `backtrace`

Fixes #5103
closed 2024-02-29 by chelmuth