genodelabs/genode #5104: lxip: Update to current DDE-Linux version

ssumpf, 2024-01-29:
Update lxip stack to Linux 6.1.20

ssumpf, 2024-01-29:
Related to #4958.

ssumpf, 2024-01-29:
The commits above are a first version of the lxip update to Linux 6.1.20. I think it would be good to integrate this into nightly testing.

Some notes:
* the `arm_v6|7|8` split is also continued so the _issue5104_lxip_update_ branches from my _genode_, _genode-  allwinner_, _genode-imx_, _genode-rpi_, and _genode-zynq_ repos are required (I have built all the driver packages I could find for all architecture they support)
* I disabled IPv6 in virt_linux for now because it caused issues
* [baa4ea6](https://github.com/ssumpf/genode/commit/baa4ea6098c1847fd5ce0001af03f3fc460ec1ee) contains a low level test that uses the C-API directly without libc or the vfs plugin, there exists a run script `lxip_raw.run`,  which we could later integrate into _depot-autopilot_
* `vfs_lxip` has been adjusted to the new C-API of lxip
* tested with `run/lighttpd_fetchurl` on x86_32, x86_64, arm_v6|7|8
* riscv support is missing
* https://github.com/genodelabs/genode/issues/4810 is probably still an issue
* the Falkon  browser behaves the same as with the old version

In the next step I would like to remove `legacy_lxip`, `vfs_legacy_lxip`, the `dde_linux.port`, and `legacy/lx_kit`.

ssumpf, 2024-01-29:
P.S. The netrperf TCP_MAERTS (sending) test shows constantly 1/2 of the performance of TCP_STREAM (receiving), this should be addressed also.

ssumpf, 2024-01-30:
5c592dd renames all legacy_lxip patches to "legacy_lxip*".

ssumpf, 2024-01-30:
@chelmuth: a483a2c add  some cosmetic changes

chelmuth, 2024-01-30:
I merged the series to staging. Thanks for the great work. Please address my remarks in the commits.

ssumpf, 2024-01-30:
@chelmuth: b57e4ca, [b691861](https://github.com/ssumpf/genode/commit/b69186191ceb4ff5274f370ce8d4288a749b3158), and [08438ca](https://github.com/ssumpf/genode/commit/08438caae1a924002231c91402ae8125991d342a) address your remarks.

ssumpf, 2024-01-30:
@chelmuth: As far as I can tell you also need to update genode-allwinner, genode-imx, genode-rpi, and genode-zynq with the commits above because of [9022f74](https://github.com/genodelabs/genode/commit/9022f74177b121347dffbee09cf401280a79e868) all dde_linux based drivers will break there.

chelmuth, 2024-01-31:
> @chelmuth: As far as I can tell you also need to update genode-allwinner, genode-imx, genode-rpi, and genode-zynq with the commits above because of [9022f74](https://github.com/genodelabs/genode/commit/9022f74177b121347dffbee09cf401280a79e868) all dde_linux based drivers will break there.

Sorry, again I missed to push the prepared branches. It's quite easy to lose track of these cross-cutting adaptions.

ssumpf, 2024-01-31:
> Sorry, again I missed to push the prepared branches. It's quite easy to lose track of these cross-cutting adaptions.

I know, but also don't have a good solution.

chelmuth, 2024-01-31:
I fixed some cap quotas in depot_autopilot (8552ae74a99b20e2b1184153d6dd7ac3aeded667), but have two remaining issues in, e.g., test-tcp_bulk_lxip.

1. ARM platforms tested in Qemu (zynq_qemu, pbxa9) now fail with `Error: jitterentropy library could not be initialized!`
2. imx53_qsb fails with `ep: undefined instruction at ip=0x116042c` which may be a lacking instruction in Cortex A8 in comparison to later versions.

I'll look into the second issue next - @ssumpf could you have a look into the first one?

ssumpf, 2024-01-31:
> I fixed some cap quotas in depot_autopilot ([8552ae7](https://github.com/genodelabs/genode/commit/8552ae74a99b20e2b1184153d6dd7ac3aeded667)), but have two remaining issues in, e.g., test-tcp_bulk_lxip.

 :+1:
 
> 1. ARM platforms tested in Qemu (zynq_qemu, pbxa9) now fail with `Error: jitterentropy library could not be initialized!`
>
>     2. imx53_qsb fails with `ep: undefined instruction at ip=0x116042c` [...]

Will look into it.

chelmuth, 2024-01-31:
Also let me put *linux/netperf_lxip* on the list.

ssumpf, 2024-01-31:
The zynq problem with jitternentropy is that

! asm volatile("mrc p15, 0, %0, c9, c13, 0" : "=r"(t));

returns always zero in `jent_get_nstime` for Cortex-A9 on Qemu (zynq and pbxa9) while Cortex-A15 (virt_qemu_arm_v7a) works as intended. I will have to find out if this is a Qemu or Cortex-A9 limitation. @skalk: Do you have more insight by any chance?

ssumpf, 2024-01-31:
@chelmuth: I took a small detour into the Qemu (6.2.5) sources, in _target/arm/cpu_tcg.c_, there in `cortex_a9_initfn` the `set_feature(&cpu->env, ARM_FEATURE_PMU);` is missing which is present in `cortex_a15_initfn`. This means performance counters are not working for Cortex-A9 (pbxa9 and zynq_qemu) which the `mrc` above tries to read, but for Cortex-A15 (virt_qemu_arm_v7a). This in turn leads to `jent_get_nstime` to always return 0 and jitterentropy not initializing. Since the `lx_emul` random implementation does require jitterentropy, lxip will not work in Qemu for these platforms. A short test to enable the PMU in Qemu for A9 worked, but requires a Qemu patch, which is most likely not what we want. The question now is how to proceed with Qemu and Cortex-A9? Maybe implement a different random approach not using jitterentropy for lxip?

nfeske, 2024-01-31:
> Maybe implement a different random approach not using jitterentropy for lxip?

It would be best to replace the hardwired use of jitterentropy by an `lx_emul` interface for obtaining seed values. `vfs_lxip` could then fill out this interface by reading seed values from a file (the path of the file could be obtained from a config attribute `random_path`). This would lay the decision about the seed source into the hands of the user, who may normally let this path point to a `vfs_jitterentopy` instance. However, our tests could let this path refer to a static `<inline>` file.

ssumpf, 2024-01-31:
@chelmuth: In the meantime I have implemented a quick fix, which does not use jitterentropy in case initialization fails. I would consider this an interim fix to get platforms without proper TSC/performance counters to work and look into the the approach suggested by @nfeske next.

chelmuth, 2024-01-31:
>     2. imx53_qsb fails with `ep: undefined instruction at ip=0x116042c` which may be a lacking instruction in Cortex A8 in comparison to later versions.

The faulting instruction is `pldw` which indeed is only available on SMP-capable systems. There quite some wizardry in Linux to detect if an SMP kernel is running on UP systems (see `ALT_SMP()` and `CONFIG_SMP_ON_UP`) which also includes boot-time binary patchin in `__fixup_smp_on_up`. Therefore, I vote for disabling `CONFIG_SMP` in virt_linux on arm_v7a.

As I got quite a pile of compiler errors on a smoke test: @ssumpf would you mind to have a look into this option?

ssumpf, 2024-01-31:
> As I got quite a pile of compiler errors on a smoke test: @ssumpf would you mind to have a look into this option?

@chelmuth: Check.

chelmuth, 2024-01-31:
> In the meantime I have implemented a quick fix, which does not use jitterentropy in case initialization fails. I would consider this an interim fix to get platforms without proper TSC/performance counters to work.

I agree that this is a valid interim solution. If you go for an lx_emul seeding interface please have a look into #4957 and f66a7db87f55ab0075d522d981965512443e783d where we designed an interface to control the initial persistent-clock time (for wireguard).

ssumpf, 2024-01-31:
> > As I got quite a pile of compiler errors on a smoke test: @ssumpf would you mind to have a look into this option?
> 

I have looked into this, the biggest problem is that the _shadow/rcu/tree.c_ gets in the way on UP. _rcutree.h_ includes cause a major mess on UP. This is because UP uses _rcu/tiny.c_ - which I will have to shadow too - and _rcutiny.h_. This looks feasible, but might take a little  longer than expected.

chelmuth, 2024-02-01:
Unfortunately, I don't know of any other feasible option. @skalk what's your opinion?

ssumpf, 2024-02-01:
8edd6a7 disabled SMP in virt_linux for arm_v7 only. adfcfe9, 1a62acf, and 5b1f033 perform the necessary adjustments for lxip, usb_net, and usb_hid.

skalk, 2024-02-01:
The solution looks good to me! And much less painful than doing some binary patching like solution similar to an ARMv7 generic SMP Linux kernel on an UP system is doing.

chelmuth, 2024-02-01:
> Edit: Unfortunately I cannot test usb_net nor usb_hid on arm_v7.

I tried to run *run/usb_hid_raw* but failed.

```
./tool/depot/create genodelabs/bin/arm_v7a/usb_hid_drv FORCE=1 UPDATE_VERSIONS=1
...
      [Linux]  .config:69:warning: override: NO_HZ_IDLE changes choice state
      [Linux]  In file included from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/include/linux/ratelimit_types.h:7,
      [Linux]                   from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/include/linux/printk.h:9,
      [Linux]                   from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/include/asm-generic/bug.h:22,
      [Linux]                   from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/arch/arm/include/asm/bug.h:60,
      [Linux]                   from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/include/linux/bug.h:5,
      [Linux]                   from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/include/linux/page-flags.h:10,
      [Linux]                   from ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/kernel/bounds.c:10:
      [Linux]  ../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/include/linux/spinlock_types_raw.h:9:11: fatal error: linux/spinlock_types_up.h: No such file or directory
      [Linux]      9 | # include <linux/spinlock_types_up.h>
      [Linux]        |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      [Linux]  compilation terminated.
      [Linux]  make[8]: *** [../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/scripts/Makefile.build:118: kernel/bounds.s] Error 1
      [Linux]  make[7]: *** [../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/src/linux/Makefile:1289: prepare0] Error 2
make[6]: *** [../genode.git/depot/genodelabs/api/virt_linux/2024-02-01/lib/mk/virt_linux_generated.inc:27: kernel_config.tag] Error 2
make[5]: *** [var/libdeps:229: virt_linux_generated.lib.a] Error 2
make[4]: *** [Makefile:351: gen_deps_and_build_targets] Error 2
make[3]: *** [../genode.git/depot/genodelabs/bin/arm_v7a/usb_hid_drv/2024-02-01.build/bin] Error 1
make[2]: *** [../genode.git/depot/var/build.mk:16: genodelabs/bin/arm_v7a/usb_hid_drv/2024-02-01] Error 2
make[1]: *** [../genode.git/tool/depot/build:159: execute_generated_build_mk_file] Error 2
make: *** [tool/depot/create:59: build] Error 2
```

ssumpf, 2024-02-01:
@chelmuth: I forgot to build the packets and update the 'dep.list', fixup will follow.

chelmuth, 2024-02-01:
With usb_hid_drv from the build directory it works. I don't know how to test usb_net_drv, though... Ideas?

chelmuth, 2024-02-01:
@ssumpf: the builder failed to `make lib/lx_kit_setjmp` for arm_v6/7a/8a. Is this library still used?

ssumpf, 2024-02-01:
Netper

> With usb_hid_drv from the build directory it works. I don't know how to test usb_net_drv, though... Ideas?

As a simple test you can use the _dde_linux/run/usb_net.run_, you will need to connect an ASIX 88179 dongle to a board and update 
```
append_if [have_spec x86] config {
  <policy label_prefix="usb_net_drv" vendor_id="0x0b95" product_id="0x1790"/> }
```
the `have_spec x86` to `have_spec arm_v7`

ssumpf, 2024-02-01:
> @ssumpf: the builder failed to `make lib/lx_kit_setjmp` for arm_v6/7a/8a. Is this library still used?

It's unused, currently this is only included in the `vfs_legacy_lxip` recipe for x86* for some reason. So you can delete them. The rest I will remove along with the `legacy/lx_kit`.

ssumpf, 2024-02-01:
@chelmuth: 5952fcb updates the 'dep.list' for virt_linux to cover tiny RCU. The packets should build with this.

chelmuth, 2024-02-05:
@ssumpf we got an unimplemented function panic

```
[2024-02-04 04:18:32] [init -> netserver_genode] could not send timeout for recv - test may not terminate
[2024-02-04 04:18:32] [init -> netserver_genode] Warning: timestamp value too big, ts=10538981 max_ts=8388607
[2024-02-04 04:18:32] Kernel: init -> nic_drv -> nic_drv -> ep tried to do cache maintainance at unallowed address 0xfff800 <- unrelated - reopened issue #4709 
[2024-02-04 04:18:32] [init -> netserver_genode] Error: Function div64_u64 not implemented yet!
```

ssumpf, 2024-02-05:
> @ssumpf we got an unimplemented function panic

@chelmuth: arm_v7 again? d6b2328 adds `div64_u64` to all 32-Bit architectures.

ssumpf, 2024-02-07:
* 4b9aa07 and e0ddfdd are my first take on letting vfs_lxip handle the random number seeding for lxip, I used a local _random.cc_ for lxip such that not all other drivers are affected 
* 5b19ee3 adds `random_path` to recipes and run scripts that use lxip as well as to the sculpt_manager
* b1e6314 adds a README to vfs_lxip
* 904e5ac removes _ipv4/tunnel4.c_ from the _source.list_ files of lxip because the  `INET_TUNNEL`  configuration is not set in virt_linux and the corresponding init call got rightfully removed from _spec/initcall_order.h_

ssumpf, 2024-02-08:
https://github.com/ssumpf/genode-world/commit/ea182f3456694175fd06f9b9ac518a7697361606 in world adds `random_path` to recipes and run scripts there

ssumpf, 2024-02-08:
53eed54, 6b1ba6f, and 531a167 should enter staging last for this issue. They remove `vfs_legacy_lxip`, `legacy_lxip`, `dde_linux.port`, and legacy `lx_kit/lx_emul`.

ssumpf, 2024-02-09:
38417a0 adjusts pc_linux to the lx_kit removal

chelmuth, 2024-02-12:
I merged all but the *random-path* commits for now, because I'd like to decouple the removal of jitterentropy from lx_emul from this issue.

ssumpf, 2024-02-16:
@chelmuth: 5f63083 and e0c079a resolve the Falkon `ERR_SOCKET_NOT_CONNECTED` problem using lxip, both as plugin Falkon and in VFS server, for me. Thanks for the hint.

ssumpf, 2024-02-16:
Just for documentation, with 5e24900 and 32ac426 there will be many error message in the VFS server case like:
```
[init -> falkon] Error: packet operation=0 failed
[init -> falkon] Error: socket_fs: reading from the connect file failed
```
But it works.