genodelabs/genode #5106: os: `Path_base::strip_double_dot_dirs()` not working correctly

cproc, 2024-01-31:
When a `Genode::Path` is initialized with `/dir1/dir2/..`, it gets converted to `/dir1` as expected, but `/..` or `/dir1/..` result in an empty path instead of `/`.

cproc, 2024-01-31:
Is `Genode::Path` supposed to be always an absolute path? I thought so at first, but `Genode::Path<128> path { "dir", "pwd" }` results in `pwd/dir`, so I'm not sure anymore.

nfeske, 2024-02-06:
The primary concern addressed by the `Path` utility is the sanitizing of the string regarding path separators and the `..` patterns (that might we misused to attempt the escaping of directory scopes). It does not try to enforce a leading slash as an invariant. In places where absolute paths are expected, there exists the typedef `Absolute_path` (vfs/types.h) to express this expectation. But it is not enforced by the `Path` utility.

cproc, 2024-02-07:
Commit 8af123f restores the leading slash in `strip_double_dot_dirs()` if the original path was absolute and the stripped path is empty, which fixes the initial cases of this issue, but there are also some problematic cases with relative paths, for example `dir1/../dir2` gets converted to `/dir2` or `../../dir1` gets converted to `/dir1`. Should a relative path starting with `..` get converted to an empty string in general to prevent directory scope escaping or should a leading `..` or `../..` be preserved instead?

nfeske, 2024-02-08:
Thanks for bringing up these corner cases. A relative path should certainly stay relative. The resulting path must not contain any `..` element after the call of `strip_double_dot_dirs`.

Would you consider adding a unit test covering those corner cases for the depot_autopilot?

cproc, 2024-02-12:
Commit dbb88e7 fixes the corner cases mentioned in this issue and commit a78777d adds the autopilot test.

cproc, 2024-02-12:
Fixup commit 649aa03 removes init from the runtime.

chelmuth, 2024-02-29:
Merged to master.