genodelabs/genode #5113: hw: make virtualization implementation more generic

atopia, 2024-02-09:
The implementation of x86 virtualization did not lend itself to a runtime selection of the x86 virtualization technology.

Make the general virtualization support less specific to AMD's SVM and encapsulate functionality in the VMCB class to facilitate adding support for Intel's VMX.

atopia, 2024-02-13:
The renewed series of commits accesses the VMCB State Save Area via the Mmio framework.

chelmuth, 2024-02-14:
@atopia I'd like to defer the review and merge until you come up with the final addition of the VMX implementation. Only then, we may discuss some design decisions you made because of the virtualization-technology duality. Do you agree?

atopia, 2024-02-19:
Sorry for the late reply, your comment slipped through. Yes, I'm fine with doing the review and merge together with the VMX implementation.

chelmuth, 2024-05-13:
I merged the series. Please address my comments in fixup commits.

atopia, 2024-06-17:
Fixed in master.

closed 2024-06-17 by atopia