genodelabs/genode #5115: hw: refine scheduler class for better comprehensibility

skalk, 2024-02-12:
Try to make API more slim, reduce state variables, find more common namings. These changes should not change any semantics at this point.

skalk, 2024-02-12:
There are still some things that could be nicer, like getting rid of the `_current` pointer to the currently scheduled context, but for the time-being I would mark this issue as being fixed.

nfeske, 2024-02-12:
I very much welcome these changes. @jschlatow do you concur?

jschlatow, 2024-02-12:
I haven't reviewed the commit series in detail (only the commit messages), yet I very much appreciate the improved readability as it will simplify future modifications.

nfeske, 2024-02-14:
I just tested these changes for Sculpt on PinePhone, where everything works fine. Hence, I have merged the commit series to staging now.

closed 2024-02-29 by chelmuth