genodelabs/genode #5117: base-hw: scheduling fairness and latency optimisation

jschlatow, 2024-02-14:
As discussed in #4796, there are a few scheduling oddities in base-hw. I did a litte bit of research on how we could improve base-hw scheduling and want to share and discuss my findings in this issue.

First of all, let me briefly summarise what I believe are the requirements:

* Scheduling must be fair, i.e. if there are N active threads (equal weight), each thread should receive a 1/N-th share of the CPU in a certain time interval. Additionally, a thread must be served proportional to its weight, i.e. a thread with weight 2 must receive double the CPU time as a thread with weight 1.
* No thread shall be able to monopolise the CPU
	* neither by running an infinite loop (follows from fairness)
	* nor by waking up frequently
* Low-latency threads shall be scheduled faster on wakeup. They might also be scheduled with smaller time slices and more often.
* High-throughput threads shall be able to run longer without preemptions (larger time slices).
* There should be support for temporarily boosting low-latency threads, e.g. as a consequence of IRQ processing. Moreover, this boosting should be possible across a chain of threads, e.g. to achieve low-latency IRQ processing all the way into a vbox guest.
* There should be support for donation/helping in case of IPC.

Next, let me share my view on common scheduling policies.

## Round Robin (RR)

* RR is fair but all threads have the same weight and get a fixed time slice (slot).
* The latency of a thread depends on the number of active threads.
* The figure below illustrates the latency issue with RR. When thread A becomes
  active just after B is scheduled the second time, it must wait for B, C and D
  before it is scheduled.

[image]

## Weighted RR (WRR), interleaved

* Threads are assigned integer weights w_i.
* A "hyperschedule" (my term) consists of max(w_i) rounds.
* A thread i gets a slot in round j if j <= w_i.
* The problem with this scheme is that the distance between two executions of a particular thread varies.
* The figure below illustrates this. Thread A has weight 6 and threads B, C and D each have weight 2. The first and second execution of thread B are 4 slots apart whereas the second and third execution are 8 slots apart.

[image]

## WRR, uniformly interleaved

* Threads are assigned integer weights w_i.
* A hyperschedule consists of N=LCM(w_i) (least common multiple of all w_i) rounds.
* A thread i gets a slot in every N/w_i round.
* This equalises the distances between two consecutive executions of threads B, C and D.
* The figure below illustrates this. The thread weights are the same as above

[image]

However, because the distance between two executions of thread A is not constant, there can still be rather long latencies as in RR. This is illustrated in the figure below in which thread A becomes active in round 6 and needs two slots to finish. Even if we allow A to execute in the already started round 6, it will only finish after 5 slots.

[image]

## WRR, "optimal"

In contrast to the above interleaving schemes, we could aim for an interleaving as illustrated below.

[image]

This scheme minimises and equalises the latency of thread A over the entire schedule. It also allows for latency tuning of individual threads by assigning higher weights and reducing their time slice.

The problem with this scheme is that even though threads B, C and D have equal weight, B is preferred significantly over D as it is scheduled already in the second slot although all threads have been activated at the same time. Moreover, I am not aware of any rule or implementation that would produce such a schedule.

## Virtual-time scheduling (e.g. stride scheduling)

* There are multiple scheduling policies that operate with virtual time.
* The concept of virtual time is an elegant trick to construct a schedule similar to that of uniformly interleaved WRR.
* The basic idea is to sort all active threads on a virtual time scale.
* For each thread, virtual time advances inversely proportional to its weight.
* After each execution, a threads virtual time is incremented by real-time slice / weight.
* The scheduler picks the thread with the lowest virtual time.
* It is common for virtual-time scheduling implementations to use a self-balancing binary tree for looking up the next thread to be scheduled.
* The figure below illustrates this for threads A, B, C and D with weights 1, 1/3, 1/3 and 1/3. Real-time is normalised to the time-slice length.

[image]

* A nice property of this scheme is that it allows for different time slices and joining/leaving threads.
* However, threads with lower weight will have higher latency. Hence, it does not  provide good latency tuning.
* Moreover, as shown in the figure, thread A gets most of its time slices only after threads B, C and D have been running.

## Earliest Eligible Virtual Deadline First (EEVDF)

* This is a scheduling scheme proposed in a paper from 1996 by Stoica and Abdel-Wahab.
* It has recently been implemented as the default scheduler in Linux as a replacement for CFS.
* It is a virtual-time scheduling scheme that, in contrast to e.g. stride scheduling, computes a virtual deadline for every thread in order to make the scheduling decisions.
* The virtual deadline is constructed by adding the thread's time slice length to its virtual start time (aka eligible time).
* By changing the time slice length, one is able to tune the latency of threads (to some degree). However, a thread's latency is basically proportional to its time slice / weight. A small weight, small time-slice thread thus might still achieve a larger latency than a large weight, large time-slice thread.
* As far as I understand, EEVDF should result in schedules similar to WF2Q but aimed at CPU scheduling where we do not know the finish time of a thread.
* The figure below illustrates EEVDF scheduling on the previous example. A EEVDF scheduler would first execute thread A multiple times since it takes the virtual time-slice length of threads B, C and D into account.

[image]

## Borrowed-Virtual-Time (BVT) scheduling

* This is a scheme proposed in a paper from 2000 by Duda and Cheriton.
* The paper is a very interesting read. The mechanisms is relatively simple and most of the paper focuses on configuration examples for different work loads.
* It is basically a virtual-time scheduling scheme like stride scheduling with an additional mechanism to let a thread borrow its own time from the future.
* It introduces an optional warp value for each thread. On thread wakeup, the warp value is substracted from the current virtual time. This allows the thread to be scheduled earlier. Warping can be restricted by a warp time limit (maximum time a thread may execute in warped state) and an unwarp time requirement (minimum time between two warp phases).
* In my opinion, BVT is a very good candidate for base-hw scheduling:
	* We can map CPU quota to weight.
	* We can map priorities to warp values.
	* We could probably implement an inheritance mechanism for warp values in order to boost IRQ handling.

jschlatow, 2024-02-14:
My plan is to implement the current base-hw scheduling as well as BVT prototypically in Python so that I can run some simulations to see what we could gain in terms of fairness and latencies. In the meantime, I would appreciate any feedback on my previous post.

m-stein, 2024-02-15:
@jschlatow Thanks for the interesting overview!

nfeske, 2024-02-15:
@jschlatow wow! What a great collection of insights. Thank you very much for bringing the topic in a digestible form for us.

Since you asked about requirements, let me add the following: For scenarios where energy consumption is the most pressing concern (think a mobile phone while the screen is blanked), I'd desire being able to express that a CPU should be idle for a specified percentage of time. While the screen is blanked, a heavy application like a browser may still continue to execute (e.g., sending keep-alive network packets from time to time to keep a network connection open) but it would merely be able to induce a tiny CPU load, even in the worst case when spinning. This would be a form of CPU throttling that is independent from any platform-specific CPU voltage/frequency-scaling opportunities.

jschlatow, 2024-02-15:
@nfeske I've never looked into energy-aware scheduling. My intuition is that we need to temporarily break the work-conserving property to address your requirement. A simple solution could be to add a sleeper thread that gets a large time slice and puts the CPU to sleep for its entire time slice once it is scheduled.

What I take from your input is that a simple mapping of CPU quantum to weight might be not expressive enough. Maybe we need to make it configurable at some level whether a process prefers having its CPU quantum in larger or smaller time slices.

jschlatow, 2024-02-23:
## Comparison Stride vs BVT (no warping)

When implementing a basic version of BVT scheduling without warping, I took the opportunity to compare it against stride scheduling. Both policies are actually very similar as they pick the job with lowest virtual execution time. However, in stride scheduling, a job gets a time slice of fixed length whereas BVT lets a job execute until its virtual time is C/w_i ahead of any other pending job. The resulting difference is that BVT achieves significantly less overhead (i.e. context switches) for long-running jobs.

I implemented both policies so that threads that wake up from sleep are preferred over already running jobs (still considering their virtual time). Moreover, if multiple threads wake up at the same time and have the same and minimum virtual time, the one with the higher weight is selected. When a thread wakes up from sleep in BVT, it will be scheduled immediately if it is eligible (i.e. it has minimum virtual time).

When comparing stride scheduling (time slice 10ms) with BVT (C=10ms), I was wondering about a few putative oddities. Note that the simulation does not consider context switch overhead, hence BVT's benefit of producing less context switches is ignored.

### First scenario

The first scenario consists of two 90ms periodic threads (A and B). A has half the weight of B. Another long-running background job arrives at 0.2s with 1/3rd of A's weight. See [test_bvt1.csv](https://github.com/jschlatow/schedsim/blob/main/input/test_bvt1.tsv)

When each job of A and B takes 15ms, stride scheduling results in slightly better maximum response times for A and B compared to BVT. Counterintuitively, the average (mean and median) response times of A were better for BVT despite the lower weight of A. Whereas B got significanly worse average response times in case of BVT. Looking at the schedule reveiled, that stride scheduling created BBAA when no background job was present whereas BVT resulted in BAB.

The figure below shows a simplified execution trace for stride scheduling. The dark blue bars show the time from a job's first execution to its completion (including preemptions). The light blue bars show the time from a job's arrival to its first execution (queueing delay). It illustrates that B completes before A is started in the second and third instance.

[image]

In the next figure (same scenario for BVT), we can see that the queueing delay for A is reduced due to the interleaved execution of B and A. The slightly longer maximum response time of A is explained by additional interference from C.

[image]

With this information, I came to the conclusion that increasing the computation requirements for A and B should benefit BVT. Increasing the job lengths to 25ms instead reversed the situtation: BVT achieves (significantly) better response times for A and B. The larger job lengths in combination with the weight difference between A and B let BVT run job B ahead of A such that B could complete a few times before A was started as illustrated in the figure below.

[image]

### Second scenario

I equalised the weights of threads A and B in the first scenario to see whether this changed anything for BVT. The difference to stride scheduling depending on the job lengths is similar. However, since A and B have equal weights, BVT does not prefer any thread. Nevertheless, for some values, BVT results in quite different response times for A and B while stride produces almost the same response times for both threads. I concluded that BVT can be quite sensitive to small changes.

The figures below show the trace for a job length of 30ms and 31ms respectively. While the latter results in similar response times, the former produces significantly better response times for A (maximum and median 10ms better) than for B. I'm not sure whether this has any practical relevance, though.

[image]

[image]

### Third scenario

In this scenario, I defined a long running job A with high weight and a 40ms periodic job C with 5ms length and 1/8th of A's weight. The weight thus reflects exactly the CPU share that job C requires.

Interestingly, stride scheduling again produced better response times (for C) than BVT in this scenario. For both, the maximum response time was quite large (30ms for stride and 35ms for BVT). Dissecting the trace, I noticed that thread C wakes up not being eligible most of the times. For stride scheduling, C is scheduled as soon as A's virtual time reaches C's virtual time. For BVT, however, A is allowed to run a little ahead of C's virtual time. This explains the greater response time. In principle, this could be mitigated by making sure that A only runs ahead of another thread's virtual time if the latter has already started. However, when C does not have minimum virtual time on wakeup, this means it has received enough of its share in the past and there should be no reason to schedule it with preference. The actual issue here is that C has too little weight. In other words, if C is a low-latency thread, its weight must be adjusted so that it is eligible when it wakes up.

I therefore experimented with changing the weight. Slightly increasing C's weight results in a much better maximum response time of 15ms for both, stride and BVT. The latter even achieves a significantly better mean and median response time. The figures below show the trace for stride and BVT scheduling respectively.

Stride:
[image]

BVT:
[image]

jschlatow, 2024-02-23:
## Comparison base-hw vs BVT (no warping)

I also compared my current BVT implementation with the base-hw simulation using the same scenarios described in my previous post.

Note that I implemented base-hw without any preferred handling of woken up threads, hence new jobs are added to the end of the round robin queue per priority.

### First scenario

With the 15ms job length, I get similar results as with stride scheduling, i.e. base-hw results in slightly smaller maximum response times. However, the minimum, median and average response time of thread A was significantly higher.

Note that A and B always arrive simultaneously and my base-hw implementation does not prefer one over the other (stride and BVT prefer B due to its higher weight). In my base-hw implementation, the order in which the threads appear in the input file has therefore an effect on which is scheduled first.

Increasing the job length to 25ms significantly worsens the reponse times for base-hw, even compared to stride scheduling (which was worse than BVT).

### Second scenario

For the 30ms job length, base-hw also shows a difference of 10ms between thread A and B. Moreover their maximum response times are approx. 8ms larger than for BVT.

With a job length of 31ms, thread B misses its deadline (=period) twice when simulated with base-hw.

[image]

### Third scenario

This scenario behaves nicely with base-hw, which can deal better with the short running job whose execution requirements exactly match its weight/quota.

With a slight increase in thread C's weight, base-hw and BVT produce the same results.

m-stein, 2024-02-24:
@jschlatow Very insightful! Thanks for this careful and visually appealing analysis!

jschlatow, 2024-03-01:
## BVT with warping

This week, I implemented warping for BVT. With warping, each thread not only has its actual virtual time (AVT) but also an effective virtual time (EVT). A thread's EVT is simply its AVT minus the thread's warp value. The BVT scheduler picks the thread with the smallest EVT. This way latency-sensitive threads can be scheduled preferrably. Note when a thread wakes up and its AVT is smaller than the minimum AVT among all other threads, its AVT is set to this minimum since a sleeping thread should not be eligible for a larger CPU share on wakeup.

There are a few points of unclarity in the original paper, though. Each thread is allowed to run a fixed (real) time C ahead of another thread. This mechanism is definied on the AVT. A thread waking up with a non-minium AVT but having the smalles EVT is supposed to be scheduled immediately, but computation of its time slice may result in a time slice length of zero (because it is already C ahead of the thread with minimum AVT). Moreover, the scheduler is supposed to switch from thread i to thread j, once thread i has run C ahead of j. Since this is defined on AVT, it is not clear how the EVT influences this switching. Must thread j have the smallest EVT or should we switch to the thread with the smalles EVT once thread i has run C ahead of any other thread?

My interpretation was to adapt the definition such thread i is allowed to run C ahead of another thread comparing their EVTs not their AVTs. This is based on the fact that the configuration scheme provided in section 6.1.3 sets the warp value `W_i` of a thread i such that it is allowed to run at least for `W_i*w_i` (where `w_i` is the thread's weight) ahead of threads on lower priorities. This scheme would not be possible when we compared the AVTs as this would limit the execution time to `C*w_i`.

In addition to the warp values, threads can be assigned a warp limit (the maximum runtime, in real time, a thread is allowed to run warped) and an unwarp time requirement (the time a thread must remain unwarped after either suspending itself or being forced to unwarp).

Let me summarise my observations after playing around with this implementation a bit:
* By setting the warp values for periodic and latency-sensitive threads according to their computation requirements, one is able to achieve minimum latencies.
* Once a thread either exceeds its expected runtime or is activated more frequently than its assigned CPU share, its latency degrades.
* When such a thread exceeds its assigned CPU share, the latency seems to degrade more than when it requires more execution time than its warp value. The reason for this is that a warping thread i is allowed to execute for `(W_i+C)w_i` time ahead of a non-warping thread. After that, a non-warping thread j is allowed to run for at least `C*w_j`. If thread i is activated before that, it has to wait until thread j got its share.
* In comparison, for the current base-hw implementation, a thread exceeding its CPU share will push the scheduling of lower-priority threads to the end of the super period.
* If there are multiple warping threads with a similar activation period and runtime, setting an unwarp time requirement likely penalises one thread systematically. This is caused by the unwarp time being measured from the suspension of each thread. If a thread has been scheduled late once (because it competed and lost against another similar thread) its unwarp time requirement may force the thread to run unwarped on its next activation. In contrast, the winning thread was already able to satisfy its unwarp time requirement before being activated again. Since the loosing thread starts unwarped it will loose the competition again.
* Configuring the warp parameters is not trivial especially when we take into consideration that dynamically adjusting weights at runtime depending on the number of (active) threads might be computationally complex. Moreover, I would like to keep the weights integers.

I think with these insights it is time to take a step back and review the timing requirements that we face for Sculpt OS and see how these could map to scheduling parameters. I will do this in a subsequent posting.

jschlatow, 2024-03-04:
## Timing requirements for Sculpt OS

First of all, I think the current limitation to four priorities (background, desktop, multimedia, device drivers) is absolutely reasonable and matches the service classes of a general purpose OS quite well. I believe it makes sense to define what the requirements w.r.t. scheduling are for each of these priorities.

Let me have a go; please correct me if I'm making wrong assumptions here:

* There is no guarantee for threads on background priority to be scheduled in any timely manner. I.e. they may be delayed for multiple seconds under high load. Moreover, I would even allow starvation for background threads.
* Threads on desktop priority must not starve under high load on higher priorities!
* Moreover, short interactive jobs should be able to achive sub-second (<500ms?) latency to keep the (interactive) system responsive even under high higher-priority load or long-running same priority jobs.
* Threads on multimedia priority have to achieve a relatively low latency most of the times. I would consider it acceptable if they missed their low-latency requirement once in a while. It's hard to come up with a realistic number, though. I believe it should be smaller than 30ms since audio artifacts will become recognisable above that.
* Threads on device-driver priority have to achieve a low latency all the time. I am aware that there have been issues with interrupts not being handled fast enough. Maybe someone could remind me of the actual latency requirement in those scenarios.

### Consequences for a BVT-based scheduler

1. We could handle the background priority as idle priority.
  * For BVT, this can be reflected by a large warp value without any warp/unwarp limit. Background threads will then still be scheduled but potentially very rarely and for a limited time slice. However, if there is a large number of background threads, at some point, every background thread will get scheduled once before any (active) desktop thread is allowed to execute again. Since this only affects cpu-bound threads, this is probably acceptable. A thread woken up by a signal is scheduled with preference over the background threads. More precisely, it is allowed to execute `C*w_i` ahead of the background thread with the smallest virtual time. Should this not be enough to complete the job, however, the signalled thread will be affected by all the background threads catching up in their virtual time. Of course the same holds for device-driver and multimedia threads.
  * Alternatively, we could employ hierarchical scheduling such that background threads are only scheduled when there is no active higher-priority thread. This, however, can lead to complete starvation of background threads. The benefit of hierarchical scheduling is the effect of lower-priority threads on higher-priority threads is not dependent on their number or their weights.
2. Device-driver and multimedia priority should be configured such that a certain CPU share is still available to the desktop priority under extreme conditions.

   This can be achieved by employing hierarchical scheduling, which effectively reserves a certain CPU share for the group of all desktop threads. Employing hierarchical scheduling has the benefit that the weights of desktop threads can easily reflect their proportional CPU share. We can adjust the individual weights arbitrarily and work with very different weights without affecting the warp parameters of higher-priority threads. In contrast, for non-hierarchical scheduling, a desktop thread i could interrupt the execution of a low-weight higher-priority thread by `C*w_i+C` (real time). The entirety of desktop threads could therefore have a huge impact on higher-priority threads. As mentioned in my previous post, this should only occur when the high-priority thread exceeds its CPU share or is activated more frequently than expected. Nevertheless, I think the potential QoS degradation for high-priority threads in case of misconfiguration should be limitted.
4. Warp parameters for device-driver and multimedia threads can be adjusted to reflect their timing requirements.
  * Device-driver and multimedia threads can all get the same weight. There is probably no need for making sure individual threads get a certain CPU share because their requirements are more about latency. Keeping the weights for device-driver and multimedia threads low and similar simplifies the adjustment of warp parameters.
  * Maybe it makes sense to weigh multimedia threads higher than device-driver threads in case they need more computation time.

jschlatow, 2024-03-07:
## Configuring BVT for Sculpt

I experimented with adjusting warp values to achieve low-latencies for multimedia (MM) and device-driver (DD) threads. With very high warp values for MM and DD, one can achieve prioritised handling of MM and DD threads. For weight assignment,
I applied the following scheme:

* assign weight 1 to DD threads
* assign weight 2 to MM threads
* assign sum of all DD+MM thread weights to a single desktop thread

I observed that the low latency of DD and MM threads is maintained even when individual threads require a larger CPU share than defined by their weight. The missing CPU share is stolen from the desktop thread. However, at some point depending on the warp values, the latencies increase because the desktop thread is allowed to catch up. The larger the warp values the later this happens. You can find my test case in [test_sculpt1.tsv](https://github.com/jschlatow/schedsim/blob/main/input/test_sculpt1.tsv).

The problem with this scheme is that, when threads are added and removed at run time, the weights of desktop threads need to be adjusted, which is not a trivial operation.

As mentioned in my previous post, employing a hierarchical scheduling scheme allows a more direct control of CPU shares and warping. I implemented such a scheme in the simulator, which schedules between the group of desktop threads and the groupd of MM+DD threads on the first level. It assigns a fixed (guaranteed) share of 33% to all desktop threads and 66% to MM+DD threads. The actual weights only matter within the groups, so that we can work with weights 1 and 2 for DD resp. MM threads and with larger weights, representing the percentaged CPU share, for desktop threads. With this scheme, MM+DD threads cannot steal from the desktop threads any more. This also means that degradation becomes visible immediately.

### Comparison with base-hw

Both, base-hw and the hierarchical BVT, produce similar schedules. However, for base-hw we need to correctly specify the CPU shares. In practice, these are unknown. Moreover, underestimating a CPU quantum drastically increases the latency of a thread because it will not run priorities until the next super period. I am confident to claim that (hierarchical) BVT simplifies configuration and maintains low latencies as long as possible.

### Thoughts on donation/helping and boosting

I think that donation and helping can still be handled by "inheriting" a time slice to another thread. For boosting, I am going to evaluate how passing warp values along with signals is going to affect latencies of lower-priority threads. I implemented the hierarchical BVT in _direct mode_ (term from the paper), i.e. the warp values of the threads (handled at second level) are passed to the first-level scheduler. This scheme lends itself to a applying an inheritance scheme for warp values across priorities.

jschlatow, 2024-03-08:
## Boosting via warp-value inheritance

I have implemented boosting in the simulator as follows:
- A thread that was activated due to a signal from another thread is assigned the maximum of its own warp value and the signalling thread's warp value.
- The signalled thread is also assigned the minimum of its own warp time limit and the remaining limit of the signalling thread (i.e. its warp time limit minus the time it has already executed in warped state).

For the simulator, the implementation was very straight forward. One needs to keep in mind, though, that when boosting is used a warp time limit should be configured for MM and DD threads. Without boosting, the maximum time such a thread can run in warped state is implicitly controlled by the warp value and the thread weights. If all those threads have similar weights, a warp time limit is not needed. However, since lower-priority threads may have totally different weight values, a warp time limit should be configured when boosting is used.

jschlatow, 2024-03-11:
## Correction w.r.t. base-hw results

After talking to @skalk, I learned that my interpretation of the current base-hw scheduler was slightly off. In my simulations, I had used an implementation that does round robin on each priority (as long as there is a thread with prioritised time left). However, the base-hw scheduler actually lets a prioritised thread execute for its entire quantum. Threads that become ready in the meantime are added to the head of the queue.

Adapting the simulator accordingly produces much worse maximum response times if there is a long-running same priority thread. On the other hand, short periodic threads can get better response times as they are scheduled right away.

skalk, 2024-08-05:
I'll try to summarize the latest (offline) discussions resp. their outcome here:

We try to approach a two-level scheduling scheme. On the first level we schedule global groups of threads, you can think of them as being abstract categories like either device-driver, desktop, background,.... These groups have weights (somehow representing CPU runtime percentages) and a warp value (this represents the priority resp. the latency pushing factor). All contexts (threads, vcpus) are assigned to a group. Weight and warp value are used to implement BVT-scheduling of groups. Therefore, groups additionally need a virtual time. When a scheduling decision needs to be done, the group with the lowest effective virtual time (EVT) and a runable context gets chosen. EVT means virtual time minus warp value.  When a schedule was done, and a new schedule is needed, the real time spent by the last context gets divided by its group's weight and gets add to the virtual time of its group.

To make a decision within a group, we track the virtual runtime of the individual contexts too, but without applying additional weights nor warp values. We simply use the context with the lowest runtime that is runable.

Whenever a context gets ready, we check its group. If it is the only runable context in there, the group's virtual time potentially needs to be set at minimum to the lowest virtual time of all other groups. Generally, the context's virtual time is set to the group's virtual time, if it was lower than that.

One thing, which I'm not hundert percent sure about, is the duration calculation for a new schedule. As far as I understood it correctly, on the top level, we compare the difference of the chosen group's EVT to the next bigger EVT. This difference multiplied by the group's weight results in the real time the scheduler's timer gets programed to trigger an alarm at. But to minor scheduling overhead this value should not fall below constant C (e.g.: 10ms).
To combine this with the inner-group scheduling, we would take the minimum in between: top-level difference to next other group, and virtual time difference of current chosen context to next context within the group. This minimum then gets multiplied by the weight of the group, and is set as the scheduler alarm, as long as it is greater than C, otherwise C is used.
I hope this is reasonably understandable. @jschlatow do you agree?

jschlatow, 2024-08-05:
> I'll try to summarize the latest (offline) discussions resp. their outcome here:
>
> We try to approach a two-level scheduling scheme. On the first level we schedule global groups of threads, you can [...]

That's correct.

> One thing, which I'm not hundert percent sure about, is the duration calculation for a new schedule. As far as I understood it correctly, on the top level, we compare the difference of the chosen [...]

A minor correction: The difference multiplied by the groups weight is added to constant C. The logic is that the chosen context is behind in its virtual time (i.e. it has got less time than the other contexts). The difference multiplied by the groups weight lets the context catch up in its virtual time. By adding C, we allow the context to advance ahead of the other context's virtual time.

> To combine this with the inner-group scheduling, we would take the minimum in between: top-level difference to next other group, and virtual time difference of current chosen context to next context [...]

If we support warp limiting, then we also need to take the remaining warp time (real time) of the selected context into account. So we get something like this `min(warp_time_left, min(diff_top, diff_bottom)*weight + C)`.

skalk, 2024-08-05:
@jschlatow: I see, thank you for your prompt answer!

skalk, 2025-06-12:
@jschlatow thank you for the fixup commits. They are very much appreciated. I'll apply them, as well as fix the unit test and merge everything together.

In my eyes the only thing left is: having a heuristic in the priority scaling of base-hw that at best matches Sculpt OS workloads for now. At least this is what @nfeske was asking for in a coffee round discussion. When we have that in place, in my eyes it is time to put everything on staging. Or are there other opinions?

jschlatow, 2025-06-20:
@skalk I have not finished parameter tuning, yet. Nevertheless, I would like to share my thoughts and state of evaluation. I tried to document a few aspects with 0089df0. It includes formulas for calculating an upper bound on scheduling latencies, which can be taken as a basis for adjusting warp times and weights.

I have also noticed that we could actually benefit from implementing a warp time limit mechanism. If a group has a CPU burner context and another context in the same group gets ready, the latter might need to wait for the group's virtual time to fall below the scheduler virtual time. This renders the group's warp value ineffective. Basically, the CPU burner used up all the group's CPU quantum, which negatively affects other contexts in the same group. Warp-limitting would change this by running the group without its warp value whenever the CPU burner context is scheduled.

The formulas from 0089df0 actually provide a quantification for this. With the presence of a CPU burner context, another context in the group `i` would experience an additional `MIN_SCHEDULE_US * w_j/w_i` from every group `j`. Where `w_x` denote the groups' weights. Thus, as soon as we use significantly different weight values, this can become an issue.

jschlatow, 2025-06-20:
@skalk I made a surprising observation while experimenting with load on different priorities/groups. The setup is as follows: I have a CPU burner on BACKGROUND priority and a single pair of ping-pong threads on DRIVER priority. The "ping" thread sends a signal to the "pong" thread and waits for a return signal. The pong thread does 2000 page-sized memcpy operations to consume some time before it sends a return signal. After receiving the return signal, the "ping" thread sleeps for 10ms. All groups have weight 1 and warp 0, except the DRIVER group, which got warp 20000. The warp value is chosen such that the ping-pong threads should always be prefered over the background burner.

The roundtrip latency measured by the "ping" thread is ~1ms larger if the background burner is present compared to the same setup without the background burner. I instrumented the scheduler to log situations whenever the BACKGROUND group is scheduled while ready contexts are present on the DRIVER group. As expected, this never occurs. However, I am still wondering why/how the background burner interferes with the load on the DRIVER priority. Do you have any idea?

jschlatow, 2025-07-10:
> > [@skalk](https://github.com/skalk) I have not finished parameter tuning, yet. Nevertheless, I would like to share my thoughts and state of evaluation. I tried to document a few aspects with [0089df0](https://github.com/genodelabs/genode/commit/0089df012df1adf280147084157b3ef4809be68d). It includes formulas for calculating an upper bound on scheduling latencies, which can be taken as a basis for adjusting warp times and weights.
> > I have also noticed that we could actually benefit from implementing a warp time limit mechanism. If a group has a CPU burner context and another context in the same group gets ready, the latter might need to wait for the group's virtual time to fall below the scheduler virtual time. This renders the group's warp value ineffective. Basically, the CPU burner used up all the group's CPU quantum, which negatively affects other contexts in the same group. Warp-limitting would change this by running the group without its warp value whenever the CPU burner context is scheduled.
> > The formulas from [0089df0](https://github.com/genodelabs/genode/commit/0089df012df1adf280147084157b3ef4809be68d) actually provide a quantification for this. With the presence of a CPU burner context, another context in the group `i` would experience an additional `MIN_SCHEDULE_US * w_j/w_i` from every group `j`. Where `w_x` denote the groups' weights. Thus, as soon as we use significantly different weight values, this can become an issue.
> 
> To me it seems unclear how to differentiate in between several contexts of one group and a group's warping facility that is enabled/disabled accordingly, e.g. no warping in case of the cpu_burner after its limit is reached, but warping for the other new context. Or do you mean every context has its separate warp valuei/limit? Wouldn't this violate the 2-level scheduling completely, and effect the fairness of the scheduling algorithm? Maybe, you can explain this in more details to me (e.g., in an offline discussion).

Commit 5132040 implements the warp limitting and adapts the scheduling parameters. Note that this does not affect the fairness of the scheduling algorithm. The main idea was that by warping, a thread gets a part of its future CPU share a bit earlier in exhange of a longer pause after that. In the original BVT paper, threads were allowed to request warping arbitrarily. Letting the warp state of a group depend on the state of its contexts does therefore not affect fairness.

jschlatow, 2025-07-31:
@skalk I just cleaned and rebased my _bvt_ branch on staging.

nfeske, 2025-08-05:
What a beautiful result! Thanks @skalk and @jschlatow for the magnificent work. Merged to staging with great pleasure.

jschlatow, 2025-08-07:
@skalk That is quite unexpected. The intention was to set the timeout to the maximum value supported by the timer when there is no other context ready. I believe, however, the helping mechanims could explain this.

skalk, 2025-08-07:
@jschlatow: you are right. It is due to helping from the main thread calling initially the entrypoint that executes Component::construct. Inside that function the measuring takes place. With the new scheduler I see approx. every 4 ms a scheduling decision when both contexts are ready here, whereby the old scheduler took 10 ms. What do you think about increasing the "minimum schedule timeout" from currently 2 ms to 5 ms? Thereby, we would get the old behavior in case of concurrent threads within the same group. Or do you see reasons for not increasing it?

commit b41b8dc1 (2025-08-05): hw: implement new scheduler scheme

Implements two-level scheduling scheme with groups of contexts
sharing a virtual time, a weight (virtual time factor), and warp
value (virtual time shift / latency boost).
Within a group each context holds its own virtual time.

On the top-level, the group with the lowest virtual time gets
selected. Within a group the context with the lowest virtual time
gets selected. Whenever the state of the scheduler needs to get
updated (timer interrupt, context gets ready/unready, or yield
gets called) the virtual time of the current context and group
gets increased, d [... cut]
commit 4d47aac6 (2025-08-27): hw: implement new scheduler scheme

Implements two-level scheduling scheme with groups of contexts
sharing a virtual time, a weight (virtual time factor), and warp
value (virtual time shift / latency boost).
Within a group each context holds its own virtual time.

On the top-level, the group with the lowest virtual time gets
selected. Within a group the context with the lowest virtual time
gets selected. Whenever the state of the scheduler needs to get
updated (timer interrupt, context gets ready/unready, or yield
gets called) the virtual time of the current context and group
gets increased, d [... cut]
closed 2025-08-28 by chelmuth