genodelabs/genode #5119: monitor: limit the 'm' command response size

cproc, 2024-02-19:
Currently, the 'm' command response can exceed the `terminal_crosslink` buffer.

cproc, 2024-02-19:
Fixed by df27b31

nfeske, 2024-02-21:
I think this is an adequate fix for the immediate problem of blocking on the terminal crosslink. So I'd like to merge it because is solves an issue. But I also suggest to pursue the following two subsequent steps towards a more sustainable/flexible solution:

* Let's make the GDB response size configurable instead of using a hard-coded value. Both the original value of 16K as well as the new value of 1K are mere heuristics. I'm a little afraid that the change to 1K may butcher the performance of scenarios where memory is dumped in bulk. By making the value configurable, both the use of GDB as well as memory-inspection scenarios could be satisfied best.
* Let's make the buffer size of the terminal-crosslink configurable. There is really no good reason why it should be limited to 4K.

nfeske, 2024-02-21:
Merged to staging, including the minor cosmetic fixup https://github.com/genodelabs/genode/commit/fceeba0d9292e7de478c74c89f1cda76ade7862f (@cproc could you check?).

closed 2024-02-29 by chelmuth

cproc, 2024-02-29:
> * Let's make the buffer size of the terminal-crosslink configurable. There is really no good reason why it should be limited to 4K.

Would it make sense to pass the receive buffer size as terminal session argument and to use the session quota for the receive buffer?

nfeske, 2024-02-29:
I would keep the buffering aspect of the terminal-crosslink separate from the communication buffer of the terminal session. So the pipe buffer size would be defined by the terminal-crosslink configuration and paid-for by the terminal crosslink.

Down the road, I can (faintly) envision that your suggestion of letting the client pay might become useful, but not until the terminal crosslink becomes a dynamically used resource multiplexer with more than two clients. But for now,  better let us keep things simple.

cproc, 2024-02-29:
The `terminal_crosslink` currently uses a `Ring_buffer` as pipe buffer, which gets its size as template parameter. Would a new `Ring_buffer` variant that gets an allocator and a size as constructor arguments be useful in general or should it be kept in the `terminal_crosslink`?

commit d08532db (2024-02-21): monitor: limit the 'm' command response size

Fixes #5119
commit ea51f1ff (2024-02-28): monitor: limit the 'm' command response size

Fixes #5119