genodelabs/genode #5123: trace_recorder: get session arguments from config

cnuke, 2024-02-20:
The following commit adds support for setting the `TRACE` session arguments via the configuration. This is useful for scenarios where the default settings are not sufficient. The component now also displays a warning in case the number of received subjects equals the limit imposed by the argument buffer size to hint at potentially truncated results.

cnuke, 2024-02-20:
Commit [98b4a44](https://github.com/cnuke/genode/commit/98b4a447378aa7f6cbf934e7c4e8e38aef5a106f) addresses this issue.

chelmuth, 2024-03-01:
I'd like to merge these commits but think the renaming went into the wrong direction. Without your patch trace_recorder and trace_logger used consistent naming of `buffer` resp. `default_buffer` attributes for buffer sizes. I suggest we stick to this naming for `default_buffer` (was `buffer_size`) and `session_arg_buffer` which unifies these aspects of the config of both components. You may squash the final commit, thanks.

cnuke, 2024-03-04:
My bad, commit [449a72d](https://github.com/cnuke/genode/commit/449a72d8a227bcbee516b3c60a49a754baec601c) contains the proper adaption.

chelmuth, 2024-03-04:
Thanks, merged to staging.

commit 449a72d8 (2024-03-04): trace_recorder: get session arguments from config

This commit adds support for setting the TRACE session parameters via
the configuration.

Fixes #5123.
closed 2024-04-12 by chelmuth