genodelabs/genode #5128: hw: add support for VMX

atopia, 2024-02-23:
Add support for Intel's Virtual Machine Extensions with nested paging.

atopia, 2024-02-23:
Works on nested virtualization under Linux/KVM with qemu:
- vmm_x86
- seoul-auto (1 vCPU)
- seoul-net
- seoul-genode
- seoul-disc
- seoul-disc-64

Does not work:
- SMP (AP initialization hangs)
- virtualbox6 (error)

atopia, 2024-05-08:
@chelmuth please have a look. The VMX virtualization feature contained in this issue is in good shape. There is an unresolved issue with (timer) interrupt injection, possibly connected to "recall" exit injection in the VMM on hw that likely transcends the VMX hypervisor implementation contained in this issue. On top of these commits, practical use of VMX under hw requires a reworked nested page table implementation that is orthogonal to VMX and will be provided in an upcoming issue.

chelmuth, 2024-05-13:
I merged cba426c139f1dcf2eb5752e540a15d3ad8fc663d but got merge conflicts with 12e87b11a3ae03953e6ec86bf6099fab1c8e190d. Please address my remarks and rebase on staging.

atopia, 2024-05-14:
The new commit should address the previously discussed points and is meant to be added on top of staging with the fixup commits from #5113 applied.

chelmuth, 2024-05-15:
Merged to staging.

commit 915cef2f (2024-05-15): hw: add support for VMX

Add support for Intel's Virtual Machine Extensions with nested paging.

Fixes #5128
commit aad80e81 (2024-05-29): hw: add support for VMX

Add support for Intel's Virtual Machine Extensions with nested paging.

Fixes #5128
closed 2024-05-30 by chelmuth