genodelabs/genode #5129: usb_net: "schedule: unexpected preempt_count"

ssumpf, 2024-02-27:
When testing with `iperf3` using RTL8156 we observe a "schedule: unexpected preempt_count" message leading to `abort` being called. This behavior is triggered when trying to `schedule` while holding a lock or mutex.

Reported by @alex-ab

ssumpf, 2024-02-27:
@alex-ab: d1e4352 does resolve the problem for me. How this goes together with https://github.com/genodelabs/genode/issues/5021 I don't know. It might not be an issue there.

alex-ab, 2024-02-27:
@ssumpf: test scenario works now as expected, nice, thx

skalk, 2024-03-26:
The issue stems from the fact that the `submit_urb` emulation code for USB client drivers was blocking when the USB packet stream was saturated. However, flow control is apparently done at another (upper) layer, and we cannot block here, but have to accept the URB. With the USB rework of issue #5021 the `submit_urb` emulation code got removed completely, and we use more original contrib code here, and do not block anymore at this layer. In my eyes, we can therefore close this issue. @ssumpf @alex-ab what do you think?

alex-ab, 2024-03-26:
I re-tested my scenario and the specific issue does not trigger anymore. However, when using the usb_net, in the test (iperf) starts with high cpu utilization (which is expected), but after while the utilization drops to 0 and the final throughput of the iperf scenario is not good. But this is another issue. So, from my side this issue can be closed.

chelmuth, 2024-03-26:
So shall it be.

closed 2024-03-26 by chelmuth