genodelabs/genode #5131: libc: select() based on poll()

cproc, 2024-03-01:
Measurements with `oprofile` for the QtWebEngine-based web browsers showed a relatively high percentage of time spent in `select()`-related functions, in particular the lookup of file descriptor objects. When I re-implemented `poll()` and `select()` in a way that `select()` uses `poll()` internally instead of the other way around and `poll()` looks up the file descriptor objects only once early-on, the corresponding `oprofile` percentages were reduced notably. The re-implementation also removed the global `Select_cb_list` and the dedicated mutex used to protect it and keeps more information local on the stacks of the individual threads instead, which might lower the risk of synchronization problems within the libc.

cproc, 2024-03-01:
Commit 34f8689 contains the re-implementation. It has been in use by the Falkon and Morph browser depot packages in the `cproc` depot for months now and appears to work well so far, at least in these scenarios.

SunriseJungle, 2024-03-02:
Hello, I hope you’re able and willing to answer my question and I’m sorry if I’m not supposed to respond to this email….. But I am wondering if Genode Will now function, most computers, generally, or if it will only work on certain computers still. I would like to try it out On my pine phone Pro Explorer or just a laptop.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

cproc, 2024-03-05:
@SunriseJungle: the mailing list [1] would be a better place for generic questions about Genode. Regarding supported hardware, there is an unofficial hardware compatibility list at [2] and some additional hardware not mentioned there like the original PinePhone or the ARM-based MNT Reform laptop. There is also a ready-to-use Sculpt OS image available which you could try on your laptop [3].

[1] https://genode.org/community/mailing-lists
[2] https://usr.sysret.de/jws/genode/hcl.html
[3] https://genode.org/download/sculpt

commit 62b3f24d (2024-04-12): libc: select() based on poll()

Fixes #5131
closed 2024-04-12 by chelmuth