genodelabs/genode #5137: monitor: make the maximum GDB response size configurable

cproc, 2024-03-07:
Follow-up of issue #5119.

cproc, 2024-03-07:
Fixed by 12e0795. Commit 78ad464 increases the buffer size of the monitor test again.

nfeske, 2024-03-07:
The `Number_of_bytes` type is intended for parsing purposes. It is not well suited for passing values around or for keeping state. Could you please replace the use of the `Number_of_bytes` argument types by a type as follows?

```
struct Max_response { size_t num_bytes; };
```

This will arguably make the code easier to read (because the purpose of the argument is clarified by its type name) and it avoids the `static_cast` from `Number_of_bytes`.

cproc, 2024-03-07:
Updated commit cfd9dcb uses `Max_response` instead of `Number_of_bytes`.

nfeske, 2024-03-07:
Would it be possible to host `Max_response` in `State` and leave the `execute` signature as is?

nfeske, 2024-03-08:
Thanks, that's much better. I merged your latest version to staging.

cproc, 2024-03-11:
Fixup commit f8d7b06 fixes the case where no `<monitor>` config node exists.

nfeske, 2024-03-11:
In the light of the recent conversation https://github.com/cnuke/genode/commit/98b4a447378aa7f6cbf934e7c4e8e38aef5a106f#r138878041, we should better remove the `_size` suffix from the attribute name. So `max_response_size` would become `max_response`. This also makes the naming consistent with init's existing `<report buffer="..."/>` argument. @cproc may you create fixup for this?

cproc, 2024-03-11:
Commits 5c2713b and 60932e8 remove the `_size` suffix.

commit 5b328c74 (2024-03-08): monitor: make maximum GDB response size configurable

Fixes #5137
commit d1a750c5 (2024-04-12): monitor: make maximum GDB response size configurable

Fixes #5137
closed 2024-04-12 by chelmuth