genodelabs/genode #5140: rump: avoid busyness when idle

nfeske, 2024-03-11:
During my investigations for #5138 when testing on Sculpt OS, I noticed that ahci-fs induced load on the timer for no good reason. Taking a closer look, I ended up at the rump kernel's `rumpuser_clock_sleep`, which is steadily called for 0 to 10 ms. Each call produces a sequence of `Timer::Session::trigger_once` calls: small ms value for the sleep, once triggered, reprogram to 0.5 s for the time interpolation, followed by another small ms value. This translates to constant load on an otherwise idle Sculpt system.

nfeske, 2024-03-11:
Commit c4d7535 enforces a sleep time of at least one second. After a few minutes of casual testing, ahci-fs on Sculpt seems to behave as before. But the idle load on my x250 gets significantly reduced (from 1% to 0.5%).

@ssumpf and @cnuke do you approve this sledgehammer change?

cnuke, 2024-03-11:
@nfeske I am fine with this change :+1:.

ssumpf, 2024-03-11:
@nfeske: Good find. I am also fine with it.

nfeske, 2024-03-11:
Thanks for quick response. I merged the commit to staging.

nfeske, 2024-04-12:
Merged to master.

closed 2024-04-12 by nfeske