genodelabs/genode #5148: tresor: remove module framework

m-stein, 2024-03-18:
In the context of this issue, several suggestions for improving the result of the last Tresor clean-up (https://github.com/genodelabs/genode/issues/5062) are applied. The main points are:
* Remove the use of the module framework for communication between modules,
* drive requests explicitly as first argument of the overloaded execute method of its module,
* make a request member of the object that requires it and let this object also drive the request,
* use a dedicated class for each request type,
* merge the state required for executing a requested functionality into the request type (removes channel classes),
* provide access to re-used resources (like VFS file handles) through the arguments of the execute methods only (management of those resources moves to the lib user, synchronization of access to them remains in the modules),
* remove the scheduler module (Request_pool) and let the user of the Tresor lib manage synchronization of the top-level requests (don't use the module scheme for that aspect as it doesn't fit),
* remove the use of the NONCOPYABLE macro.

m-stein, 2024-03-22:
@chelmuth I tried to compress my merge_to_staging branch. If you opt for more compression, please let me know!

chelmuth, 2024-03-25:
The merge revealed (maybe 32-bit related) hiccups in the following depot packages.

- checker/bin/arm_v6/tresor
- checker/bin/arm_v6/file_vault
- checker/bin/x86_32/tresor
- checker/bin/x86_32/file_vault

The builder confirms this with errors on arm_v6, arm_v7a, and x86_32.

chelmuth, 2024-03-25:
run/tresor_tester fails on all platforms but linux.

```
[init] child "block_io_fs" requests resources: ram_quota=2101248
```

x86_64.pc.nova.file_vault_client
```
Line 9 of output is unexpected
expected: '[init -> dynamic_init -> log_terminal] Und Sonderzeichen: /§($)=%!'
got:      '[init -> dynamic_init -> log_terminal] Und Sonderzeichen: /..($)=%!'
Error: Test failed, 1 unexpected lines of output
```

arm_v8a.imx8q_evk.hw.file_vault_client
```
Line 3 of output is unexpected
expected: '[init -> dynamic_init -> log_terminal] -rwx------ 1 root 0 18 Jan  1  1970 file_1'
got:      '[init -> dynamic_init -> log_terminal] -rw------- 1 root 0 18 Jan  1  1970 file_1'

Line 9 of output is unexpected
```

m-stein, 2024-03-25:
@chelmuth I'm at it. Thanks for the pointers!

m-stein, 2024-03-26:
@chelmuth I started with re-enabling support for x86_32 (testing with nova) and found it sensible to introduce a new type ``Number_of_disk_bytes``, because many places that used to use addr_t, size_t or something related actually referred to addresses, sizes, etc. on either the "virtual" (virtual block device) or "physical" (block back-end) block device. I think that these address spaces should be assumed to have 64-bit addresses independently from the target platform. Here's the commit that could be merged already: 7bac2ffafb.

I will continue with the ARM platforms.

m-stein, 2024-03-26:
@chelmuth My arm_v7a and arm_v6 tests succeeded as well with the above commit in place.

m-stein, 2024-03-27:
@chelmuth I couldn't reproduce the nova error in qemu. I can investigate this further in office if you opt for it or we just remove the apparently troublesome § character for now: 24e1e1615f

m-stein, 2024-03-27:
@chelmuth This should fix the file-mode issue on arm_v8: 041ab6320f

chelmuth, 2024-03-27:
> @chelmuth I started with re-enabling support for x86_32 (testing with nova) and found it sensible to introduce a new type `Number_of_disk_bytes`, because many places that used to use addr_t, size_t [...]

Currently, I'd appreciate just a fix of the problem at hand: Don't use addr_t and size_t where uint64_t should be used. Your intention to clean up and tighten the implementation by means of stronger typing is well-meant but your implementation as plain typedef to uint64_t does not address your intention. Therefore please, let's only fix the build errors for now. We may discuss clean-up options offline in the next meeting.

m-stein, 2024-03-27:
@chelmuth Thanks for your feedback! This is a less invasive approach:
47cb1cd813 run/file_vault_client: consider varying file modes
2fcb541960 run/file_vault_client: remove troublesome § char
ab50b9def5 file_vault: re-enable support for 32-bit platforms

chelmuth, 2024-03-28:
Merged 337e0ecae857f1a4cc9a4fe88b786ce8bdc5de1d, 583e5bee0589fc3c1dce5754c7dd5ff98fc27ef7, and 8adb297903097132dc787c63eec601fc0d3b155d as fixup. Do you approve?

m-stein, 2024-03-28:
@chelmuth Yes, thank you!

m-stein, 2024-04-01:
@chelmuth With these commits, tresor_tester should succeed on all platforms and file_vault_client should fail on less platforms:
6f0b211612 file_vault: raise child resources
e7fd3f1d58 file_vault_client.run: raise access timeout
77490c9312 file_vault_client.run: disable for riscv
02610d305c tresor_tester.run: raise block_io_fs caps
74e87de503 tresor_tester.run: raise test timeout

There are some fails with file_vault_client that I couldn't reproduce yet (for instance qemu/x86_32/fiasco). I assumed that it is related to the used qemu version (I use 6.1). However, I couldn't figure out which version is used in the nightly tests because my login failed and therefore tried some older qemu versions (4,4.2,5,5.2) but without a result.

m-stein, 2024-04-02:
@chelmuth Ok, it seems, that the remaining fails where also timeout-related and I just lost myself in the logs. I noticed that testing with qemu on the test machine is significantly slower than on my machine, presumably because of ´´-no-kvm´´.

cnuke, 2024-04-03:
Commit [bb7d61d](https://github.com/cnuke/genode/commit/bb7d61d7fa1226fb870b6dbe259418f057814c12) changes the default permissions to `0777` (cover +x as it is not possible to set it otherwise) to prevent regressions, which should make [file_vault_client.run: fix expected file modes](https://github.com/genodelabs/genode/commit/583e5bee0589fc3c1dce5754c7dd5ff98fc27ef7) dispensable.

nfeske, 2024-04-03:
Thank you for the fixup. I merged it to staging and reverted the file_vault_client.run commit.

m-stein, 2024-04-08:
@chelmuth This commit should fix most of the remaining fails with file_vault_client:
14ecd4db34 file_vault_client.run: raise lock timeout and caps

closed 2024-04-12 by chelmuth