genodelabs/genode #5150: Consolidation of the driver manager into the sculpt manager

nfeske, 2024-03-18:
To make the driver management of Sculpt more flexible, runtime-dependent driver components should better move from the static drivers subsystem to the dynamic runtime subsystem. This brings the following benefits:

- The sculpt manager can dynamically expand the driver's resource quotas. E.g., the RAM needed by Intel FB when encountering large resolutions can be granted by the Sculpt manager by responding to resource requests.
- Drivers like Intel FB can be removed and restarted during a suspend-resume cycle.
- Drivers appear as nodes in the component graph, which opens up the potential of hosting driver-specific settings in the node dialogs.
- USB device management becomes simpler because the host-driver configuration would no longer require an interplay of the Sculpt manager with the static drivers subsystem. Instead, the USB config could follow the same pattern as other configurations. That is, config/usb contains the user-dictated rules whereas config/managed/usb contains the effective configuration (generated by the Sculpt manager, consumed by the USB host driver) that includes the supplements of the Sculpt manager.

nfeske, 2024-03-18:
Commits https://github.com/genodelabs/genode/commit/9d19410f12e3e4b851e46d56c30749bf84598c6e and https://github.com/genodelabs/genode/commit/1b1545794364420ac3dbd35d1c901a98d127943e implement the first step, covering the PC framebuffer drivers.

nfeske, 2024-03-27:
I wrapped up this line of work today. Note that I merged the commit series to staging to let our CI chew on it. It goes without saying that I'll continue with thoroughly testing the new version.

chelmuth, 2024-03-28:
Great, I'm looking forward to use this work.

I have a related question: During testing with modular Sculpt I usually cherry-pick 73862c4 to enable the Sculpt partition in RAM by default for benefit from a full-fledged depot and the *prepare* mechanics (e.g., a *config/\<VERSION\>* directory). Will this still be possible with your changes or may it even be worthwhile to add a config option for this with 24.04?

nfeske, 2024-03-28:
Good idea -> https://github.com/genodelabs/genode/issues/5166

nfeske, 2024-03-28:
Fixup 3907a9890702795382023a20280f7cc149d30a5b addresses a partition-discovery glitch I noticed today.

nfeske, 2024-04-12:
The change entered the master branch.

closed 2024-04-12 by nfeske