genodelabs/genode #5151: pc_nic_drv: high cpu load under high network load

alex-ab, 2024-03-19:
In #5149  the cpu load of the ipxe driver could be reduced, in order to achieve more stable results under high network load  in vbox #5146. The same optimization of ipxe are already part of the ported pc_nic_drv, but nevertheless the pc_nic_drv causes high CPU load. It turns out that the used backend allocators seems to be the main culprit. To show the effect, a simple cache was set up which shows the effect:

The iperf server runs outside Genode, the iperf client inside a vbox VM , ( ```iperf -c IP -t 60``` resp. ```iperf -c IP -t 60 -R```)

| x | pc_nic_drv (0,0) | nic_router (1,0) | vbox (2,0) + (2,1) |
| --- | --- | --- | --- |
| iperf, w/o cache | ~66%| ~14%| ~60%, ~931 MBit/s |
| iperf, simple cache | ~25%| ~14%| ~45%, ~962 MBit/s |
| iperf -R, w/o cache | ~75%| ~16%| ~90%, ~851 MBit/s |
| iperf -R, simple cache | ~45%| ~16% | ~90%, ~827 MBit/s |

Applying the simple cache to usb_net reduces for this driver the CPU load by approximately 10-15%.

alex-ab, 2024-06-26:
I was requested by @chelmuth to redo the measurements with the work-in-progress state of #5236. I added the SLUB version to the table below. The overall observation is that the CPU load in the pc_nic_driver drops during the measurement to the same level as with the simple cache version.

| U7411 | pc_nic_drv (0,0) | nic_router (1,0) | vbox (2,0) + (2,1) |
| --- | --- | --- | --- |
| iperf, w/o cache | ~66%| ~14%| ~60%, ~931 MBit/s |
| iperf, simple cache | ~25%| ~14%| ~45%, ~962 MBit/s |
| iperf, SLUB #5236 | ~28%| ~14%| ~45%, ~930 MBit/s |
| iperf -R, w/o cache | ~75%| ~16%| ~90%, ~851 MBit/s |
| iperf -R, simple cache | ~45%| ~16% | ~90%, ~827 MBit/s |
| iperf -R, SLUB #5236 | ~40%| ~14%| ~90%, ~850 MBit/s |

| S938 | pc_nic_drv (0,0) | nic_router (1,0) | vbox (2,0) + (2,1) |
| --- | --- | --- | --- |
| iperf, w/o cache | ~90%| ~20%| ~60%, ~864 MBit/s |
| iperf, simple cache | ~45%| ~20%| ~90%, ~914 MBit/s |
| iperf, SLUB #5236 | ~55%| ~21%| ~65%, ~929 MBit/s |
| iperf -R, w/o cache | ~85%| ~20%| > 100%, ~577 MBit/s |
| iperf -R, simple cache | ~65%| ~20% | > 100%, ~832 MBit/s |
| iperf -R, SLUB #5236 | ~60%| ~22% | > 100%, ~775 MBit/s |

nfeske, 2024-06-26:
Thanks @alex-ab for the excellent news!

nfeske, 2024-08-29:
Fixed in Genode 24.08.

closed 2024-08-29 by nfeske