genodelabs/genode #5152: lighttpd fails to deliver files that apear in file system after its createion

trimpim, 2024-03-19:
If `lighttpd` is setup to deliver files from an externally provided file system, files that appear after the creation of the 'vfs1 node fail to be delivered with the following error log:

```
[init -> lighttpd] 2000-01-01 00:00:23: (/data/genode/depot/trimpim/src/lighttpd/2024-03-19T1035/src/app/lighttpd/src/network_write.c.179) read: Resource temporarily unavailable
[init -> lighttpd] 2000-01-01 00:00:23: (/data/genode/depot/trimpim/src/lighttpd/2024-03-19T1035/src/app/lighttpd/src/connections.c.325) connection closed: write failed on fd 33
```

As long as the file hasn't been created an error 404 is reported as expected.

I have modified the `lighttpd` run script to demonstrate the error. I was able to trace the error back to the commit 914508bf7ac6f66c9709f4fb32ade6ab4d5d0633.

I'm currently tying to figure out what is going wrong.

cnuke, 2024-03-19:
@trimpim please cherry-pick 83afc6e438d1c44c3e55da4d8027ad967774777a from staging and try if that fixes your problem,

trimpim, 2024-03-19:
@cnuke Thanks for the quick response, indeed this fixes the issue.

trimpim, 2024-03-19:
Closing issue as it is already fixed in latest staging.

closed 2024-03-19 by trimpim