genodelabs/genode #5157: black_hole: add Play and Record support

ttcoder, 2024-03-20:
I've started migrating from mixer to record_play_mixer before the former gets retired, and have some tentative code to test.

My testing typically starts with a very basic scenario, running in Qemu -- hence the usage of black_hole. In fact even in more advanced (bare metal) scenarios, I tend to redirect Audio-in to black_hole even if Audio-in would, in fact, be available, as I only want to enable Audio-Out.

However, I just realized that black_hole was not yet migrated, it only provides the <audio_in> and <audio_out> nodes but not `<play>` and `<record>` yet.

(BTW, black_hole silently fails, it does not report a warning like "unknown service type: `<play>`" or such, it silently ignores the unknown tag -- not sure whether that deserves a separate ticket ?)

nfeske, 2024-03-21:
This is a good idea.

As an interim solution, you could try using the `record_play_mixer` with an empty `<config/>` as a black hole for the record and play sessions.

> it silently ignores the unknown tag

This is the case for many components. The components don't even try to validate XML schemas but merely pick up the information that interests them. However, configs can be validated offline using the accompanied xsd files. For the back hole, you can find it at https://github.com/genodelabs/genode/blob/master/repos/os/src/server/black_hole/config.xsd

In some cases, we deliberately use the fact that XML nodes are allowed to contain superfluous content. For example, the monitor component consumes init configurations but parses additional monitor-specific supplements that are not known/supported by init. A monitor configuration (with these supplements) can directly be passed to the sandbox library (the actual init code) with no filtering needed.

In other cases, where configs are generated, the nodes can be augmented with additional meta data for diagnostic purposes, which can also be quite handy. If components would reject such configs, we'd loose this flexibility.

As a further consideration, the complexity of runtime XML schema validation along with the need for a formalism of supplementing schema definitions to the executable binaries seems an excessive price to pay for the gain of a little more diagnostic assistance. With offline validation, we can nicely side step these complexities.

cnuke, 2024-04-03:
Commit [6918f10](https://github.com/cnuke/genode/commit/6918f1081b85d88fdf61206e3c3c015ac7dabe6d) adds the dummy implementation for both sessions to the `black_hole` component.

nfeske, 2024-04-03:
Thank you @cnuke. Merged to staging.

commit 0a39c494 (2024-04-03): black_hole: add Play and Record session

This commit introduces support for the Play and Record session to
the black hole component in the most minimal way possible.

Fixes #5157.
closed 2024-04-12 by chelmuth