genodelabs/genode #5164: dde_linux: improve handling of interrupt masking and acknowledgement

chelmuth, 2024-03-25:
While debugging pc_nic_drv link handling, I discovered that unmasked pending interrupts are sometimes not delivered immediately. This was caused by sloppy task execution that is fixed by 1ea641a5311167a150719bbdbbbede42dd3ed2b1 paired with shaky handling of interrupt masking and acknowledgement. I addressed the latter in a work-in-progress commit on x86/pc, but then discovered that ARM comes with a slightly different implementation of irqchip.c regarding EOI ans ACK.

May we merge the both implementations?

If not, could someone check if arm/irqchip.c contradicts my work in progress?

chelmuth, 2024-03-25:
Commit fb00f4ab4802d18baa189ff124c9d6ffb96c9b20 reflects my WIP.

skalk, 2024-03-26:
I would very much appreciate, if we can merge both irqchip variants (x86, arm). They are in my eyes not so much architecture dependent, it's more a question of configuration. The arm variant reflects the different needs already a bit by the `CONFIG_IRQ_DOMAIN_HIERARCHY` optional part. If we put several parts into a `CONFIG_OF` optional part too, we would gain an almost identical version.
The EOI when doing unmask in the arm version might be located there for no particular reason. The development history was such that x86 came later, and EOI got placed differently. I would think we can check whether it works in the ARM case the same way.
Another function provided in the arm case is `.irq_set_wake`, which seems to have no value regarding functionality, I think we can simply remove it. The function `.irq_set_type` (only in arm) might be necessary, but probably it does not disturb the x86 scenarios.

chelmuth, 2024-07-29:
Commit b51ea01a36ba9121b72a9adcc76867fb6bc59f00 on staging includes the fixes but also addresses further concerns of our discussion. @skalk please have look especially at how I addressed `CONFIG_OF` in *irqchip.c*.

skalk, 2024-07-29:
I've reviewed it, and I'm fine with it, like we discussed right away offline.

commit b51ea01a (2024-07-29): dde_linux: improve handling of IRQ masking and ack

Unmasking of a pending interrupt did not lead to immediate IRQ handler
execution in all cases.

This commit also addresses some style concerns risen during the issue
discussion.

- Replace multi-boolean IRQ state by state enum
- EOI and ACK should be same in DDE context
- Unify x86 and ARM irqchip.c
- Remove Pending_irq type
- Remove dde_irq_set_wake()

Fixes #5164
commit 044d8bca (2024-08-27): dde_linux: improve handling of IRQ masking and ack

Unmasking of a pending interrupt did not lead to immediate IRQ handler
execution in all cases.

This commit also addresses some style concerns risen during the issue
discussion.

- Replace multi-boolean IRQ state by state enum
- EOI and ACK should be same in DDE context
- Unify x86 and ARM irqchip.c
- Remove Pending_irq type
- Remove dde_irq_set_wake()

Fixes #5164
closed 2024-08-29 by chelmuth