genodelabs/genode #5167: New VFS OSS plugin using Play/Record sessions

cnuke, 2024-03-28:
This issue is about integrating the new `Play` and `Record` sessions.

The new VFS plugin uses the `Play` and `Record` session and for now is a drop-in replacement for the old plugin employing `Audio_out` and `Audio_in`. For the time being, however, it is constraint by the same limitations (format, sample rate and available channels), which will be lifted step by step eventually.

cnuke, 2024-03-28:
The commit series above makes the new plugin available. Existing users should just have to change the services routes in their runtimes.

I have tested SDL2 applications, cmus, VBox and Falkon (non-jemalloc variant) with good results.

mewmew, 2024-03-31:
Really wonderful to see this development! Great work @cnuke.

Both Daniel and I are excited to take this out for a spin : )

nfeske, 2024-04-02:
Thanks @cnuke for the excellent commit series, which I just merged to staging.

ttcoder, 2024-04-03:
I'm going to try this out ! Couldn't get "native" Play to work well so far (the audio is barely recognizable, as if mistaking little endian for big endian or something, couldn't find what I'm doing wrong for the past couple weeks). This plug-in might save my bacon.

mewmew, 2024-04-03:
> I'm going to try this out ! Couldn't get "native" Play to work well so far (the audio is barely recognizable, as if mistaking little endian for big endian or something, couldn't find what I'm doing wrong for the past couple weeks). This plug-in might save my bacon.

@ttcoder would you be willing to share your experiences if you get it working? It would be really great to see some more examples of how to get autio playback running on Genode.

Cheers,
Robin

mewmew, 2024-04-03:
> I have tested SDL2 applications, cmus, VBox and Falkon (non-jemalloc variant) with good results.

@cnuke, a quick question. Did you test also SDL2_mixer?

I've been trying to get it working on an older version of Sculpt, and haven't yet reached successful results :blush:

cnuke, 2024-04-03:
@mewmew So far I only tested the SDL2_mixer's `playmus` example (the commits are on my [world](https://github.com/cnuke/genode-world/commits/issue%235167-vfs_oss_record_play-2024-03-28/) branch) that exercises the mixer in a fairly simple fashion. Testing more applications (mainly the games in my goa-projects repo) that may make use of it is still on my todo-list.

> I've been trying to get it working on an older version of Sculpt, and haven't yet reached successful results 😊

What issues do you encountered in particular?

ttcoder, 2024-04-04:
I cherry-picked the files from c992740 in my 24.02 checkout, made sure the relevant part of libcache/ was empty, and built the .so, but it won't load : `LD: binary is not an elf.. exception .. Linker::incompatible.. Failed to create <oss> VFS node..` ; didn't checksum it but the stripped variant is 213312 bytes big.

Will try again from a fresh check-out of Genode 24.04/sculpt when it's released, probably not a good idea to mix and match different code bases.

mewmew, 2024-04-06:
> @mewmew You mean, the old mixer code ? I've been quite successful with the 'old' mixer, see here : https://chiselapp.com/user/ttcoder/repository/genode-haiku/dir?name=genode-haiku/haiku-on-genode/kits/media ; disregard all files except those that start with a "_" underscore, those are the ones containing Genode code (_ge_avdecoder uses the Genode "libav" package to read mp3 etc files, _ge_audioout_oss targets the legacy OSS plugin, and _bmediafile_temp "coordinates" the previous two). That stack has been rock solid for me (in 'native' mode at least).
> 
> However it might not be the best example to follow for "pure" Genode audio support, as it's encumbered with the need to provide compatibility : it converts one API into another API, it's basically a compatibility layer. The official Genode sample code (provided by cnuke esp.) is cleaner and much more compact than what mine provides.
> 
> Plus, much of that code will be retired once I manage to transition to the new mixer (record_play_mixer) and session API. Anyway feel free to get in touch (or write to the mailing-list) if I can provide more info, but basically I'm just your average struggling developer ^^

Awesome. Thanks @ttcoder for sharing your experiences! I appreciate it a lot!

Cheerful regards from a fellow struggling developer <3

cnuke, 2024-04-10:
Commit [2cccefc](https://github.com/cnuke/genode/commit/2cccefc71081b5db4161478f453e9a8e09065938) adjusts the missing parent-provides routes in the vbox6 pkgs.

nfeske, 2024-04-11:
@cnuke thanks a lot for this consideration. Given the variety of current topics, I appreciate it very much. :)

cnuke, 2024-04-11:
The commits

* ad44d591d9 - sculpt_distribution: replace mixer pkg
* 4a97820102 - dde_bsd: use 'mic_' prefix for Play sessions
* 86572f4b23 - fixup 'os: add pkg for record_player_mixer' (mic_left)
* 0f568e1442 - sculpt: add audio and mixer launcher
* f39875112e - fixup 'vbox6: switch pkgs to new VFS OSS plugin' (parent routes + frag sizes)
* 61c8deb727 - fixup 'libports: add test/oss recipes' (oss name)
* 5bd3640ed1 - fixup 'fixup 'gems: VFS OSS plugin for Record/Play session' (name)
* 5c62331f1f - fixup 'gems: VFS OSS plugin for Record/Play session' (halt when closing handle)

clean the existing commits up a bit and provide the examplary configuration (launcher) for Sculpt,
while the following commits introduce support for multiple sample-rates

* 17a549047c - Add plugin version to VFS OSS plugins
* 0217aad570 - vfs/oss: support different sample rates
* c6333f330f - libc: allow sample rate change via SNDCTL
* edb551beb3 - libc: print unknown fcntl command in hex

This features was tested only with cmus (playing .opus files) so far.

nfeske, 2024-04-11:
Excellent! I merged the commit series just now.

mewmew, 2024-04-12:
Just as a data point, with the recent commit series, audio playback using the record/play interfaces works well on x220, when running the `sdl2_playmus` example app.

Thanks for the incredible work @cnuke and the Genode team : )

closed 2024-04-12 by nfeske