genodelabs/genode #5173: sculpt_distribution-pc: error: GMP is missing or unusable (on staging)

mewmew, 2024-04-06:
Building `sculpt_distribution-pc` gives an error on staging (89fbc085fa69f4575f63a132ade202d7024ad7a4) but works just fine for git tag [24.02](https://github.com/genodelabs/genode/tree/24.02).

Command (on staging branch):
```bash
time ./tool/depot/create \
   UPDATE_VERSIONS=1 FORCE=1 REBUILD= \
   mewmew/pkg/x86_64/sculpt_distribution-pc
```

Output:
```
      [gdb]  mkdir -p -- ./gdbsupport
      [gdb]  Configuring in ./gdbsupport
      [gdb]  egrep: warning: egrep is obsolescent; using grep -E
      [gdb]  configure: WARNING: result yes guessed because of cross compilation
      [gdb]  configure: WARNING: linux/perf_event.h missing or too old; some features may be unavailable.
      [gdb]  configure: WARNING: libipt is missing or unusable; some features may be unavailable.
      [gdb]    CXX      agent.o
...
      [gdb]    CXX      event-pipe.o
      [gdb]    AR       libgdbsupport.a
      [gdb]  mkdir -p -- ./gdb
      [gdb]  Configuring in ./gdb
      [gdb]  configure: WARNING: using cross tools not prefixed with host triplet
      [gdb]  configure: WARNING: libdebuginfod is missing or unusable; some features may be unavailable.
      [gdb]  egrep: warning: egrep is obsolescent; using grep -E
      [gdb]  configure: WARNING: no enhanced curses library found; disabling TUI
      [gdb]  configure: WARNING: expat support disabled; some features may be unavailable.
      [gdb]  configure: error: GMP is missing or unusable
      [gdb]  make[8]: *** [Makefile:11447: configure-gdb] Error 1
      [gdb]  make[7]: *** [Makefile:1004: all] Error 2
make[6]: *** [/home/u/Desktop/genode_staging/depot/mewmew/api/noux/2023-06-15/mk/gnu_build.mk:190: built.tag] Error 2
make[5]: *** [var/libdeps:266: gdb.prg] Error 2
make[4]: *** [Makefile:351: gen_deps_and_build_targets] Error 2
make[3]: *** [/home/u/Desktop/genode_staging/tool/depot/mk/build_bin_archive:213: /home/u/Desktop/genode_staging/depot/mewmew/bin/x86_64/gdb_x86/2024-04-07.build/bin] Error 1
make[2]: *** [/home/u/Desktop/genode_staging/depot/var/build.mk:380: mewmew/bin/x86_64/gdb_x86/2024-04-07] Error 2
make[1]: *** [/home/u/Desktop/genode_staging/tool/depot/build:159: execute_generated_build_mk_file] Error 2
make: *** [tool/depot/create:59: build] Error 2
```

---

For reference:

Command successful on 24.02 tag:
```bash
$ time ./tool/depot/create \
   UPDATE_VERSIONS=1 FORCE=1 REBUILD= \
   mewmew/pkg/x86_64/sculpt_distribution-pc

created mewmew/api/aes_cbc_4k/2022-11-16
...
created mewmew/pkg/sculpt_distribution-pc/2024-03-31
```

mewmew, 2024-04-06:
Note: this is when using the 23.05 tool chain, and running the build on Arch Linux.

```bash
$ uname -a
Linux x1 6.8.2-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 28 Mar 2024 17:06:35 +0000 x86_64 GNU/Linux
```

cproc, 2024-04-09:
Commit 0a52c96 should fix the problem.

nfeske, 2024-04-10:
Thanks @cproc. I merged your fix to staging.

mewmew, 2024-04-10:
> Commit https://github.com/genodelabs/genode/commit/0a52c9668c844f13a657e47b65f79eb1bef28de0 should fix the problem.

I can confirm that the issue has now been fixed.

commit 82e6cf96 (2024-04-10): gdb: prevent use of host libgmp in configure test

Fixes #5173
closed 2024-04-12 by chelmuth