genodelabs/genode #5175: record_play_mixer: periodic warning after stop() called

alex-ab, 2024-04-10:
After temporarily calling stop(), the mixer seems still to operate the sessions and start complaining in the Sculpt log regulatory:

```
[init -> mixer] Warning: Play (seoul -> left) required sample value is no longer available
[init -> mixer] Warning: Play (seoul -> left) (jitter config or period too high?)
```

even so the client announced, that no new data will be enqueued.

Can/should the _stopped() method check also be applied to the warning case, as done for the later_than_avail_samples case,  or should the processing be stopped already earlier (saving CPU time) ?

https://github.com/genodelabs/genode/blob/master/repos/os/src/server/record_play_mixer/play_session.h#L327

```
if (probe_result == Probe_result::MISSING) {
	if (earlier_than_avail_samples) {
		warning("required sample value is no longer available");
		warning("(jitter config or period too high?)");
	}
	else if (later_than_avail_samples && !_stopped()) {
		warning("required sample is not yet available");
		warning("(increase 'jitter_ms' config attribute?)");
	}
}
```

nfeske, 2024-04-11:
Thanks for bringing this behavior to my attention. Your suggestion to apply the `_stopped` to both branches (moving it to the compounding `if` condition) is sensible. Even though the message is prefixes with the play session, it is triggered in the context of the record client's interaction with the mixer (when trying to obtain the sample data for mixing). When the condition of no more play data occurs, the depletion to reflected to the record client, which can use it to stop recording.

chelmuth, 2024-04-12:
Does 3af3773 fix this issue?

alex-ab, 2024-04-12:
@chelmuth: Yes.

chelmuth, 2024-04-19:
Fix merged to master.

closed 2024-04-19 by chelmuth