genodelabs/genode #5177: gpu/intel: support Intel HD Graphics 500

jschlatow, 2024-04-11:
The ZimaBlade (Celeron N3450) comes with IHD 500, which is basically a gen9 GPU. Linux handles the GPU as _Broxton_ and uses `GEN9_LP_FEATURES` (as opposed to `GEN9_FEATURES` that is used for IHD 520/530`) for it. I tried adding the GPU to the driver's config as normal Skylake/gen9 and, after fixing #5176, got mesa_gears and glmark2 working with that. Although glmark2 is able to complete, the log is flooded with `exec list is not empty`, which needs further investigation.

chelmuth, 2024-04-19:
@jschlatow the commits above were merged to master. Can we close this issue as fixed?

jschlatow, 2024-04-19:
Since I'm not planing to investigate on the "exec list not empty" messages for now, let's close the issue.

closed 2024-04-19 by jschlatow