genodelabs/genode #5182: Allow bootloader config tweaks for Sculpt/x86 system updates

nfeske, 2024-04-12:
Currently, Sculpt's system update mechanism merely copies the files found in the boot/ directory, most importantly hypervisor, bender, and image.elf.gz. Anticipating our plan to swap out Sculpt's kernel, the update mechanism must also capture boot-loader config changes.

It would be nice a retain Sculpt's ability to switch between different versions even in the event when swapping out the kernel. My suggestion: Starting from Sculpt 24.04, the update mechanism copies all files found in boot/grub/ (i.e., grub.cfg) as well as boot/grub/i386-pc/ (grub modules). So the grub.cfg is always installed along with the matching sculpt image. BTW, all previous system images happen to feature the grub files, they just were not used until now. So the rollback to earlier versions would work just fine.

There is one limitation though: It will not be possible to switch directly from a version older than 24.04 to a Sculpt version using a different kernel because the older version won't update the boot-loader config. I wonder, should we safeguard against this situation? One option would be to name the base-hw boot files differently (not image.elf.gz) so that this unsupported update path would result in the boot of the previously installed NOVA version. But I'm not really sure if this caution is exaggerated.

Users who want to manage a custom grub config should host their grub config at a distinct place, not in the boot directory of the Genode partition. I guess that this is the typical situation anyway in dual-boot setups where grub resides on a Linux boot partition.

I'd appreciate any feedback.

alex-ab, 2024-04-12:
> I wonder, should we safeguard against this situation? One option would be to name the base-hw boot files differently (not image.elf.gz) so that this unsupported update path would result in the boot of the previously installed NOVA version. But I'm not really sure if this caution is exaggerated.

imo: documenting it  is sufficient and don't take extra caution

> Users who want to manage a custom grub config should host their grub config at a distinct place, not in the boot directory of the Genode partition. I guess that this is the typical situation anyway in dual-boot setups where grub resides on a Linux boot partition.

Yes, I agree.  I personally do so.

nfeske, 2024-04-16:
After sleeping over it, I came to the realization that adding the kernel name as image suffix (e.g., `image-<kernel>.elf.gz`) is actually not a bad idea. Besides nicely sidestepping possible inconsistencies between the installed kernel and image.elf (the residual risk outlined above), it gives the user an useful indicator about the kind of image.elf, which is probably good for diagnostic purposes. An image.elf is not kernel-agnostic after all.

@alex-ab and @jschlatow, what do you think?

alex-ab, 2024-04-16:
I'm fine with it.

jschlatow, 2024-04-16:
That's very reasonable.

nfeske, 2024-04-17:
@cnuke as I'm currently in the suspend/resume trenches, could you do me the favor of testing this particular change (https://github.com/genodelabs/genode-allwinner/commit/3f88eca9028226dc9a83b0cc5c3d6d957532603d) by booting Sculpt on the PinePhone using a freshly written SD-card image produced after this change?

cnuke, 2024-04-18:
@nfeske I am on it.

cnuke, 2024-04-18:
@nfeske the commit appears to work on the PinePhone :+1:.

nfeske, 2024-04-18:
Thank you for testing!

nfeske, 2024-04-19:
Merged to master.

I don't plan to apply the change to other kernels than base-hw (covered by the commits) for now.

closed 2024-04-19 by nfeske