genodelabs/genode #5184: curl: update to version 8.7.1

cnuke, 2024-04-15:


cnuke, 2024-04-15:
Commit [5957f64](https://github.com/cnuke/genode/commit/5957f642dea48915a1c4dabe76f038608df12613) addresses this issue while commit [d00b6f6](https://github.com/cnuke/genode-world/commit/d00b6f6af6efd370b87aa8a7c93359b09dc811b9) and [c785ef0](https://github.com/cnuke/genodians.org/commit/c785ef04655275a550660fa8b3ffa89e01df29c8) adapt the external repositories.

Commit [c076e04](https://github.com/cnuke/genode/commit/c076e04331629ca0d4add554e1093d24b4f228bd) allows for setting the verbose option via the configuration.

chelmuth, 2024-04-16:
Commit 5957f642dea48915a1c4dabe76f038608df12613 misses the *curl_setup.patch* file.

cnuke, 2024-04-16:
> Commit [5957f64](https://github.com/genodelabs/genode/commit/5957f642dea48915a1c4dabe76f038608df12613) misses the _curl_setup.patch_ file.

You are right - sorry, commit [acabff6](https://github.com/cnuke/genode/commit/acabff611c471e27ac679b1aaaba8d47c0017611) adds the missing patch file.

cproc, 2024-04-17:
Fixup commit fa7118f adapts the `dbg_download` recipe as well.

commit 27be438e (2024-04-16): curl: update to version 8.7.1

This commits updates the contrib sources to version 8.7.1.

This version requires more random entropy as it queries OpenSSL
about the current random state and will bail if it is not sufficient.
Doubling the content of the '<inline>' VFS plugin as used in static
configurations seems satisfactory.

Furthermore DNS resolving needs a configured '<pipe>' plugin to work
properly.

Fixes #5184.
commit 344831ec (2024-04-19): curl: update to version 8.7.1

This commits updates the contrib sources to version 8.7.1.

This version requires more random entropy as it queries OpenSSL
about the current random state and will bail if it is not sufficient.
Doubling the content of the '<inline>' VFS plugin as used in static
configurations seems satisfactory.

Furthermore DNS resolving needs a configured '<pipe>' plugin to work
properly.

Fixes #5184.
closed 2024-04-19 by chelmuth