genodelabs/genode #5185: wifi: Error: Function iommu_device_unuse_default_domain not implemented

nfeske, 2024-04-18:
While testing Sculpt (staging), the above error occurs on an x201 (after suspend-resume) and on a Gen13 Framework (directly after booting when enabling wifi).

cnuke, 2024-04-18:
Commit [5301134](https://github.com/cnuke/genode/commit/5301134f7a4677a7f4db99b50beb7b2492895cef) replace the generated dummy.

The generated dummy is triggered in the first place because the driver cannot initialize the device on the X201 (thanks Alex for testing):

Good:
```
[…]
[runtime -> wifi] Intel(R) Wireless WiFi driver for Linux
[runtime -> wifi] iwlwifi 02:00.0: pci_enable_msi failed - -38
[runtime -> wifi] task_work_add: task: 0000000078e4a67f work: 000000002f7ef564 notify: 0
[runtime -> wifi] NET: Registered PF_PACKET protocol family
[runtime -> wifi] sched_clock: Marking stable (46880000, 28539000)->(147583000, -72164000)
[runtime -> wifi] wireless-reg: Loading compiled-in X.509 certificates for regulatory database
[runtime -> wifi] Warning: /dev/rtc not readable, returning 0
[runtime -> wifi] wireless-reg: loaded regulatory.db is malformed or signature is missing/invalid
[runtime -> wifi] iwlwifi 02:00.0: loaded firmware version 9.221.4.1 build 25532 6000-4.ucode op_mode iwldvm
[runtime -> wifi] iwlwifi 02:00.0: CONFIG_IWLWIFI_DEBUG disabled
[runtime -> wifi] iwlwifi 02:00.0: CONFIG_IWLWIFI_DEBUGFS disabled
[runtime -> wifi] iwlwifi 02:00.0: CONFIG_IWLWIFI_DEVICE_TRACING disabled
[runtime -> wifi] iwlwifi 02:00.0: Detected Intel(R) Centrino(R) Ultimate-N 6300 AGN, REV=0x74
[runtime -> wifi] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[runtime -> wifi] iwlwifi 02:00.0: Radio type=0x0-0x3-0x1
[runtime -> wifi] iwlwifi 02:00.0: Radio type=0x0-0x3-0x1
[…]
```

Bad:
```
[…]
[runtime -> wifi] Intel(R) Wireless WiFi driver for Linux
[runtime -> wifi] iwlwifi 02:00.0: HW_REV=0xFFFFFFFF, PCI issues?
[runtime -> wifi] iwlwifi: probe of 02:00.0 failed with error -5
[runtime -> wifi] Error: Function iommu_device_unuse_default_domain not implemented yet!
[runtime -> wifi] backtrace "ep"
[runtime -> wifi] Will sleep forever...
[…]
```

It looks like the wifi device does not respond after the resume (judging by `HW_REV` being all 1s).

chelmuth, 2024-04-19:
The fix for this issue is merged to master. Let's open a new issue when addressing the PCI-bridge handling on suspend-resume.