genodelabs/genode #5186: record_play_mixer: reduce volume of diagnostic messages by default

nfeske, 2024-04-19:
In Sculpt, the latency warnings by the record-play mixer are too many (rate of 1 second). Those warnings were intended for diagnosis. But Sculpt users will hardly participate in the debugging of latency issues but have to ensure all his warning noise.

To address this issue, a `warning_rate_ms` attribute could be added to the record_play_mixer config. By using 10000 ms default, regular users won't be bothered when seeing a message every now and then. But for latency diagnosis, we can set the attribute even to a low value like 200 ms.

Related to #5174

chelmuth, 2024-04-19:
I would even be more user-oriented and implement `warning_rate_ms` as mere debugging feature with a default of `0` (no warnings).

nfeske, 2024-04-19:
@cnuke what do you think?

nfeske, 2024-04-19:
@chelmuth this would be fine for me as well.

cnuke, 2024-04-19:
@nfeske I think that is a good idea (introducing the feature and setting it to `0` be default).

commit 084a14b1 (2024-04-26): record_play_mixer: make warning rate configurable

This patch disables latency warnings by default. The warnings can be
enabled by setting the 'warning_rate_ms` value to the desired
maximum rate.

Fixes #5186
Issue #5174
closed 2024-04-26 by chelmuth