genodelabs/genode #5190: file_vault_gui: dialog-based front end for the file vault

m-stein, 2024-04-23:
The File Vault used to implement both, managing a Tresor container + file system and providing a GUI via menu_view. Furthermore, for controlling the GUI, the File Vault didn't use the Dialog API. These two aspects made the component quite bulky and complex. In this issue, I will implement a Dialog-based GUI front end for the File Vault in a separate component that uses the Report/Config interface of the File Vault as back end. Thereby, I hope to be able to reduce the overall complexity of the system significantly.

m-stein, 2024-04-23:
I just tested the Sculpt package of the new approach:
5dd2679197 file_vault: further reduce typical quota needs
a33bf088e1 file_vault: do not consider ui_config version
88d9e1c0b4 file_vault: fix locking issue
f4d84b5574 file_vault: clean up
e12bd58e07 file_vault: remove need for state file
c3640cf754 file_vault_gui: dialog-based file_vault front end
54b82b4866 text_area: remove unused min_width, min_height

chelmuth, 2024-04-23:
Merged to staging.

nfeske, 2024-04-26:
Fixed in master.

closed 2024-04-26 by nfeske