genodelabs/genode #5192: fix nic_router_uplinks fails

m-stein, 2024-04-24:
The nic_router_uplinks test currently fails on x86_64/x86_64/pc/hw|nova|sel4.

m-stein, 2024-04-24:
@cnuke I failed to fix it on sel4 because the wifi driver doesn't come up. Do you have a clue what to do about this?
```
[init -> wifi_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[init -> wifi_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> wifi_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> wifi_drv] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[init -> wifi_drv] time-clocksource: Switched to clocksource dde_counter
[init -> wifi_drv] Intel(R) Wireless WiFi driver for Linux
[init -> wifi_drv] iwlwifi 02:00.0: pci_enable_msi failed - -38
[init -> wifi_drv] task_work_add: task: 0000000037b9efe4 work: 0000000000b84fe6 notify: 0
[init -> wifi_drv] NET: Registered PF_PACKET protocol family
[init -> wifi_drv] sched_clock: Marking stable (54300000, 752117000)->(854355000, -47938000)
[init -> wifi_drv] wireless-reg: Loading compiled-in X.509 certificates for regulatory database
[init -> wifi_drv] wireless-reg: loaded regulatory.db is malformed or signature is missing/invalid
[init -> wifi_drv] iwlwifi 02:00.0: api flags index 2 larger than supported by driver
[init -> wifi_drv] iwlwifi 02:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
[init -> wifi_drv] iwlwifi 02:00.0: loaded firmware version 68.01d30b0c.0 ty-a0-gf-a0-68.ucode op_mode iwlmvm
[init -> wifi_drv] iwlwifi 02:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
Warning: invalid signal-context capability
Warning: flush page table entries - mapping cache full - PD: init -> wifi_drv
[init -> wifi_drv] iwlwifi 02:00.0: loaded PNVM version dda57f4f
[init -> wifi_drv] iwlwifi 02:00.0: Detected RF GF, rfid=0x10d000
[init -> wifi_drv] iwlwifi 02:00.0: base HW address: 3c:e9:f7:de:a5:17
Warning: flush page table entries - mapping cache full - PD: init -> wifi_drv
Warning: void Core::Rpc_cap_factory::free(Genode::Native_capability) not implemented - resources leaked: 0x100
[init -> wifi_drv] iwlwifi 02:00.0: firmware didn't ACK the reset - continue anyway
[init -> wifi_drv] iwlwifi 02:00.0: Start IWL Error Log Dump:
[init -> wifi_drv] iwlwifi 02:00.0: Transport status: 0x0000004A, valid: 6
[init -> wifi_drv] iwlwifi 02:00.0: Loaded firmware version: 68.01d30b0c.0 ty-a0-gf-a0-68.ucode
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN       
[init -> wifi_drv] iwlwifi 02:00.0: 0x00A082F0 | trm_hw_status0
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | trm_hw_status1
[init -> wifi_drv] iwlwifi 02:00.0: 0x004DAF86 | branchlink2
[init -> wifi_drv] iwlwifi 02:00.0: 0x004D13D6 | interruptlink1
[init -> wifi_drv] iwlwifi 02:00.0: 0x004D13D6 | interruptlink2
[init -> wifi_drv] iwlwifi 02:00.0: 0x0001699A | data1
[init -> wifi_drv] iwlwifi 02:00.0: 0x01000000 | data2
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | data3
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | beacon time
[init -> wifi_drv] iwlwifi 02:00.0: 0x00034382 | tsf low
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | tsf hi
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | time gp1
[init -> wifi_drv] iwlwifi 02:00.0: 0x0004848E | time gp2
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000001 | uCode revision type
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000044 | uCode version major
[init -> wifi_drv] iwlwifi 02:00.0: 0x01D30B0C | uCode version minor
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000420 | hw version
[init -> wifi_drv] iwlwifi 02:00.0: 0x00C89002 | board version
[init -> wifi_drv] iwlwifi 02:00.0: 0x8061FF00 | hcmd
[init -> wifi_drv] iwlwifi 02:00.0: 0x20020000 | isr0
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | isr1
[init -> wifi_drv] iwlwifi 02:00.0: 0x48F00002 | isr2
[init -> wifi_drv] iwlwifi 02:00.0: 0x00C0001C | isr3
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | isr4
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | last cmd Id
[init -> wifi_drv] iwlwifi 02:00.0: 0x0001699A | wait_event
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | l2p_control
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | l2p_duration
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | l2p_mhvalid
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | l2p_addr_match
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000009 | lmpm_pmg_sel
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | timestamp
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000028 | flow_handler
[init -> wifi_drv] iwlwifi 02:00.0: Start IWL Error Log Dump:
[init -> wifi_drv] iwlwifi 02:00.0: Transport status: 0x0000004A, valid: 7
[init -> wifi_drv] iwlwifi 02:00.0: 0x20000074 | ADVANCED_SYSASSERT
[init -> wifi_drv] iwlwifi 02:00.0: 0x00000000 | umac branchlink1
[init -> wifi_drv] iwlwifi 02:00.0: 0x8045E21C | umac branchlink2
[init -> wifi_drv] iwlwifi 02:00.0: 0x804804AE | umac interruptlink1
[init -> wifi_drv] iwlwifi 02:00.0: 0x804804AE | umac interruptlink2
[... 60 lines cut ...]
[init -> wifi_drv] iwlwifi 02:00.0: Session protection failure
[init -> wifi_drv] iwlwifi 02:00.0: Not associated and the session protection is over already...
[init -> wifi_drv] wlan0: Connection to AP 4c:1b:86:5b:de:18 lost
Warning: flush page table entries - mapping cache full - PD: init -> wifi_drv
[init -> wifi_drv] wlan0: send auth to 4c:1b:86:5b:de:18 (try 3/3)
[init -> wifi_drv] iwlwifi 02:00.0: Session protection failure
[init -> wifi_drv] iwlwifi 02:00.0: Not associated and the session protection is over already...
[init -> wifi_drv] wlan0: Connection to AP 4c:1b:86:5b:de:18 lost
[... cut ...]
[init -> wifi_drv] iwlwifi 02:00.0: Not associated and the session protection is over already...
[init -> wifi_drv] wlan0: Connection to AP 4c:1b:86:5b:de:18 lost
[init -> wifi_drv] wlan0: send auth to 4c:1b:86:5b:de:18 (try 3/3)
[init -> wifi_drv] iwlwifi 02:00.0: Session protection failure
[init -> wifi_drv] iwlwifi 02:00.0: Not associated and the session protection is over already...
[init -> wifi_drv] wlan0: Connection to AP 4c:1b:86:5b:de:18 lost
[init -> wifi_drv] wlan0: authentication with 4c:1b:86:5b:de:18 timed out
```

m-stein, 2024-04-24:
@chelmuth This 467b5c96ab should fix hw and nova.

chelmuth, 2024-04-24:
Merged to staging.

cnuke, 2024-04-25:
@m-stein I am afraid I have no insight I could provide but judging by the history of CI results the `wifi_drv` has not been tested or has had issue on `seL4` for quite some time now. The messages that look suspicious though are

```
[…]
Warning: invalid signal-context capability
Warning: flush page table entries - mapping cache full - PD: init -> wifi_drv
[init -> wifi_drv] iwlwifi 02:00.0: loaded PNVM version dda57f4f
[init -> wifi_drv] iwlwifi 02:00.0: Detected RF GF, rfid=0x10d000
[init -> wifi_drv] iwlwifi 02:00.0: base HW address: 3c:e9:f7:de:a5:17
Warning: flush page table entries - mapping cache full - PD: init -> wifi_drv
Warning: void Core::Rpc_cap_factory::free(Genode::Native_capability) not implemented - resources leaked: 0x100
[…]
```

as I would expect the working-set of memory of the driver to be small enought to not blow the static page table limit.

m-stein, 2024-04-25:
@chelmuth @cnuke Considering cnuke's answer, may I disable sel4 for the test?

alex-ab, 2024-04-25:
@m-stein: may you try to give the wifi drv more memory as test case for seL4, it may effect the size of the page cache (so avoiding the flush) ?

m-stein, 2024-04-25:
@alex-ab Thanks for the hint! Will try that.

chelmuth, 2024-05-30:
What are the plans with this issue? Any news after trying @alex-ab's advice?

m-stein, 2024-05-31:
I'm sorry, I forgot to report my findings. Raising quota apparently doesn't fix the sel4 issue with the wifi_drv. Beside that, I'm still at it. However, so far, I prioritize the SSH issue higher.

m-stein, 2024-06-14:
@chelmuth With these, I could successively test all failing targets:
e83de3bffc run/nic_router_uplinks: use test component
8b4a9e9a51 ping: support reporting results

commit e83de3bf (2024-06-14): run/nic_router_uplinks: use test component

So far, this test used dynamic_rom for the re-configuration of the nic router
and tested for the expected ping results by inspecting the log with the run
tool. However, this approach had two issues:

* Timing differs significantly on different targets and so the dynamic_rom had
  the difficult task of compensating with heuristics without bloating the test
  duration too much.

* In case of a failing test, it was difficult to determine the cause as the
  test kept running and produced output for quite some time and there was also
  no specific error m [... cut]
closed 2024-06-20 by nfeske