genodelabs/genode #5199: sculpt: make inspect vfs available to other components

jschlatow, 2024-04-29:
Currently, there is no easy way to access arbitrary USB sticks/storages from within system_shell or any other customised unix-like shell.  I therefore end up using Sculpt's inspect view, which, however, provides only a limited set of tools. By separating the vfs part from the inspect sub-init, we could provide a route to this vfs in Sculpt to other components such as system_shell.

Related to this: It would also be nice to have a way to "automount" (i.e. auto-inspect) particular file systems at boot. In my case, I have an additional nvme partition that stores my VMs.

nfeske, 2024-05-02:
Given my plan to remove the inspect view once the file manager got into a usable shape, I'd rather not extend the current mechanism.

That said, we could/should contemplate a nice way to cover your need in a different way. E.g., by allowing the leitzentrale user to create a vfs component where one can select the mounted file systems as checkboxes. I have planned to work on the dynamic vfs reconfiguration at the end of the year (according to our road map). An interactively composable vfs would be a nice user-visible application of this work.

jschlatow, 2024-05-06:
> That said, we could/should contemplate a nice way to cover your need in a different way. E.g., by allowing the leitzentrale user to create a vfs component where one can select the mounted file [...]

Perfect!