genodelabs/genode #5202: sculpt: Adding French Bépo keyboaurd

goutnet, 2024-04-29:
Hi there,

I am starting to use scuplt OS, but alas my keyboard layout (bépo: https://en.wikipedia.org/wiki/B%C3%89PO ) is not yet supported.

As discussed with @nfeske  the best course of action would be to create an issue and a pull request to fill this need.

Will tag this issue in my next commits :)

goutnet, 2024-04-29:
The above commits fixes this issue for me (and add the Bépo support)

With that said: while doing the test, I realized the terminal does not show accentuated characters properly (while the font does include those).

The issue can be seen when trying to type characters in a command set for example.

I will look into this too, and if the fix is simple, I'll tag this issue there. (else that will be for a new issue).

goutnet, 2024-04-30:
@chelmuth  I just pushed a lighter version of the bepo layout, let me know what you think

chelmuth, 2024-04-30:
Look good to me. Merged to staging.

nfeske, 2024-05-30:
Fixed in Genode 24.05

closed 2024-05-30 by nfeske