genodelabs/genode #5211: sculpt:  add support for ventoy

stavpup, 2024-05-11:
[support for ventoy](sculpt:  add support for ventoy)

Ventoy is an open source tool to create bootable USB drive for ISO/WIM/IMG/VHD(x)/EFI files.
With ventoy, you don't need to format the disk over and over, you just need to copy the ISO/WIM/IMG/VHD(x)/EFI files to the USB drive and boot them directly.

https://www.ventoy.net/en/index.html

nfeske, 2024-05-30:
@stavpup thanks for making us Genode developers aware of the project. However, I fail to see an obvious connection to our work, or an actionable plan.

Do you intend to work on this issue? If not, I'd appreciate you closing this issue.

stavpup, 2024-05-30:
> @stavpup thanks for making us Genode developers aware of the project. However, I fail to see an obvious connection to our work, or an actionable plan.
> 
> Do you intend to work on this issue? If not, I'd appreciate you closing this issue.

Thank you for your response. I understand the need for a clear connection to Genode's work. My intention was to suggest an opportunity to integrate Ventoy's functionality into Genode, making it easier to create bootable USB drives for Genode's ISO images and other files.
I'm currently unable to work on this myself, but I wanted to share the idea in case any contributors or developers find it interesting and want to explore it further. I apologize if this isn't the appropriate platform for such suggestions. If it's alright, I'll keep the issue open in case anyone would like to pick it up. If not, please let me know, and I'll close it accordingly.

nfeske, 2024-06-17:
Please note that Genode's build system already produces ISO and disk images that are immediately bootable. The benefit of Ventoy over the existing solution seems to be the convenience of hosting multiple OSes on a single USB stick. But as this (presumed) convenience apparently comes at the price of added uncertainty (https://news.ycombinator.com/item?id=40689629) I think we should not embrace it.

stavpup, 2024-06-20:
> Please note that Genode's build system already produces ISO and disk images that are immediately bootable. The benefit of Ventoy over the existing solution seems to be the convenience of hosting [...]

There are some comments that may address those issues about blobs

"The three files explicitly mentioned on the original message opening the thread.
They have either direct source code / notes referencing where the source code came from in the same folder.

Cryptsetup: https://github.com/ventoy/Ventoy/tree/v1.0.99/cryptsetup
BSD: https://github.com/ventoy/Ventoy/tree/v1.0.99/Unix
Dmsetup: https://github.com/ventoy/Ventoy/tree/v1.0.99/DMSETUP

Until you can point out a binary that has no source/readme next to it there isn't a point in yapping in this thread any more about blobs."

https://github.com/ventoy/Ventoy/issues/2795

chelmuth, 2024-07-08:
We will not look into ventoy in the foreseeable future.

closed 2024-07-08 by chelmuth