genodelabs/genode #5214: Busybox port for reproduceable Linux image creation

skalk, 2024-05-13:
In the process of Linux device driver porting, as well as to build simple Linux VM images it is helpful to provide a common way to build a simple initramfs for the Linux kernel via the Busybox project.

chelmuth, 2024-05-14:
Nice, I just tested this feature and have some remarks for discussion.

1. It would be nice to have a little documentation what I could do if I get `*** Need to have 'arm-linux-gnueabihf-gcc' installed`. Maybe a note in the port file that suggest `sudo apt install gcc-arm-linux-gnueabihf`?
2. For clarity, I suggest to use `BB_MK_ARGS := ARCH=x86_64 CROSS_COMPILE=x86_64-linux-gnu-gcc` in *busybox/x86_64/target.mk*.
3. Does the port create the rootfs as directory tree only or is there some image file?

skalk, 2024-05-15:
@chelmuth I've added the documentation including the requirements in a README file, and have add the x86_64 suggestion.

chelmuth, 2024-05-15:
Great, merged to staging.

closed 2024-05-30 by chelmuth