genodelabs/genode #5215: base-hw: busy loop while booting Tigerlake+ platforms

chelmuth, 2024-05-13:
It seems the initialization enters a busy loop on more CPUs. I successfully ran simple scenarios on Whiskeylake (i7-8665UE) but failed on Tigerlake (i7-1185G7) and newer.

chelmuth, 2024-06-03:
With IOMMU support progressing pretty well on base-hw (https://github.com/genodelabs/genode/issues/5066#issuecomment-2139050160), testing on recent platforms (e.g., Framework or T14 with Intel Gen12) becomes more and more relevant. Are there any ideas how to proceed with this issue?

jschlatow, 2024-07-23:
@chelmuth I am wondering whether the workaround mentioned in #5304 has any effect on this issue.

chelmuth, 2024-07-24:
Most certainly you identified the right spot. Will give it a try on Tigerlake.

chelmuth, 2024-07-24:
Heureka, I successfully ran `make run/log KERNEL=hw` on i7-1185G7 for the time just now. So, this issue is a duplicate of #5304 (or vice versa).

jschlatow, 2024-07-24:
> Heureka, I successfully ran `make run/log KERNEL=hw` on i7-1185G7 for the time just now. So, this issue is a duplicate of #5304 (or vice versa).

Perfect. I closed the other issue as a duplicate.

I'm still wondering how to solve this issue properly. I believe we could add info about the availability of an ACPI timer during bootstrap, store this in the boot info, and use the ACPI timer (if available) for timer calibration. The PIT-based calibration could stay in place as a fallback.

chelmuth, 2024-07-24:
I suggest to replace the PIT-based calibration with an implementation based on the ACPI timer (which should be always available?). Later we could carefully add code to discover exact values from CPUID for most relevant platforms, e.g., Haswell/Broadwell and later as well as our chosen Qemu-emulated CPU Nehalem. In any case, all this code should stay in bootstrap and, thus, outside of the kernel. Especially, everything from *base-hw/src/include/hw/spec/x86_64 /x86_64.h* that was added for this purpose must be removed again.

atopia, 2025-01-15:
@chelmuth My [/5215-hw-bootstrap-calibration](https://github.com/atopia/genode/tree/5215-hw-bootstrap-calibration) branch (rebased on staging) is ready for review. My vbox modular sculpt test scenario on the test machine worked fine with the changes on top of master but is failing on current staging - but so is staging without my changes, so the error is outside of the scope this issue.

jschlatow, 2025-01-20:
@atopia I cherry-picked your latest four commits onto my _sculpt_24.10_. Unfortunately, building _bootstrap_ fails with the following message:

```
  Library bootstrap-hw-pc.lib.a
    CONVERT  hw/memory_consts.h
    COMPILE  bootstrap/init.o
    COMPILE  bootstrap/lock.o
    COMPILE  bootstrap/log.o
    COMPILE  bootstrap/platform.o
    ASSEMBLE bootstrap/spec/x86_64/crt0.o
    ASSEMBLE bootstrap/spec/x86_64/crt0_translation_table.o
    COMPILE  bootstrap/spec/x86_64/platform.o
/home/johannes/repos/genode/repos/base-hw/src/bootstrap/spec/x86_64/platform.cc:21:10: fatal error: port_io.h: No such file or directory
   21 | #include <port_io.h>
      |          ^~~~~~~~~~~

```

Have I missed any commit?

atopia, 2025-01-20:
@jschlatow sorry my bad, I missed a copy of `port_io.h` in my working tree. [65a0939](https://github.com/atopia/genode/commit/65a09392075de26d1fef89a2502468b74a5a4ba8) should fix the issue.

jschlatow, 2025-01-22:
@atopia Thanks for the fixup. Unfortunately, I'm unable to boot Sculpt (hw) on the StarLite since I encountered several issues:

1. The device reboots shortly after showing the Genode boot logo. When I skip bender and directly boot into hw, it stops at the boot logo without resetting. (I am able to boot this mod on my t490s, though.)
2. Even with the original workaround, I am unable to boot Sculpt 24.10 (hw) on the StarLite.

I think I will try to bisect this unless anyone already has an idea what change(s) since Sculpt 24.04 might be responsible for issue 1 and/or 2.

chelmuth, 2025-01-22:
I merged the series plus my fixups for readability and a bug in `calibrate_lapic_frequency()`.

be7df4fa823 fixup "hw: always serialize rdtsc reads" (final cosmetic touches)
6793143d31e hw: always serialize rdtsc reads
28ecbbbb71e fixup "hw: calibrate Local APIC via ACPI timer" (final cosmetic touches)
8bdddbd46a3 hw: calibrate Local APIC via ACPI timer
103d03b5906 fixup "hw: calibrate TSC via ACPI timer" (final cosmetic touches)
5def882001a hw: calibrate TSC via ACPI timer

atopia, 2025-01-22:
> I merged the series plus my fixups for readability and a bug in `calibrate_lapic_frequency()`.

Please also consider bf140c1 to make the serialized rdtsc robust against reorderings by the compiler.

jschlatow, 2025-01-23:
> > I think I will try to bisect this unless anyone already has an idea what change(s) since Sculpt 24.04 might be responsible for issue 1 and/or 2.
> 
> May commit [eb7aea8](https://github.com/genodelabs/genode/commit/eb7aea82b82fc014bd9cb0933f601060dfd61fa3) be related to this issue?

The issue is already present with 24.08. According to _git bisect_, 9258004cc7ff3bcf3af953c7a5ed17601500ed5f appears to be the culprit. If I revert this commit along with a7b4add27c95591a175aa0b6485a47ccae2d6c96, I am able to boot _run/intel_fb_ on the StarLite.

@skalk Do you have any idea whay the actual issue might be?

skalk, 2025-01-23:
Ouch, that is a quite invasive commit unfortunately. Maybe it is the way the CPU index is read now? In `src/bootstrap/spec/x86_64/platform.cc` within `Board::Board()`? But I'm just guessing. I'm afraid you have to instrument the bootstrap code to the point it resets.

jschlatow, 2025-01-24:
@skalk Looking at _/proc/cpuinfo_, I noticed that the apic IDs of the four cores are not numbered consecutively but actually 0, 2, 4, 6. Can base-hw deal with this already?

skalk, 2025-01-24:
> [@skalk](https://github.com/skalk) Looking at _/proc/cpuinfo_, I noticed that the apic IDs of the four cores are not numbered consecutively but actually 0, 2, 4, 6. Can base-hw deal with this already?

It is intended to do so. But as your hardware platform seems to be the first with this sparse apic ids, its a hot lead.
You do not have some serial output for this machine?

jschlatow, 2025-01-24:
> It is intended to do so. But as your hardware platform seems to be the first with this sparse apic ids, its a hot lead. You do not have some serial output for this machine?

Unfortunately, I don't. My usual approach was to trigger a conditional device reset for debugging. I'm thinking about implementing logging to RAM like the logmem option for NOVA so that I can at least look at the log via GRUB after a warm reboot.

atopia, 2025-01-27:
**TL;DR:** for me the state with the reverted commits works with sparse APIC IDs

My Lenovo X1 Nano Gen 2 also has sparse APIC IDs. With https://github.com/genodelabs/genode/commit/9258004cc7ff3bcf3af953c7a5ed17601500ed5f and https://github.com/genodelabs/genode/commit/a7b4add27c95591a175aa0b6485a47ccae2d6c96 reverted, I am able to boot hw. Due to an issue with the wifi driver on hw I haven't been able to do thorough testing yet, but just as an observation I want to mention that while the P-Cores with  hyperthreads as well as the E-cores are presented as a 2 by 10 matrix on Nova, on hw with the reverted commits the CPUs are presented as a 1 by 20 matrix (this might be fully expected e.g. if hw does not detect the hyperthreads, I haven't looked into it deeper).

skalk, 2025-01-27:
> **TL;DR:** for me the state with the reverted commits works with sparse APIC IDs
> 
> My Lenovo X1 Nano Gen 2 also has sparse APIC IDs. With [9258004](https://github.com/genodelabs/genode/commit/9258004cc7ff3bcf3af953c7a5ed17601500ed5f) and [a7b4add](https://github.com/genodelabs/genode/commit/a7b4add27c95591a175aa0b6485a47ccae2d6c96) reverted, I am able to boot hw. Due to an issue with the wifi driver on hw I haven't been able to do thorough testing yet, but just as an observation I want to mention that while the P-Cores with hyperthreads as well as the E-cores are presented as a 2 by 10 matrix on Nova, on hw with the reverted commits the CPUs are presented as a 1 by 20 matrix (this might be fully expected e.g. if hw does not detect the hyperthreads, I haven't looked into it deeper).

Can we use your Lenovo X1 with AMT or similar to debug the failing changes of the sparse CPU area?

chelmuth, 2025-01-27:
NOVA gathers the following info for U7412 (fuji6), which is a i7-1270P.

```
[-1] parse_lapic: cpu=0 acpi_id=0 apic_id=0
[-1] parse_lapic: cpu=1 acpi_id=1 apic_id=1
[-1] parse_lapic: cpu=2 acpi_id=2 apic_id=8
[-1] parse_lapic: cpu=3 acpi_id=3 apic_id=9
[-1] parse_lapic: cpu=4 acpi_id=4 apic_id=16
[-1] parse_lapic: cpu=5 acpi_id=5 apic_id=17
[-1] parse_lapic: cpu=6 acpi_id=6 apic_id=24
[-1] parse_lapic: cpu=7 acpi_id=7 apic_id=25
[-1] parse_lapic: cpu=8 acpi_id=8 apic_id=32
[-1] parse_lapic: cpu=9 acpi_id=9 apic_id=34
[-1] parse_lapic: cpu=10 acpi_id=10 apic_id=36
[-1] parse_lapic: cpu=11 acpi_id=11 apic_id=38
[-1] parse_lapic: cpu=12 acpi_id=12 apic_id=40
[-1] parse_lapic: cpu=13 acpi_id=13 apic_id=42
[-1] parse_lapic: cpu=14 acpi_id=14 apic_id=44
[-1] parse_lapic: cpu=15 acpi_id=15 apic_id=46
```

Maybe the issue can be investigated on this device? `make run/log KERNEL=hw` failed to boot during a smoke test and logged `Kernel: Cxx_heap_uninitialized`.

skalk, 2025-01-27:
@atopia I assume the machine referenced by @chelmuth triggers the same issue, and should be ready for debugging, thank you.

commit 8bdddbd4 (2025-01-22): hw: calibrate Local APIC via ACPI timer

Upto now, bootstrap used the Programmable Interval Timer to set a
suitable divider and determine the frequency of the Local APIC.
The PIT is not available on recent x86_64 hardware anymore.

Move Local APIC calibration to bootstrap and use the ACPI timer as a
reference. Clean up hw's timer implementation a little and disable the
PIT in bootstrap.

Fixes #5215
closed 2025-01-30 by chelmuth