genodelabs/genode #5217: consolidate page table implementations

atopia, 2024-05-15:
Currently, the Intel IOMMU driver in the pc platform driver uses a page table implementation that is heavily borrowed and generalized from the x86 page table implementation in hw. The duplication of existing hw definitions hinders adoption of the generalized page table code for implementing nested page tables.

To clean up the current page table implementations:
- Make the x86 page table template base compatible with existing interfaces
- Move common definitions and implementations to base so they can be used throughout the Genode OS framework
- Adapt hw to use the page table allocator from base
- Adapt hw to use the page table template on x86_64

atopia, 2024-05-15:
After consulting with @nfeske, we resolved to take this issue out of the critical path for #5218 as much as possible. I have force-pushed and rebased my [branch](atopia/genode/tree/5217-consolidate-page-tables) on current staging. The first three commits mostly move headers around, I hope they are pretty uncontroversial. Commit [b9054d7](https://github.com/atopia/genode/commit/b9054d7bbb8810432631b50e6b4304c415c5c0b2) makes the extra arguments in @jschlatow's optional to bring the interface in line with hw's current implementation.

I will offer the commits changing hw's page table allocator and x86 page table in a separate issue, where there will also be room to discuss the allocator interface changes.

chelmuth, 2024-05-21:
Commit 184d06cb3a40285ffebbe6d8aa43012cb3c128e7 seems to break all ARM builds. Please have a look.

atopia, 2024-05-21:
What breakage are you seeing exactly? A `make KERNEL=hw run/log` works for me on all architectures on current staging both when I used my long-standing build directories and with fresh build directories.

atopia, 2024-05-21:
Ah I see, for ARM v7 get get the error with `BOARD=pbxa9`, but when I was testing locally with `BOARD=virt_qemu_arm_v7a` it worked.

chelmuth, 2024-05-21:
For record the error was like follows (from the nightly build-packages email).

```
In file included from /data/depot/checker/src/base-hw-virt_qemu_riscv/2024-05-20/src/include/hw/spec/riscv/qemu_board.h:19,
                 from /data/depot/checker/src/base-hw-virt_qemu_riscv/2024-05-20/src/core/board/virt_qemu_riscv/board.h:18,
                 from /data/depot/checker/src/base-hw-virt_qemu_riscv/2024-05-20/src/core/kernel/pd.h:22,
                 from /data/depot/checker/src/base-hw-virt_qemu_riscv/2024-05-20/src/core/platform.h:31,
                 from /data/depot/checker/src/base-hw-virt_qemu_riscv/2024-05-20/src/core/thread_start.cc:26:
/data/depot/checker/src/base-hw-virt_qemu_riscv/2024-05-20/src/include/hw/spec/riscv/page_table.h:18:10: fatal error: hw/page_flags.h: No such file or directory
   18 | #include <hw/page_flags.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[6]: *** [/data/genode/repos/base/mk/generic.mk:72: thread_start.o] Error 1
make[6]: Target 'all' not remade because of errors.
make[5]: *** [var/libdeps:199: core-hw-virt_qemu_riscv.lib.a] Error 2
```

atopia, 2024-05-21:
@chelmuth [7ca7bd9](https://github.com/atopia/genode/commit/7ca7bd917ce11c2ec581019992a40eca2e4d0878) should fix the issue.

chelmuth, 2024-05-30:
I must admit that I lost track of the state of this issue and also how it is related to #5221. @atopia would you mind to clarify the situation?

atopia, 2024-06-10:
> I must admit that I lost track of the state of this issue and also how it is related to #5221. @atopia would you mind to clarify the situation?

This issue can be closed as completed, as the narrowed scope for the release has been merged. As stated [above](https://github.com/genodelabs/genode/issues/5217#issuecomment-2112841194), I will open another issue with the left out consolidation of hw's page table allocator and x86 page table implementation.

https://github.com/genodelabs/genode/issues/5221 is an orthogonal issue only relevant in the scope of x86 virtualization on hw, but not to the larger consolidation work. I intend to discuss a clean(er) implementation with @nfeske when I'm at the office.

closed 2024-06-11 by chelmuth