genodelabs/genode #5218: hw: implement nested page table

atopia, 2024-05-15:
Implement a nested page table for x86 virtualization using the x86 page table template base implementation.

atopia, 2024-05-16:
I've successfully tested [this branch](atopia/genode/tree/5218-hw-nested-page-table) with Sculpt and @jschlatow's firefox-av VM package albeit somehow audio wasn't playing, but we're still talking about a Linux VM here :smiley: No seriously, I still have to check that audio works in principle but from my quick test, it seems that the current state can be offered as "First Sculpt PC variant on the base-hw kernel" with the firefox-av VM as test case.

chelmuth, 2024-05-16:
That's great news, however I have follow-up questions.

- Looking at the patch only, it looks like both - EPT and NPT - are handled by the same page-table implementation, right?
- If *clflush.h* must be moved from *drivers/platform/pc* to *base/include*, it should be placed in the  *spec/x86/cpu/* subdirectory.

I must admit I'm unsure about the state of the branch. Thus, it would be good if you notify me if it's ready for merge and @jschlatow as well as @skalk approve. Setting the [fixed] label would be sensible, then.

atopia, 2024-05-16:
>     * If _clflush.h_ must be moved from _drivers/platform/pc_ to _base/include_, it should be placed in the  _spec/x86/cpu/_ subdirectory.

If fixed this with a fixup commit in #5217.

atopia, 2024-05-16:
Maybe @nfeske has an idea how to get to a better working version?

nfeske, 2024-05-17:
> Maybe @nfeske has an idea how to get to a better working version?

I think the first version is fine for now. I agree that your second commit goes in a good direction. But in my opinion we should not force it in the short term of the current release. Let us keep the clear separation of both page table formats for a subsequent step (without haste) after the release.

atopia, 2024-05-17:
OK. I pushed the first version of my `Vm_page_table` runtime selection code with an updated commit message. I would suggest to merge this version on top of the EPT implementation commit, as the suboptimal part
* is self-contained in `vm_page_table.h`
* does not introduce unnecessary computational overhead (accessing the underlying implementation via branches should have a similar overhead to accessing a vtable)
* does not incur memory overhead, safe for a static enum to cache the result of the virtualization technology detection

If this implementation is acceptable for now, I will open an issue to clean up the `Vm_page_table` implementation and provide cached virtualization technology detection at a higher level.

chelmuth, 2024-05-17:
@atopia please check current staging for missing commits.

atopia, 2024-05-17:
Staging looks good to me, thanks!

commit b7450810 (2024-05-17): hw: implement EPT page table

Implement a nested page table to use with x86 virtualization.

Fixes #5218
commit 221d0c6c (2024-05-29): hw: implement EPT page table

Implement a nested page table to use with x86 virtualization.

Fixes #5218