genodelabs/genode #5221: hw: clean up runtime page table switching for x86 virtualization

atopia, 2024-05-17:
The `Vm_page_table` implementation in #5218 needs to be cleaned up. At minimum, virtualization technology detection needs to be cached at a higher level. On top of that, it should be considered to implement the runtime switch in a more idiomatic manner, such as by implementing `Vm_page_table` as a virtual base class and constructing a matching implementation at runtime. Since the actual page table implementations are accessed by hardware, this solution would obviously require wrapper classes that use the correct underlying implementation. The separation of the actual page table implementation in memory (e.g. `Ept`) and its accessor class (e.g. `Vm_page_table_ept`) requires changes to the x86-specific implementation of `Vm_session_component`, as `_table` would now store a class instance that is separate from the memory provided by `_alloc_table()`, and memory management as well as providing the physical address to hardware virtualization needs to be adapted accordingly. Commit [fe899c7](https://github.com/atopia/genode/commit/fe899c719b6fe22281f778bd558cf75740de1a45) on branch [5218-hw-nested-page-table-lamdba](https://github.com/atopia/genode/tree/5218-hw-nested-page-table-lamdba) is a first, discfunctional draft implementation. We need to figure out if the clarity in language constructs outweighs the complexity in making `Vm_page_table` a virtual class that - contrary to the other architectures in hw - is separate from the actual nested page table.

nfeske, 2024-05-21:
Thank you for opening this issue.

Regarding your question about the direction, I'd welcome a solution w/o using virtual functions. Not because of performance considerations but because virtual base classes often yield to entangled designs. In my experience, such interfaces (which embody the often well-intended but overzealous attempt to generalize) tend to introduce ceremony and artificial terminology that is better avoided. Let us strive for maximum clarity, using specific types, as little indirections as possible, and no side effects (like the caching of the type check).

Following this line of thinking, it would be nice to consider separate types entirely and keep the distinction only at the `Vm_session_component`, very similar to what @atopia is already suggesting but without virtual interfaces. The `_component` could hold two factories one for each type (SVN, VMX). If the factories are small, there is no need to use a union. Repeated checks for the type can be avoided by hosting the check in a few private `_with_...` style methods. Note that I'm not entirely against a common base type but I'd keep at as bare-bones as possible.

I assume much of the code between the two types looking very similar. Let us try to factor out these similarities to library-like utility functions and types that can be reused individually by each types. In short, let's try to factor out similarities, not factoring out dissimilarities. This approach can keep the amount of duplicated code reasonably low, documents similarities where they exist, while not tapering over dis-similarities (by using most specific type as the default starting point).

atopia, 2024-11-11:
@nfeske please have a look at [05248a4](https://github.com/atopia/genode/commit/05248a4f4009e4097aaebd05af3ff598470c93d9) and at the inline comments in the commit on Github. While the commit may look like I just did the minimum to separate the two classes, I actually went full circle reuniting separate implementations.
The key issue for me was trying to contain the complexity of the common functionality, especially the `create_vcpu` method. Having different page table implementations led to keeping the `Kernel::Vm::Identity` member (`_id`) to the derived class, because it contains the physical address of the page table, which is only known once the virtualization technology specific page table has been constructed. However, keeping `_id` a separate member of the VMX/SVM class led to a less readable specific implementation of `create_vcpu` which would in turn call methods of the `Vm_session_component` base class for common tasks etc.
As you can see in the inline comments, the current state has the downside of having to modify the physical address of the nested page table in `_id` from the constructor of the derived class.
I'm aware this intermediate result is maybe not what you envisioned. I did attempt to split out common functionality by composition, but I ultimately found that it only led to clunky helper methods whereas especially `create_vcpu` needed to replicate common logic and basically touch all of the data members of the `Vm_session_component` across the different virtualization implementations.

Maybe you have a better idea for splitting this up that we can discuss here or in person.

nfeske, 2024-11-11:
I think the crux of the misunderstanding is that I'm pressing to abolish the common base class whereas your patch retains a *large* `Vm_session_component`. The only thing both classes `Vmx_session_component` and `Svm_session_component` should have in common is that each of them is compatible with the `Session_object<Vm_session>` interface.

Using inheritance for code reuse is a bad pattern. Shared code must have the form of utilities *aggregated* by the specific implementations. Those utilities can be templates. E.g., `create_vcpu` (should it be worth sharing) can be a template taking Vmx/Svm-specifics as template arguments.

As a minor nitpick, please let us shorten `Svm_vm_` to `Svm_`, and `Vmx_vm_` to `Vmx_` respectively.

atopia, 2024-11-11:
I was aware that you were pushing for composition over inheritance, I just didn't see a good way to do it. I did previously give the CRTP a thought but of course that wasn't solving my inheritance problem around `_id` (or any other problem for that matter, as my issue wasn't with the vtable). Thanks for the idea of using a template for the `create_cpu` logic. I will try to  use that for a more composition based approach.

atopia, 2024-11-14:
@nfeske please see 4f8c503 for a starting point to refactor the separated Vm_session_component  implementations.

nfeske, 2024-11-28:
@nfeske 2024-11-25:
> This is what I had in mind. It is good to see the use of C++ exceptions reduced as the new `Guest_ram` interface uses `Guest_ram_result`.
> You may consider moving this type renamed to `Attach_result` under the umbrella of `Guest_ram::` because that is the scope it is intended for.

> The return type of `table_phys_addr` should better not be a `void *`. The value is not meant to be de-referenced after all, or is it? It seems to be merely passed to `Kernel::Vm::Identity`.
> A dedicated type could clarify this relation, clarifying the rationale behind the relation between `table_phys_addr` and `Vm::Identity`.

> In my opinion, there is no need to move `Core::Guest_ram<TABLE>::attach` to a .cc file.
> I sense that you got the gist the approach I was pressing for. Do you see the next opportunities for de-duplication in the same style (factoring out commonalities into re-usable and self-explanatory utilities), or do you need my assistance to identify the next sensible step?

@atopia 2024-11-28:
> > This is what I had in mind. It is good to see the use of C++ exceptions reduced as the new `Guest_ram` interface uses `Guest_ram_result`.
> > You may consider moving this type renamed to `Attach_result` under the umbrella of `Guest_ram::` because that is the scope it is intended for.
> I did have something like that initially, but iirc that would require specifying a template instance when matching the result from the outside in `(Svm|Vmx)_session_component::attach()`.

> > The return type of `table_phys_addr` should better not be a `void *`. The value is not meant to be de-referenced after all, or is it? It seems to be merely passed to `Kernel::Vm::Identity`.
> > A dedicated type could clarify this relation, clarifying the rationale behind the relation between `table_phys_addr` and `Vm::Identity`.
> Sure. Since `Kernel::Vm::Identity` is used across architectures, I'll fix this once I expand the use of `Guest_ram` across architectures in the final version of that change.

> > In my opinion, there is no need to move `Core::Guest_ram<TABLE>::attach` to a .cc file.
> > I sense that you got the gist the approach I was pressing for. Do you see the next opportunities for de-duplication in the same style (factoring out commonalities into re-usable and self-explanatory utilities), or do you need my assistance to identify the next sensible step?
> I was planing to look into expanding the use of Guest_ram to the other architectures first (although part of the implementation complexity was shared across kernels in repos/base/src/core/vm_session_common.cc originally, so we'll see how sensible that really is).
> I'll have a look at what to factor out next and will let you know if I fail to identify a next opportunity.

@nfeske 2024-11-26:
> After sleeping over the `Guest_ram` interface, it occurred to me that the clarity of the code would potentially benefit from removing
> the _table member and passing it as auto &pt argument to the methods like attach that operate on it (and passing it along
> further to the private methods of course). This has several benefits:
> * The weird allocation/construction (via construct_at ) can stay outside of the otherwise clear-cut abstraction of `Guest_ram`.
> * The `Guest_ram` would become easier to subject to unit testing because the code becomes less frameworky and more library-like.
> * `Guest_ram` no longer needs to be a class template, making the code easier on the eyes with `<` and `>` noise removed.
> * Your remark about the complication of naming `Guest_ram::Attach_result` is addressed.

> I propose hosting the methods inside the Guest_ram class to making the code less scattered.

> Upon a little further reflection on the name Guest_ram I sensed that Guest_memory might actually be a more appropriate name because not all dataspaces mapped to the guest need to be RAM dataspaces. They may be ROM or even IO_MEM dataspaces.
> Speaking of `Guest_memory` would avoid the implied connection to the (otherwise nicely consistent) notion of RAM within core.

@atopia 2024-11-26:
> > After sleeping over the `Guest_ram` interface, it occurred to me that the clarity of the code would potentially benefit from removing
> > the _table member and passing it as auto &pt argument to the methods like attach that operate on it (and passing it along
> > further to the private methods of course). This has several benefits:
> > * The weird allocation/construction (via construct_at ) can stay outside of the otherwise clear-cut abstraction of `Guest_ram`.
> > * The `Guest_ram` would become easier to subject to unit testing because the code becomes less frameworky and more library-like.
> > * `Guest_ram` no longer needs to be a class template, making the code easier on the eyes with `<` and `>` noise removed.
> > * Your remark about the complication of naming `Guest_ram::Attach_result` is addressed.

> I attempted to implement your suggestion in 4479202, but I don't know how to get around the issue in line 263 of guest_memory.h, namely that ~Guest_memory() calls detach_at() , which needs to access table and table_array.
> And honestly while I sympathize with the idea of moving construction of table and table_array out of Guest_memory , I find the result to be utterly unreadable. The whole code just screams that table and table_array are integral to the task encapsulated in `Guest_memory`, and that they have no business in `(Svm|Vmx)_session_component` other than awkwardly being passed in [... cut ...]
> The ep is only needed for the ds-cap -> dsc lookup. The `Attached_dataspace takes` care of the core-local mapping, which is bound to the lifetime of ds.

> With this utility in place, you can host an instance of `Constructible<Phys_allocated<Ept>>` as a plain member variable.
> Why a `Constructible`? The allocation must be performed after the quota accounting. So the constructor of > `Svn_session_component` would do the `try_withdraw` as a precondition for calling `_ept.construct(ep, ram, rm, ...)`.
> The interaction with the ept conveniently becomes `_ept->obj.method(...)`.

atopia, 2024-11-28:
You're absolutely right. Thanks for your review comments, they are addressed by [6ba9241](https://github.com/atopia/genode/commit/6ba9241086fe71b78dd0e95119ccfa7f2c8e4904).

atopia, 2024-11-28:
[repost of https://github.com/atopia/genode/commit/0a1521cb514b0fca9eb5bc6645997a14375610b0#commitcomment-149702003 ]

Thanks for the many tips to tackle the use of `cma()`! I'm looking forward to straighten out this part of base-hw's VM support.

> Could you be more specific what you mean with "not satisfied"? Do you refer to the code repetition between detach and detach_at?

No, what I mean is that keeping the tables in `(Svm|Vmx)_session_component` requires us to implement the `Region_map_detach` interface at that level, which in turn leads to passing the instance of the session into `attach()`:

> https://github.com/atopia/genode/blob/3ee1851440bcb7e8836db56b5db5d217018d200e/repos/base-hw/src/core/spec/x86_64/virtualization/svm_session_component.cc#L171-L175

I don't have a better suggestion (other than going back to my original implementation and making guest page table handling part of `Guest_memory`) and with the current scope of `Guest_memory` it is just not equipped to implement the interface, but I would have preferred if `Region_map_detach` could have been encapsulated in `Guest_memory`, especially since we now need to pass it into `attach()`.

atopia, 2024-11-28:
@nfeske please add [3481d6e](https://github.com/atopia/genode/commit/3481d6e8254ed961b79f985b67e3a06cc7a02c79) to staging.

nfeske, 2024-11-28:
Merged with minor style adjustments in b716ef9594cf529b2990b13a9604d687766ca7cd

atopia, 2024-12-03:
Thanks to your extensive hints it has been easy enough to adapt `_table` to the proposed `Constructible<Phys_allocated<..>`, but I'm currently struggling to extend the idea to the `Vm_page_table_array`, because the constructor requires a lambda which will convert an internal virtual address to a physical. I have already consulted with Alex about it and I have reached out to Christian Helmuth (because both I can ask more easily via chat). Anyway since Alex wasn't able to help and I haven't heard back from Christian I might as well describe my problem here:
So as I mentioned `Vm_page_table_array` expects to learn about the physical address of an internal data structure via a lambda that is passed in the constructor. The problem arises since we also want to make it a constructible, so the backing memory is only acquired after the Ram quota has been withdrawn from the vmm. So the following won't work:
```
	_table_array.construct(_ep, _core_ram_alloc, _region_map, [&](void *) { return _table_array->phys_addr(); });
```
because obviously at this point `_table_array` hasn't been constructed yet. The obvious but ugly alternative would be to forego the idea of forwarding constructor arguments in `Phys_allocated` (neither the page tables nor other candidates have any) and pass the lambda from within `Phys_allocated` via a special constructor, but I was wondering if there was a way to get hold of a Dataspace capability at a given virtual address from the `(Svm|Vmx)_session_component` e.g. from the `Region_map`? I don't see a solution, but that wouldn't clutter `Phys_allocated` and be a *much* cleaner interface.

atopia, 2024-12-03:
to make the discussion easier, [51516a7](https://github.com/atopia/genode/commit/51516a7d8b4f0d6ba0204d881aa3bb70a0957d7d) provides a POC implementation that uses a special constructor to illustrate the feasible but not very clean solution.

atopia, 2024-12-03:
Thanks for your suggestions! If I understand proposal 1 correctly I could forego passing the lambda but would still need a special "pass in a physical address" constructor in `Phys_allocated`. Anyway, it seems that with a lot of the problems I'm running into I'm more hesitant to deviate from the previous common implementation of `Vm_root` and `Vm_session_component` and therefore go for local solutions instead of considering to create specialized implementations. I'll implement suggestion 2, eventually that line of work should be generalized to the other architectures anyway.

atopia, 2024-12-04:
I've quickly checked the previous "local" solution with the lambda inside `Phys_allocated` on VMX with my Virtualbox scenario and it turns out that the lambda that captures `_table_array` in fact **doesn't** cause the potential nullptr dereference. Honestly I haven't run the complex Virtualbox test in a while.
While I pin down the root cause I would appreciate opinions on the lambda:
On second thought I'm thinking that capturing the reference to the object under construction is pretty bad style. Then again it seems to technically work akin to accessing object members and methods from the body of the constructor, and it would keep the special requirements of `Vm_table_array()` out of `Phys_allocated` and save the need for a wrapper struct around `Vm_table_array` just to house the lambda.
After reconsidering what I did with the lambda, my analysis is that `_table_array` is not constructed yet, because the constructor hasn't finished, and therefore accessing it via the captured reference is pretty bad style and I really should implement another solution even though that will be less convenient in other places. At the same time the lambda should be guaranteed to work, because `_table_array`'s member variables are guaranteed to be initialized when the constructor body is run, and the capture-by-reference of `_table_array` in the lambda should therefore provide access to `_table_array`'s public methods and data members just like using them from the constructor's body. Any other opinions?

nfeske, 2024-12-05:
> The problem arises since we also want to make it a constructible [...]

So for the record, this assessment was misguided. The use of `Constructible` was not the problem. The problem is generally that the arguments of a constructor cannot depend on the object being constructed.

The `Phys_allocated` utility imposed this problem on you because it executes `construct_at` in the constructor body while merely forwarding the constructor arguments. Hence, `phys` cannot participate in the construction of `obj` regardless of whether in the form on additional argument or as part of a lambda body. It is not part of `args...` after all.

You can solve this knot by moving `construct_at` out of the `Phys_allocated` constructor to the caller site. Like so:

```
struct Phys_allocated
{
   ...
  /*
     modified constructor that replaces the forwarding of constructor args by a
     functor that performs the construction of `obj`. The trick is to expose `*this`
     to `construct_fn`. So `construct_fn` can get hold of `phys`.
  */
  Phys_allocated(Rpc_entrypoint &ep, Ram_allocator &ram, Region_map &local_rm, auto const &construct_fn)
  :
     _ep(ep), _ram(ram), _rm(rm)
  {
     construct_fn(*this, &obj);
  }
 ...
};

...

/* construction example */
using Table = Phys_allocated<Page_table_array>;
Table _table {
  _ep, _ram, _rm, [] (Table &table, auto *obj_ptr) {
    construct_at(obj_ptr, table.phys, any, arguments, you, like);
  })};
```

atopia, 2024-12-06:
I was hoping to publish the adapted version in [b7f5348](https://github.com/atopia/genode/commit/b7f5348de31d2d7ec8214619728879aa21495d4a) together with a fix for the runtime error on Virtualbox, but that has proven more elusive than I thougth. From what I gathered my test scenario breaks at the factored out [attach_addr](https://github.com/atopia/genode/blob/5f8bf0841cf2b0d6fd0a002a6456e40615352d2a/repos/base-hw/src/core/guest_memory.h#L98) (and it was already broken with the refactoring in dc3712ca5a6f6b77fcff4e03a2ad9af667a76033) but I don't yet understand how the refactoring introduced different runtime behavior.

nfeske, 2024-12-06:
The new version definitely gained clarity. It is good to see the use of manual `destroy` calls vanish and the proper accounting added.

> I don't yet understand how the refactoring introduced different runtime behavior.

Since you already have a suspicion, have you tried comparing the logged sequences of attach/detach_at operations in the good and bad cases? That is, logging the arguments and results?

If you cannot spot the issue by cross-correlating or by careful reviewing the patch regarding semantic changes, I'm afraid you will need to split the patch into multiple steps that can be individually tested.

atopia, 2024-12-10:
I've spotted the error: instead of passing a reference to the Constrained RAM Allocator into the `Guest_memory` constructor, I accidentally passed in a copy. Fixed by  [8df4007](https://github.com/atopia/genode/commit/8df400710b28fa7cd25c32c0cc209921f567d12b). This makes me wonder if there is any legitimate reason to copy-construct RAM allocators or if `Genode::Ram_allocator` should actually be noncopyable?

nfeske, 2024-12-10:
> wonder [...] if `Genode::Ram_allocator` should actually be noncopyable?

That's right. I'd appreciate a separate commit changing just that.

atopia, 2024-12-10:
[940c2db](https://github.com/atopia/genode/commit/940c2db98a205714a9b950c10710db73db193f3d) makes the `Ram_allocator` interface noncopyable.

atopia, 2024-12-10:
My [5221-vm-session-refactor](https://github.com/atopia/genode/tree/5221-vm-session-refactor) branch is the current state of the refactoring rebased on master and tested with my complex VMX Linux test scenario as well as with `run/vmm_x86` on qemu's SVM implementation. My next step will be converting the allocations in `Vcpu_data` to use the `Core_ram_allocator` as well.

nfeske, 2024-12-11:
Merged 9b0b0d8 to staging.

nfeske, 2024-12-11:
> My next step will be converting the allocations in Vcpu_data to use the Core_ram_allocator as well.

Thanks for the updated branch, which clearly gets into a good shape now.

I agree that the VCPU creation is good next point to focus on.

Two other potential improvements are
* The suggestion https://github.com/genodelabs/genode/issues/5221#issuecomment-2506033959
* The introduction of a `with_dataspace` utility function implementing the (identical) ceremonial parts of both `attach` implementations.

atopia, 2024-12-11:
I have tried to come up with a more generic way to deal with the x86-specific allocation of pages for virtualization technologies. I haven't come up a totally convincing cross-platform solution. If we end up generalizing the support classes for ARM, the obvious first step would be to untangle `Vcpu_data` from `Vcpu_state`, but I don't have a good idea for a solution that would allow the re-use of the `Vcpu` class on ARM. Possibly here the solution could as well be to factor out common bits (e.g. around `Vcpu_state`) in a common helper class used by the platform-specific `Vcpu` but maybe that is taking the composition pattern too far?

Anyway, for now I've pushed [09e260b](https://github.com/atopia/genode/commit/09e260b984f1a0f7a77ae9e46c96cf04baad092f) that improves on the situation on x86.

As a side note, just FYI @nfeske  despite changing the repository's settings today a notification for commits on atopia/genode was still sent to my private email account. The `(Svm|Vmx)_session_component` classes are starting to look comfortably empty, I will look into your other suggestions for further improvements tomorrow.

chelmuth, 2024-12-12:
Commit ccdbefd1581254f858f4c20e872b867f192122b1 is responsible for a number of build errors last night. @atopia please consult the CI emails and provide fixups.

atopia, 2024-12-12:
@chelmuth  Unless I have missed other implementations that need to be copy-constructible, cdc80a5 should fix the build errors (please disregard [7a7eb19](https://github.com/atopia/genode/commit/7a7eb19509655bf672448743ba9f45f8c0100919)).

nfeske, 2024-12-12:
Commit https://github.com/atopia/genode/commit/09e260b984f1a0f7a77ae9e46c96cf04baad092f looks very nice. It's good to see that the `Phys_allocated` utility fits in so nicely.

As far as I can see, `_vcpu_id_alloc` is merely used as an allocator for the elements of the fixed dimensioned `_vcpus` array. It may be worth considering replacing the array by the following:

```
Registry<Registered<Vcpu>> _vcpus { };
```

Creating a new `Vcpu`:
```
Vcpu &vcpu = *new (_constrained_md_alloc)
  Registered<Vcpu>(_vcpus, all, the, other, arguments);
```

Destroying all `Vcpu` objects at session destruction:
```
_vcpus.for_each([&] (Registered<Vcpu> &vcpu) {
  destroy(_constrained_md_alloc, &vcpu); });
```

This simplifies the code, removes the C-ish array (which are always risky), reduces the base line of session costs, and eliminates the need to define a `VCPU_MAX`.

Hope I'm not oversimplifying things. Maybe I fail to see the whole picture?

atopia, 2024-12-12:
No, that is a neat idea. You pretty much nailed it, commit [f3c20d3](https://github.com/atopia/genode/commit/f3c20d3b6521d66cc01630ffbfb126c9fcd7760d) moves the vCPUs to a Registry.

atopia, 2024-12-12:
>     * The suggestion [hw: clean up runtime page table switching for x86 virtualization #5221 (comment)](https://github.com/genodelabs/genode/issues/5221#issuecomment-2506033959)

I didn't comment on it, but I didn't implement that idea because I felt that because both the VM Session and the `Region_map_detach` interfaces depend on data members like `_memory` and `_table`, the readability of the code wouldn't benefit from a separate struct, and after considering it once more today, I still feel that way.
 
>     * The introduction of a `with_dataspace` utility function implementing the (identical) ceremonial parts of both `attach` implementations.

I've tried but as much as I'd want to reduce the shared code further, having to pass all the data members to such a utility function didn't seem to be worth obscuring relatively simple logic and the exceptions thrown. Or did I somehow misinterpret your proposal?

nfeske, 2024-12-13:
> I didn't comment on it, but I didn't implement that idea because I felt that because both the VM Session and the `Region_map_detach` interfaces depend on data members like `_memory` and `_table`.

The implementation of `detach_at` would stay in the `Svm_session_component` but merely in the form of a private `_detach_at` which is not a virtual function. The `Detach::detach_at` method (implementing the `Region_map_detach` interface) would come down to:

```
struct Detach : Region_map_detach
{
  Svm_session_component &_session;
  void detach_at(addr_t at) override { _session._detach_at(at); }
  ...
} _detach { *this };
```

Same for `unmap_region`.

It looks simple to me. More importantly, it eliminates the question of why `Svm_session_component` implements the `Region_map_detach` interface, because that's no longer needed.

> I've tried but as much as I'd want to reduce the shared code further, having to pass all the data members to such a utility function didn't seem to be worth [...]

I'm afraid that you misinterpreted my suggestion. I don't want to obscure anything. But a trivial utility like the following would make the code of `Svm_session_component::attach` more clear (less obscure):

```
void with_dataspace(Rpc_entrypoint &ep, Dataspace_capability cap,
                    auto const &fn, auto const &missing_fn)
{
  bool valid = false;
  if (cap.valid())
    ep.apply(cap, [&] (Dataspace_component *ptr) {
      if (ptr) {
        fn(*ptr);
        valid = true; }; });
  if (!valid)
    missing_fn();
}
```

That said, we don't need to go overboard here. I'm not insisting. It is probably more sensible to implement this idea in a wider scope independent from this issue, e.g., by replacing the current `Rpc_entrypoint::apply` with an `with_object` template method in the spirit of the utility sketched above. So let's keep the current implementation for now. I still invite you to try out the experiment of how the above `with_dataspace` would positively affect the clarity of `attach`. Just to get the hang of approaching *simplification* by introducing sensible (self-explanatory) *terminology* and then benefiting from using it.

> The (Svm|Vmx)_session_component classes are starting to look comfortably empty

I agree. Given how small the .cc files have become, I'd rather move the remaining code to the respective headers. This creates one holistic picture at one place. It also avoids the repeated (and thus uninteresting) stating of the long list of `Svn_sesson_component` constructor arguments (keeping an eye on the SLOC count). It also eases the comparison of `Svn/Vtx_session_component` side by side for further unification attempts.

atopia, 2024-12-19:
Hi @nfeske, [d59448d](https://github.com/atopia/genode/commit/d59448d4b69f59f11f329534512dabb395f748ce) moves the implementations into the header and [d59448d](https://github.com/atopia/genode/commit/d59448d4b69f59f11f329534512dabb395f748ce) moves the `Region_map_detach` interface to a member object.

nfeske, 2024-12-19:
I wonder, what is the current state of affairs? Are you working on further consolidation steps or do you consider the current state as satisfactory?

atopia, 2024-12-19:
If there aren't any more wishes for broad changes from your side I would consider the clean up done for the scope of x86. I was thinking to postpone any attempts to generalize the refactoring to ARM (where many of the issues e.g. with regard to core memory persist) until the x86 changes have passed review and to tackle that adaption (where sensible) in a follow-up issue where it should not need as much discussion.

nfeske, 2024-12-19:
To take on the review of the changes as a whole, please take the following steps.

1. Rebase your branch to staging.
2. Squash the commits into a single one that can be reviewed as a one entity.
3. Review the single commit by yourself, *repeatedly*, until you are happy with everything you see. Please pay attention to the following aspects:
   - Follow the coding style, keep an eye on the SLOC count, e.g., by putting arguments on a single line where it fits on 80 characters.
   - If code can be formatted in a tabular way, like the cases of a switch statement and corresponding single-line return values, lay out the code as a (kind of) table.
   - Consider inverting logic if this would make the code simpler.
   - Remove ceremony wherever it brings no benefit. E.g., the `Virt_type` enum is superficial. Better take the two branches for SVM and VMX using a plain `if` statement. Clarity should win over formality. Avoid repeated prefixing of names by using `enum` class with short names for the values.
   - Shun global variables, even if disguised as local static variables (looking at `virt_type`).
   - Make extra sure that all allocations are accounted to the session. That's the purpose of `Constrained_ram_allocator` after all. Don't keep a reference to the `core_ram` argument because is must never be used directly (because allocations are not accounted to the session when using `core_ram` directly).
   - Try to keep locally used names short and expressive. E.g., Within the `Virt_cpu` you can speak of `_ram` instead of `_contrained_md_ram_alloc`.
   - Ask yourself critical questions about your code. E.g., what happens with the priority argument passed to the sessions?
   - Follow the existing conventions. E.g., don't include headers using `"base/ram_allocator.h"` but via `<base/ram_allocator.h>`. Try to remove redundant includes if possible.
   - Don't print error messages that can be forcibly triggered by the untrusted VM client. E.g., the detaching of an invalid region in a loop must not set core under water. To aid debugging, you can consider printing messages up to an upper count.
   - As a matter of conduct, it would be fair to mention the original author of code in the file headers where appropriate. E.g., phys_allocated.h states you as the sole author whereby the code was almost verbatim dictated and explained by me. Give credits where credits are due. 
4. Review the whole commit again, constantly asking the questions: "Is this line really needed?", "Can this code be expressed even more clearly", and "What could Norman possibly argue about it?".
5. To double check that the cosmetic polishing above introduced no regressions, test the final version with a real VM.

atopia, 2025-01-07:
My [5221-vm-session-refactor-squashed](https://github.com/atopia/genode/tree/5221-vm-session-refactor-squashed) branch addresses your comments and should be ready for review, with the following remarks:

> * Make extra sure that all allocations are accounted to the session. That's the purpose of `Constrained_ram_allocator` after all. Don't keep a reference to the `core_ram` argument because is must [...]

The reference is necessary because the core ram allocator needs to be passed to the Vcpu constructor for per-CPU physically-addressable memory (e.g. the VMCB/VMCS). The core ram allocations are deducted from the VMM's quota before allocating the memory and VM/VCPU creation will fail before allocating (core) memory if the quota is not available from the VMM.

>        * Ask yourself critical questions about your code. E.g., what happens with the priority argument passed to the sessions?

As discussed offline, setting the priority of a vCPU thread isn't yet implemented on hw and requires adapting the kernel's vCPU thread class ("Vm").

>        * Follow the existing conventions. E.g., don't include headers using `"base/ram_allocator.h"` but via `<base/ram_allocator.h>`. Try to remove redundant includes if possible.

The local include path was incorrectly added by a LSP server and already removed in later commits, sorry for the noise.

nfeske, 2025-01-13:
Thanks for the squashed branch. Even though you left a few of my remarks unaddressed (i.e., naming, excessive vertical space), I merged the branch to staging now. I can refine those points myself.

> The reference is necessary because the core ram allocator needs to be passed to the Vcpu constructor for per-CPU physically-addressable memory

But I wondered why cannot `constrained_md_ram_alloc` be used for this purpose?

> As discussed offline, setting the priority of a vCPU thread isn't yet implemented on hw and requires adapting the kernel's vCPU thread class ("Vm").

Thanks for clarifying. From this information, I take that the extraction, clamping, and forwarding of the `priority` argument in vm_root.h does not have any effect. In my opinion, this code along with the priority vm-session constructor argument should better not exist.

atopia, 2025-01-13:
> But I wondered why cannot `constrained_md_ram_alloc` be used for this purpose?

Right, it came up at some point in the discussion but then I misunderstood the question to be about keeping the argument outside of passing it into `Phys_allocated` but somehow memorized that using `Core_ram_allocator`  there was generally necessary. I'm really sorry for dragging this on, commit [a71c979](https://github.com/atopia/genode/commit/a71c9797819153cd38e005a1b3c41a1a0461adbb) (tested in qemu for SVM and on hardware with VirtualBox) addresses your question.

> Thanks for clarifying. From this information, I take that the extraction, clamping, and forwarding of the `priority` argument in vm_root.h does not have any effect. In my opinion, this code along with the priority vm-session constructor argument should better not exist.

Instead of axing the code in question (which of course I can still do), I found it more productive to implement vCPUs in hw in commit [0f2d899](https://github.com/atopia/genode/commit/0f2d899fbe6167eecd89e0e8dfd5be80f9a5bfb8). Reworking `Vm` (kernel vCPU threads) was already on my list, e.g. the implementation around `Vcpu_data` would also need to be refactored. For now, this implementation needs support for another system call argument, added in [de135ff](https://github.com/atopia/genode/commit/de135ff30b99d25762ad8c3f088741697efa108b). I've tested that the priority implementation doesn't break the code across architectures to the best of my ability, but I haven't tested the actual effect of setting the vCPU thread priority yet.

nfeske, 2025-01-14:
I merged https://github.com/genodelabs/genode/commit/cea5a16abba798feb172ac37f8b4330de1c6e020 to staging.

> Instead of axing the code in question (which of course I can still do), I found it more productive to implement vCPU priorities in hw in commit.
> [...]
> but I haven't tested the actual effect of setting the vCPU thread priority yet.

I'm afraid that it won't have any effect with the current scheduler because the VCPU would require a quantum in addition to the priority. As this topic is closely related to the kernel scheduling work by @skalk and @jschlatow in the context of issue #5117, it is best to leave the passing of scheduling parameters to be addressed there, along with introducing a proper type instead of passing around a naked unsigned value.

Remember that the current issue is strictly about cleaning up. To wrap up the issue, please remove the misleading code and call it a day.

atopia, 2025-01-14:
Sure, [fa1c465](https://github.com/atopia/genode/commit/fa1c465b095a4a1480e182cb299383c9c48705a5) removes the unused vCPU priority support.

nfeske, 2025-01-14:
Thanks. I merged your fixup to staging.

nfeske, 2025-01-14:
That's already fixed on staging.

nfeske, 2025-01-30:
The changes entered the master branch today.