genodelabs/genode #5224: Mesa: Update to version 24.0.1

ssumpf, 2024-05-22:
In order to support newer GPU generations on Intel (Alder Lake+) it's time to update our three year old (21.0.0) port to current Mesa.

ssumpf, 2024-05-22:
@chelmuth: 9fe2e8a and 59ddb53 resample the first version of the Mesa update. I test compiled everything for x86_64, allwinner, and imx. I will do more in depth testing next and push possible fixup commits here.

ssumpf, 2024-05-22:
35232d4 removes accidentally committed recipe hashes from the Mesa update.

ssumpf, 2024-05-22:
Current testing status:

* x86_64: VBox6 works on Skylake and Tigerlake as expected, dhewm3 gets completely stuck at random times -> will start to investigate tomorrow
* PinePhone: glMark2 runs as expected
* i.MX8: glMark2 test will be the first thing tomorrow

ssumpf, 2024-05-23:
@chelmuth: 3b247c9 updates the port hash for Mesa.

nfeske, 2024-05-23:
Wow! And just perfectly timed for the upcoming release. :)

ssumpf, 2024-05-24:
@chelmuth: 45f0dc7 fixes dhewm3 (Skylake/Tigerlake). However I have encountered strange artifacts in VBox6 on Tigerlake only - investigating.

jschlatow, 2024-05-24:
@ssumpf I tried the commits on the ZimaBlade and am getting a few warnings before mesa_gears crashes with an illegal WRITE at 0x0.

```
[runtime -> mesa_gears] Warning: ignoring unsupported RTLD_GLOBAL in dlopen()
[runtime -> mesa_gears] libEGL warning: MESA-LOADER: failed to retrieve device information
[runtime -> mesa_gears] Warning: _device_gem_get_aperture_size: available_gtt_size (4KB) is not properly accounted
[runtime -> mesa_gears] 
[runtime -> mesa_gears] EGL_VERSION = 1.5
[core] Error: illegal WRITE at address 0x0 by pager_object: pd='init -> runtime -> mesa_gears' thread='ep' ip=0x11d4ef2
```

jschlatow, 2024-05-24:
The page fault appears to occur in `iris_fine_fence_init()`. I'm investigating. @ssumpf Let me know if you have any ideas already.

jschlatow, 2024-05-24:
> ...which calls `i915_bo_set_caching()` that fails for some reason.

If I'm not mistaken, the reason is that our libdrm does not implement `DRM_I915_GEM_SET_CACHING`. I have no clue how this should be implemented, though. I believe it is supposed to change the caching attributes in the GTT. I'll try a dummy implementation.

jschlatow, 2024-05-24:
@ssumpf  With commit 85239ab, mesa_gears and glmarks2 are spinning happily on the ZimaBlade again. Let me know if you have a better idea than this simplistic dummy implementation.

ssumpf, 2024-05-26:
> @ssumpf With commit [85239ab](https://github.com/genodelabs/genode/commit/85239aba41c3587f9936182138e030a519e009e4), mesa_gears and glmarks2 are spinning happily on the ZimaBlade again. Let me know if you have a better idea than this simplistic dummy implementation.

@jschlatow: I am fine with the commit. I tested it it nothing breaks here. The less we have to implement, the better because special cases can get out of control for GPUs.

ssumpf, 2024-05-26:
@jschlatow: I run into some errors when testing VirtualBox6 with 3D enabled and dhewm (Doom3). The first one is

```
[init -> drivers -> platform_drv] Error: Could not allocate DMA area of size: 16777216 alignment: 24 size with guard page: 16781312 total avail: 2136502272 (error: DENIED)
```
What happens here? Is the DMA allocator exhausted. dhewm requires quite a lot of DMA memory (~2GB), but the ~4GB in the allocator should suffice. 

The second one is rather rare but worrisome:
```
[init -> drivers -> platform_drv] Error: Uncaught exception of type 'Genode::Final_table<Intel::Level_1_descriptor<12u> >::Double_insertion'
```
Have you seen this one somewhere, yet?

jschlatow, 2024-05-27:
> What happens here? Is the DMA allocator exhausted. dhewm requires quite a lot of DMA memory (~2GB), but the ~4GB in the allocator should suffice.

Can you try setting `_use_guard_page` to false? I'm wondering whether this causes too much fragmentation. Does dhewm allocate large chunks of DMA buffers?

> Have you seen this one somewhere, yet?

Not yet. I'll have a look. I am probably going to get back to you with some debugging commits.

ssumpf, 2024-05-27:
> Can you try setting `_use_guard_page` to false? I'm wondering whether this causes too much fragmentation. Does dhewm allocate large chunks of DMA buffers?

`libdrm` allocates 16MB junks only, so you are probably right with the guard page + plus 24 bit alignment leading to heavy fragmentation. Will test it. 
> 
> > Have you seen this one somewhere, yet?
> 
> Not yet. I'll have a look. I am probably going to get back to you with some debugging commits.

First, I will have to find a way how to reproduce this because I only saw it once.

ssumpf, 2024-05-27:
@jschlatow: With `_use_guard_page { false }` it works. So, fragmentation. Thx.

jschlatow, 2024-05-27:
> @jschlatow: With `_use_guard_page { false }` it works. So, fragmentation. Thx.

 Should we lift the 4GB limit for the DMA allocator then?

jschlatow, 2024-05-27:
> > > Have you seen this one somewhere, yet?
> > 
> > 
> > Not yet. I'll have a look. I am probably going to get back to you with some debugging commits.
> 
> First, I will have to find a way how to reproduce this because I only saw it once.

I reviewed the code and came up with a theory how this could be triggered. The DMA allocator should actually prevent a double insertion (i.e. using the same address range for different DMA buffers). For this to work, the reserved memory regions are also passed to the DMA allocator, but only as soon as the corresponding device is acquired. Normally, the DMA allocator will log an error if any range from the reserved memory has already been allocated. However, I believe, the check in the DMA allocator is shadowed by the mechanisms that sets up the default mappings. This mechanism was introduced because the reserved memory regions must already be present in the IOMMU translation table before the device gets acquired (at least in case of the graphics controller). The mappings are therefore already inserted during initialisation of the platform driver. When a component allocates DMA buffers before it acquires a device that has any reserved memory specified, the DMA allocator might actually use any address from the reserved memory regions but only fail at the insertion into the translation table.

The solution would be to let the DMA allocator know of all known reserved memory regions in the very beginning. Yet, that's easier said than done. The reserved memory regions are evaluated whenever the devices ROM changes and I need to come up with a solution that passes every reserved memory region only once to the DMA allocator. I think this calls for a list model. I'd rather implement this after the release, though, as this is not a trivial change to the platform driver.

ssumpf, 2024-05-27:
> > @jschlatow: With `_use_guard_page { false }` it works. So, fragmentation. Thx.
> 
> Should we lift the 4GB limit for the DMA allocator then?

I think for now it is okay as is, since it only happens in case Mesa leaks tons of memory, which I plan not to allow ;)

ssumpf, 2024-05-27:
> I reviewed the code and came up with a theory how this could be triggered. The DMA allocator should actually prevent a double insertion (i.e. using the same address range for different DMA buffers). [...]

Okay, I think I can see the problem. But it happened only once during testing things which put a lot of pressure on the platform driver and the circumstances are not clear or reproducible right now. I would suggest to keep it in mind in case the error shows up again. If it does, I will save the scenario on a branch and we can go from there.

jschlatow, 2024-05-28:
@chelmuth Can you please merge 85239ab?

chelmuth, 2024-05-28:
Sorry, it seems I was instinctively waiting for @ssumpf explicit approvement ;-) merged to staging now.

ssumpf, 2024-05-28:
> Sorry, it seems I was instinctively waiting for @ssumpf explicit approvement ;-) merged to staging now.

@chelmuth: Approved this two days ago. Will write: Please merge to staging next time ;)

jschlatow, 2024-05-29:
> Okay, I think I can see the problem. But it happened only once during testing things which put a lot of pressure on the platform driver and the circumstances are not clear or reproducible right now. [...]

I can reproduce the error with a small artificial test case. I'm gonna move this into a separate issue.

nfeske, 2024-05-30:
Fixed in Genode 24.05

ssumpf, 2024-06-04:
@chelmuth: 561570b updates to Mesa 24.0.8, 1ad7353 fixes the VBox6 issue for Tiger Lake, 6f4884c removes Genode locks from libdrm and uses `phtread_mutex` instead, 70d057b is a long overdue cleanup, and d66def7 removes an error in case there is more than one syncobj present in the wait call. I have tested this on all supported platforms. Please consider for staging.

chelmuth, 2024-06-04:
Merged the series to staging.

nfeske, 2024-06-20:
@ssumpf, since all commits of this issue entered the master branch today, can this issue be closed?

ssumpf, 2024-06-20:
@nfeske: I will close it for now in case something shows up on Broadwell I can re-open it.

closed 2024-06-20 by ssumpf