genodelabs/genode #5226: sculpt: vbox6 fails to create file-system session if vm_fs is started after vbox6

jschlatow, 2024-05-22:
Testing my firefox-av@vbox6 package on Sculpt, the following issue popped up. When I route the "vm" file-system request to a vm_fs child and start the latter after adding firefox-av@vbox6, vbox6 fails with an Error complaining about failure to create the corresponding file-system session. Interestingly, this issue only seems to occur if vm_fs uses a persistent storage (I'm using the USB thumb drive) and not if it uses ram_fs.

In the working case, the vbox child of firefox-av@vbox6 is started and appears to waits until vm_fs is present. In the error case, the vbox child of firefox-av@vbox6 already produces the error output before runtime logs the start of the vm_fs child.

This is reproducible with the official Sculpt 24.04.1 image.

jschlatow, 2024-05-22:
This is also reproducible with a single vfs component instead of firefox-av@vbox6. In this case, the error also occurs when routing the file system to ram_fs via a chroot component.

jschlatow, 2024-05-22:
It looks as if the runtime config is updated twice. The first update does not contain the required chroot component, which gets added by the second update. I believe the Sculpt manager is actually supposed to skip the first update.

nfeske, 2024-05-22:
This looks like the sculpt manager fails to detect that the vbox component is an "unsatisfied" child until the vm_fs announces its service, and starts the vbox component prematurely. I'd follow the search term "unsatisfied" in both app/sculpt_manager and app/depot_deploy (included by the sculpt_manager).

jschlatow, 2024-05-23:
@nfeske I believe the issue is that the runtime state assumes a launched child as running as soon as the user enabled the launcher option. The vbox component becomes a "satisfied" child at this point already, however, the vm_fs is not immediately added to the runtime config because it is not yet configured.

nfeske, 2024-05-23:
Bin begeistert! Thanks a lot for getting to the bottom of it. I merged your fix to staging.

closed 2024-05-30 by chelmuth