genodelabs/genode #5227: Update framework APIs to C++20 function template syntax

nfeske, 2024-05-22:
Thanks to C++20, we can `auto` in many places where template types had to be declared manually. For example,
```
template <typename FN>
void with_payload(FN const &fn) const
```
becomes
```
void with_payload(auto const &fn) const
```

nfeske, 2024-05-22:
The commits https://github.com/nfeske/genode/commit/63ed8aa544d38a4f643197b4b353347f340982da and https://github.com/nfeske/genode/commit/7460bb6090f7fb990fb245640172c8db7248069b adjust the public headers in the base and os repositories.

chelmuth, 2024-05-24:
With this welcome change `-std=c++20` (or `-std=gnu++20`) becomes mandatory for API users. virtualbox6 could be adapted by adding `-fconcepts` to `CC_CXX_OPT_STD = -std=gnu++17`. Other overrides of `CC_CXX_OPT_STD` I found are in *iris.mk* and *stdcxx-\*.mk*, which should be noncritical.

chelmuth, 2024-05-27:
Please note that Qt applications which use the base API also log many warnings like the following now.

```
In file included from /data/depot/checker/api/base/2024-05-24/include/pd_session/pd_session.h:18,
                 from /data/depot/checker/api/base/2024-05-24/include/cpu_session/cpu_session.h:23:
/data/depot/checker/api/base/2024-05-24/include/util/attempt.h:56:29: warning: use of ‘auto’ in parameter declaration only available with ‘-std=c++20’ or ‘-fconcepts’
   56 |                 RET convert(auto const &access_fn, auto const &fail_fn) const
      |                             ^~~~
/data/depot/checker/api/base/2024-05-24/include/util/attempt.h:56:52: warning: use of ‘auto’ in parameter declaration only available with ‘-std=c++20’ or ‘-fconcepts’
   56 |                 RET convert(auto const &access_fn, auto const &fail_fn) const
      |                                                    ^~~~
/data/depot/checker/api/base/2024-05-24/include/util/attempt.h:62:34: warning: use of ‘auto’ in parameter declaration only available with ‘-std=c++20’ or ‘-fconcepts’
   62 |                 void with_result(auto const &access_fn, auto const &fail_fn) const
      |                                  ^~~~
/data/depot/checker/api/base/2024-05-24/include/util/attempt.h:62:57: warning: use of ‘auto’ in parameter declaration only available with ‘-std=c++20’ or ‘-fconcepts’
   62 |                 void with_result(auto const &access_fn, auto const &fail_fn) const
      |                                                         ^~~~
/data/depot/checker/api/base/2024-05-24/include/util/attempt.h:67:33: warning: use of ‘auto’ in parameter declaration only available with ‘-std=c++20’ or ‘-fconcepts’
   67 |                 void with_error(auto const &fail_fn) const
      |                                 ^~~~
```

chelmuth, 2024-05-27:
> Please note that Qt applications which use the base API also log many warnings like the following now.

@cproc could we use `-fconcepts` for Qt apps too? I expect the qt5 port has to be adapted, right?

cproc, 2024-05-27:
> > Please note that Qt applications which use the base API also log many warnings like the following now.
> 
> @cproc could we use `-fconcepts` for Qt apps too? I expect the qt5 port has to be adapted, right?

Commit 3fe14ca seems to do it.

closed 2024-05-30 by nfeske