genodelabs/genode #5230: Support to sign grub2 bootloader for shim usage

alex-ab, 2024-05-24:
During Hack'n'Hike #4989 we had a x86 notebook, where I (and others) did not manage to disable the UEFI secure boot feature in the boot menu, which excludes Sculpt to be testable easily. 

Even so a user may enroll its own UEFI keys (https://wiki.linuxquestions.org/wiki/How_to_use_Secure_Boot_with_your_own_keys), the step may not be suitable for all scenarios, e.g. if you still want to use other OSes and it can be problematic (see preparations chapter).

Another option is to use the shim bootloader, which is nowadays common in the Linux community. Shim is signed by the "trusted" CA. Shim itself has a feature called MOK (machine owner key), that permits the owner of a machine to add personal certificates which it en-trust in order to boot self build+signed bootloader or linux kernels, without wiping the "built in" en-trusted CA certificates. 

This option would give us the option, to ship Genode framework based scenarios, e.g. Sculpt OS, with a self created certificate and signed grub2 bootloader.

If the user is willing to en-trust the self-signed certificate (or the Genode Labs provided certificate) and install it as MOK, he would be able to boot a Sculpt scenario on secure-boot only systems.

alex-ab, 2024-05-24:
With the wip commit on a Ubuntu 22.04 installation, I get a bootable USB stick for T490 with secure boot enabled. Debian 12 does not work for me currently. Please try also other distros so we can get an idea how for this takes us.

The commit expects that you have installed the shim packages of your Linux host distribution. For Debian 12 it was for me
shim-signed and pesign.

The commit in the current form adds the option, to specify a x509 certificate, for either image/uefi or image/disk, e.g:

--include image/disk --image-disk_shim genode.cer
--include image/uefi --image-uefi_shim genode.cer

When invoked without existing certificate file, it will give you some basic hints to create a key and certificate (mainly using the efigenkey tool as proposed by the basic tutorials). If all is fine, the grub2 efi files will be signed and embedded to the generated Genode image.

When booted on a secure boot enabled machine, it will be denied, because the signing key is unknown. I attached a raw slide show, what you will have to do. Mainly looking for the certificate, which is on the USB stick, checking that all is fine, from your perspective and enroll/accept the certificate. Afterwards you reboot, and if all is fine, your USB stick should boot  ...

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

If you want to remove the self-sign certificate from the machine again, you will need a native Linux installation or live medium. First list your enrolled certificate (you will have to use sudo or have to be root):

mokutil --list-enrolled

Export the certificates into files

mokutil --export

You will  get several MOK-number.der files in the current directory.
Schedule the entries you want to remove, e.g.

mokutil --delete MOK-number.der

You will be requested to enter a new password, which you will require later to confirm the removal during reboot.
Check finally the certificates which are scheduled to be deleted with:

mokutil --list-delete

Finally, reboot.
You will be presented with the same blue screen of the key manager, where you have to acknowledge the removal.

nfeske, 2024-05-30:
Just noticed that the GRUB output has become quite noisy since this commit was merged to staging.

alex-ab, 2024-05-30:
I will craft a fixup.

alex-ab, 2024-05-31:
@nfeske: please add 4c9fa3c4ffcf1ef9cb5da546da97b71d6159b893 to staging

commit aaa18adf (2024-05-30): image: support shim usage for uefi and disk

Fixes #5230
commit a3a84b25 (2024-06-20): image: support shim usage for uefi and disk

Fixes #5230
closed 2024-06-20 by nfeske