genodelabs/genode #5232: platform: potential conflict between DMA buffers and reserved memory

jschlatow, 2024-05-29:
The DMA allocator takes care of preventing conflicts between reserved memory regions and DMA buffers. However, since a device's reserved memory regions are only registered at the DMA allocator when the device is acquired, DMA buffers may actually occupy reserved memory regions. The issue popped up in #5224:

> The DMA allocator should actually prevent a double insertion (i.e. using the same address range for different DMA buffers). For this to work, the reserved memory regions are also passed to the DMA allocator, but only as soon as the corresponding device is acquired. Normally, the DMA allocator will log an error if any range from the reserved memory has already been allocated. However, I believe, the check in the DMA allocator is shadowed by the mechanisms that sets up the default mappings. This mechanism was introduced because the reserved memory regions must already be present in the IOMMU translation table before the device gets acquired (at least in case of the graphics controller). The mappings are therefore already inserted during initialisation of the platform driver. When a component allocates DMA buffers before it acquires a device that has any reserved memory specified, the DMA allocator might actually use any address from the reserved memory regions but only fail at the insertion into the translation table.

> The solution would be to let the DMA allocator know of all known reserved memory regions in the very beginning. Yet, that's easier said than done. The reserved memory regions are evaluated whenever the devices ROM changes and I need to come up with a solution that passes every reserved memory region only once to the DMA allocator. I think this calls for a list model. I'd rather implement this after the release, though, as this is not a trivial change to the platform driver.

jschlatow, 2024-06-10:
Commit 4cfef99 fixes the issue. Commit bfc6850 further makes use of the added interface to omit calls to `add_default_mapping()` for already added memory regions.

chelmuth, 2024-06-11:
@skalk I'm fine with the commits. Do you confirm the merge?

skalk, 2024-06-12:
@chelmuth: I'm fine with the commits. Thank you @jschlatow.

chelmuth, 2024-06-12:
Great, merged to staging.

commit 55d3e823 (2024-06-12): platform: handle reserved memory on devices update

Reserved memory regions must be excluded from the corresponding DMA
allocators irrespective of whether the device is in use. Otherwise, an
early allocation of DMA buffers may use the reserved memory regions of a
late acquired device.

Fixes #5232
closed 2024-06-20 by nfeske