genodelabs/genode #5234: acpica: failed IO_MEM session on t490s and base-hw

jschlatow, 2024-05-30:
Moving to base-hw on my t490s, I noticed that acpica logs the following error. I cannot reproduce this on the x260, though. @alex-ab Do you have any idea?

```
[runtime -> acpi_support] detected - ec      
[runtime -> acpi_support] Error: IO_MEM-session creation failed (label="", ram_quota=6K, cap_quota=3, base=0x10bf00000, size=0x1000, wc=no)
[runtime -> acpi_support] failed   - sleep button registration - error=22
[runtime -> acpi_support] detected - ac
[runtime -> acpi_support] online   - ac (0)
[runtime -> acpi_support] detected - battery 'BAT0' type='LiP' OEM='SMP' state=0x1f model=' 1179' serial='5B10W13910'
[runtime -> acpi_support] detected - lid
[runtime -> acpi_support] closed   - lid (0)
[core] Error: I/O memory [000000010bf00000,000000010bf01000) used by RAM allocator
[core] Error: Local MMIO mapping failed!
```

chelmuth, 2024-05-30:
Reincarnation of #4643 on base-hw. We just removed this page from the RAM allocator...

jschlatow, 2024-05-30:
@chelmuth Thanks for the reminder. Fixed by 2ce8757

alex-ab, 2024-06-11:
@chelmuth, @nfeske: branch issue_5234 contains the commits to move the quirk from core to acpica. It is more or less a very limited "emulation".

chelmuth, 2024-06-12:
> @chelmuth: commit [4d444ec](https://github.com/genodelabs/genode/commit/4d444ec61a86282af302a1e6b338625569a466bf) should be reverted on staging

Thanks for the reminder. Reverted on staging.

commit bff2b63a (2024-06-11): acpica: avoid special mmio address request

Add special handling for root bridge access for

12th Gen Framework laptop and Thinkpad X1 Nano Gen2

to avoid bogus MMIO memory accesses as seen in #5234 and #4643.

Fixes #5234
closed 2024-06-20 by nfeske