genodelabs/genode #5239: Make util/geometry.h C++20 friendly

nfeske, 2024-06-06:
With C++20, the `Point`, `Area`, and `Rect` types could better be expressed as compound types. Since the most common way defining rectangles is the use of a `Point` and an `Area`, the internal representation should change from two points `p1` and `p2` (as of today) to one point `at` and an `area`. So the construction of the compound remains natural, and a default-initialized `Rect` is invalid (because a default-initialized `Area` is invalid). With the switch to a compound type, `Rect` can then also be constructed via `{ .at = ..., .area = { .w = ..., .h = ...} }`. For the case where a rectangle is constructed by two points, a new `Rect::compound(p1, p2)` class function can be used.

While being at it, the types could benefit from a bit of renovation like replacing out parameters of `cut` by a compound return type.

To make the types applicable in base, e.g., for affinity.h, we should relocate the header from os/include/util to base/include/util/.

nfeske, 2024-06-06:
@cproc could you help out with adapting the Qt5 port to this change? I think it mostly comes down to removing the `()` pairs when accessing `Area::w` and `Area::h`.

cproc, 2024-06-07:
I'm having some trouble with the Qt5 adaptation, for example with the following error:

```
/.../contrib/qt5-0bf1a25e14bf04d6b6d8c4ecca3f611a8a9f364d/src/lib/qt5/qtbase/src/plugins/platforms/genode/qgenodeplatformwindow.cpp: In member function 'virtual void QGenodePlatformWindow::setVisible(bool)':
/.../contrib/qt5-0bf1a25e14bf04d6b6d8c4ecca3f611a8a9f364d/src/lib/qt5/qtbase/src/plugins/platforms/genode/qgenodeplatformwindow.cpp:640:68: error: no matching function for call to 'Genode::Point<int>::Point(int, int)'
  640 |                         Gui::Rect::compound(Gui::Point(g.x(), g.y()),
      |                                                                    ^
/.../repos/base/include/util/geometry.h:33:16: note: candidate: 'constexpr Genode::Point<int>::Point()'
   33 | struct Genode::Point
      |                ^~~~~
/.../repos/base/include/util/geometry.h:33:16: note:   candidate expects 0 arguments, 2 provided
/.../repos/base/include/util/geometry.h:33:16: note: candidate: 'constexpr Genode::Point<int>::Point(const Genode::Point<int>&)'
/.../repos/base/include/util/geometry.h:33:16: note:   candidate expects 1 argument, 2 provided
/.../repos/base/include/util/geometry.h:33:16: note: candidate: 'constexpr Genode::Point<int>::Point(Genode::Point<int>&&)'
/.../repos/base/include/util/geometry.h:33:16: note:   candidate expects 1 argument, 2 provided
```

nfeske, 2024-06-07:
I guess this problem stems from the fact that C++20 is not enabled for Qt5.

Can you try using `Point{...}` instead of `Point(...)`?

cproc, 2024-06-07:
> Can you try using `Point{...}` instead of `Point(...)`?

Yes, that worked. But I had to make similar changes in `base/include/util/geometry.h`, `os/include/os/surface.h` and `os/include/input/event.h` as well. I'll try to enable C++20 for Qt5.

cproc, 2024-06-08:
Commit 5f43ae4 adapts Qt5 to the API change. Enabling C++20 for whole Qt caused multiple errors, but enabling it for the QPA plugin worked fine. Also, the `-fconcepts` addition for Qt5 from issue #5227 caused errors with QtWebEngine, so I reverted that change and enabled C++20 for the API-using components instead.

cproc, 2024-06-08:
Commit 01d9c9c adapts qt_avplay.

commit 8c466161 (2024-06-06): Make util/geometry.h C++20 friendly

- Move header to base/include to make it applicable for base types
  like 'Affinity' down the road.
- Represent 'Rect' as typle of point and area, which is the most
  common form of initialization, creates in valid 'Rect' by default.
- Turn Point, Area, and Rect into compound types, making x, y, w, h, at,
  area accessible without a method call
- 'Rect::Compound' function for constructing a 'Rect' from two points,
  replacing a former constructor
- Use result type 'Rect::Cut_remainder' instead of out parameters.

Fixes #5239
commit 72dc8592 (2024-10-04): menu_view: restore horizontal cursor positioning

This is a follow-up fix for "Make util/geometry.h C++20 friendly".

Fixes #5239
Issue #5356
closed 2024-06-20 by nfeske