genodelabs/genode #5240: sculpt/hw: noticeable performance degradation with increased uptime in virtualbox

jschlatow, 2024-06-06:
After almost a week of running Sculpt 24.04.1 with base-hw on my T490s, I repeatedly observed the following effect:

- When running Sculpt for a couple of hours (approx. 3h), I do not get any key repeat events in my Linux VM anymore.
- The key repeat events do still work in the Inspect view though.
- The Linux VM also becomes noticeably slow. I particularly notice this when there is some bulk network traffic, such as `goa run --target sculpt`, which uploads a bunch of binaries. This works flawlessly on a freshly booted system. However, once I lost the key repeat events, the upload is slower and occasionally makes my VM unusable for a few seconds.

jschlatow, 2024-06-06:
I just noticed that I had accidentally created this issue in the Goa repository and therefore transferred the issue.

ssumpf, 2024-06-06:
@jschlatow: I actually can confirm this. A hard restart in Leitzentrale helps. So it looks like a VBox6 problem or in one of its dependencies. As you stated: The problem requires a couple of hours to appear.

ssumpf, 2024-06-06:
As an additional information: I don't observe any system load in `top_view` nor CPU frequencies going up in `msr_gui` where the remain at 800MHz or below.

nfeske, 2024-06-07:
> I instrumented the method as you suggested. The counter overflowed once, [...]

That is very worrying because this can have all kinds of effects, ranging from the missed (or hours late) firing of timeouts to the wrapping of the values returned by `Timer::Session::elapsed_us`, which is unexpected by clients. The continuity of kernel time is an ultimate assumption. Therefore, before investigating the problem anywhere else, we need to eliminate the wrapping of the time value reported by the kernel.

In my understanding, the kernel tick counter is a 64-bit value that counts at a rate of around 2 GHz. These values are converted to microseconds (`timer_ticks_to_us`) using a calibration value `ticks_per_ms`. This conversion must anticipate an uptime longer than a few hours, which is seemingly not the case right now.

I'll try to reproduce the wrap-around using a synthetic test.

jschlatow, 2024-06-07:
@nfeske I haven't written down the counter values but I just remembered that I noticed the values were very close. Thinking about this, it didn't seem like a typical overflow situation.

chelmuth, 2024-06-07:
By brute force testing I discovered  that `timer_ticks_to_us()` does not always yield increasing results for increasing `ticks` values, for example with `0x4ffffffff`, `0x500000000`, and `500000001` as input the clock ticks backwards about 588 us. As I was not able to fix the current implementation up to now, you may just replace it by a simple `return ticks * 1000 / ticks_per_ms` for testing.

chelmuth, 2024-06-11:
@jschlatow does the suggested change above or 6b6f31190c7089ff471599b5f5ef6eeecf1ec226 have positive impact on this issue?

jschlatow, 2024-06-12:
@chelmuth I'm still in the process of testing the fix since the issue is a bit sporadic. Without 6b6f311, I noticed that the degradation occurred with and without a preceding timer overflow and, vice versa, I have seen timer overflows without noticing any degradation. I'll keep you updated.

jschlatow, 2024-11-27:
The effect is still reproducible with Sculpt 24.10.2.

jschlatow, 2025-08-12:
Apparently, the problem is only triggered when a) the VM uses multiple cores and b) the first core used by the VM is the one that also runs the timer component. One notable difference is that, on hw, the timer component calls the kernel to get the current time, which can take a bit. This might mess with potential TSC synchronisation in vbox. Yet, I have not been able to locate the exact point in vbox that causes it to trip. Moreover, I am quite puzzled why it does not recover by restarting the vbox component.

Note that commit 616324f implements a TSC sync scenario that demonstrates the interference with hw's timer component. It also shows that on NOVA and hw there is no time drift between the TSCs of the different CPU cores.

jschlatow, 2025-08-14:
There are two more things I noticed:

1. I am able to "recover" from the problem by changing the affinity assignment of the vbox component so that it does not share a CPU core with the time component.
2. I am also able to trigger the problem by enabling extensive logging (irrespective of the affinity assignment). Hence this is potentially related to a high frequency of preemptions/kernel calls. Interestingly, I can recover by reducing the logging details.

When the problem occurs, vbox is unable to keep the virtual clock at sync for too long and doe not try to recover from that. It occurs before the VMM switches its TSC emulation mode to real TSC (with offset).

jschlatow, 2025-08-15:
> 2\. I am also able to trigger the problem by enabling extensive logging [...]

This is also reproducible on base-nova.

If my understanding is correct, the failure to sync the virtual clock is caused by a too big jump in the virtual timestamp. It's still a mystery, however, why this occurs and, in particular, why this only occurs on base-hw when the VM has been running for some time. It is unrelated to the uptime of Sculpt, though.

chelmuth, 2025-09-17:
So, does the effect disappear if you add `<libc> <pthread placement="single-cpu"/> </libc>` to the test?

On base-hw, the timeout framework uses the following implementation for `_timestamp()`.

```
Trace::Timestamp Timer::Connection::_timestamp()
{
	return (Trace::Timestamp)Kernel::time();
}
```

For testing, you may try to replace *hw/timer_connection_timestamp.cc* by the generic *timer_connection_timestamp.cc* in *base-hw/lib/mk/timeout-hw.mk*.

jschlatow, 2025-09-17:
> So, does the effect disappear if you add `<libc> <pthread placement="single-cpu"/> </libc>` to the test?

@chelmuth 
The effect disappears indeed with single-cpu placement (irrespective of affinity assignment). I also played around with the _all-cpus_ and _manual_ settings and different affinity assignments. As soon as cpu 0.0 is in the component's affinity (and thus the main thread is running on this core), the effect is present. In all other configurations, the effect disappears.

> On base-hw, the timeout framework uses the following implementation for `_timestamp()`.
>
> Trace::Timestamp Timer::Connection::_timestamp()
> {
> 	return (Trace::Timestamp)Kernel::time();
> }
> For testing, [...]

With the generic implementation, the effect disappears as well (also validated with my vbox test case).

chelmuth, 2025-09-18:
> I must admit that I am puzzled why the base-hw implementation of `_timestamp()` uses `Kernel::time()` at all. What is the point of interpolating between `elapsed_us()`, which is derived from `Kernel::time()`, and `_timestamp() = Kernel::time()`?

That's only one puzzling piece of the timeout-framework puzzle. The revamp of the implementation is on our agenda for years but up to now we only accomplished the first step by decoupling the timer implementations from the framework.

jschlatow, 2025-09-18:
@skalk I've looked into the base-hw kernel timer implementations. For x86, it uses the Local APIC timer in one-shot mode. While this is reasonable for scheduling, it is not suitable for establishing a reliable time source. This explains the time drift on x86. All other platform use a continuously counting timer as a time source. Now, I'm wondering how to fix `Kernel::time()` on base-hw as I see two options here:

1. Use the Local APIC timer in TSC deadline mode. AFAIK, this has been introduced by Intel with Sandybridge. Yet, I'm not sure whether this is supported by all newer CPUs.
2. Keep using the one-shot mode but use the TSC for kernel time calculation (basically `_time` and `_duration()`) so that `Kernel::time()` will be a derivative of the TSC.

chelmuth, 2025-09-19:
base-hw/x86: APIC timer implementation may loose ticks #5144

chelmuth, 2025-09-19:
> 1. Use the Local APIC timer in TSC deadline mode. AFAIK, this has been introduced by Intel with Sandybridge. Yet, I'm not sure whether this is supported by all newer CPUs.

Intel CPUs since Sandy Bridge support CPUID.01H:ECX.TSC_Deadline[bit 24]. Unfortunately, Qemu does support APIC tsc-deadline only in KVM mode but not with `-accel tcg` (which is the setting when running in VirtualBox on Sculpt). Also, the precious x201 does not support this mode because it's Arrandale, the predecessor of Sandy Bridge.

skalk, 2025-09-19:
@jschlatow thank you for your impressive investigation. It's quite unfortunate that this shortcoming was already noticed before, but wasn't fixed yet. I feel part of responsible therefore.
When summarizing the discussion until now correctly, we should immediatedly change time measurement in the kernel to use the TSC, but keep the Local APIC one-shot mode as timeout device. Do you (@all) agree? Without being too much in that whole x86 architectural details, I wonder whether TSC not being serialized in between different cpu cores will only lower the observed drift, or whether this is an academical discussion.

jschlatow, 2025-09-19:
> Without being too much in that whole x86 architectural details, I wonder whether TSC not being serialized in between different cpu cores will only lower the observed drift, or whether this is an academical discussion.

In my earlier tests (commit 616324f), I haven't seen any TSC drift. This does not mean, though, that there could not be any drift in certain scenarios (e.g. temperature-related drift). However, I'd assume that TSC drift is rather uncommon and therefore go for simple detection mechanism (implemented in the timeout framework) to warn if there any significant drift pops up.

skalk, 2025-09-19:
@jschlatow very cool! I'm fine with the kernel commit.

chelmuth, 2025-09-25:
> I also added a (hw-specific) test case with commit [b1afee6](https://github.com/genodelabs/genode/commit/b1afee65a671ec08d7e3581bd4dcf44ca57e0a47).

Last night, the test failed on qemu/pbxa9. Could (or should) the test be disabled on Qemu?

skalk, 2025-09-30:
The test fails on arbitrary non-SMP platforms with:
```
 [init -> test-kernel_time] Error: Test requires SMP
 [init] child "test-kernel_time" exited with exit value 1

 Line 0 of output is unexpected
 expected: '[init] child "test-kernel_time" exited with exit value 0'
 got:      '[init] child "test-kernel_time" exited with exit value 1'
Error: Test failed, 1 unexpected lines of output
```
namely on: imx53_qsb, imx53_qsb_tz, x86_64 on qemu, rpi. Commit 3aa6ce9 fixes that.