genodelabs/genode #5242: Tighten `Gui::Session` interface

nfeske, 2024-06-10:
- Avoid the use of optional parameter by splitting `create_child_view` from `create_view` as a separate RPC function.
- Replace use of ancient `Handle_registry` by `Id_space`. Speak of view IDs instead of handles.
- Allocate view IDs at the client side.

nfeske, 2024-08-14:
I merged the commit series for revisiting the `Gui::Session` interface to staging today.

The commits adjust all components in the Genode repository. In https://github.com/genodelabs/genode-world/commit/31620c775feebb31c0190bef5230d06e988fd954 I adjusted the components in the world repository as well.

As a note of caution, even though I did my best with thoroughly testing the changes, there remain a few blind spots. I would particularly appreciate feedback about virtualbox5, virtualbox6, mesa_demos, usb_webcam, glmark2, qt_avplay, and the ARM vmm.

@cproc Could you adapt Qt5 (and the forthcoming Qt6) to these changes? In order to do that, you may first skim over the commit messages prefixed with "gui_session:" to get the broad picture of the changes, and then take a look at the adaptation commit in genode-world (that applies the changes at once). I hope that you won't run into difficulties. Should you run into uncertainties or problems, please give me a note.

nfeske, 2024-08-20:
Super cool, thanks a ton, @cproc! Merged to staging.

nfeske, 2024-08-29:
Fixed in master.

alex-ab, 2024-09-02:
@nfeske: attached the abstract of an "evil" user (by accident I trigged it)

61 .   .   Gui::View_id const               _view         { 1 };                     
62 .   .   Gui::View_id const               _view_text { 1 };   <- this is obviously an error

147 .   .   .   log("first view");
148 .   .   .  _gui.view(_view,           { .title = { }, .rect = r_all,   .front = true  });
149 .   .   .   log("second view");
150 .   .   .   _gui.view(_view_text,  { .title = { }, .rect = r_view,  .front = false });
151 .   .   .   log("third view");

execution of 150 leads to:

```
[init -> graph] first view
[init -> graph] second view
[init -> wm -> wm] Error: Uncaught exception of type 'Genode::Id_space<Gui::View_ref>::Conflicting_id'
[init -> wm -> wm] Warning: abort called - thread: ep
```

[wm - entry](https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/wm/gui.h#L985)
[wm - unchecked use of ID](https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/wm/gui.h#L993)
[wm - exception caused](https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/wm/gui.h#L953)

nfeske, 2024-09-02:
Thanks @alex-ab for the excellent report, which allowed my to pin point the problem straight away!

May you give the presumed fix ce0ef5c4d9fe36281b8ba222ff40e84bcd42bcad a try?

alex-ab, 2024-09-03:
@nfeske: the commit helps, I don't get an exception anymore :+1:

nfeske, 2024-09-03:
Thanks for the feedback. Merged the fix to staging now.