genodelabs/genode #5245: Reduce reliance on C++ exceptions in the base framework

nfeske, 2024-06-11:
Down the road, I'd like to clear the base framework from the use of C++ exceptions, using the attempt pattern for error handling instead (see https://genode.org/documentation/genode-foundations/24.05/api/Fundamental_types.html#Exception-less_error_handling). Let's move gradually to this direction.

nfeske, 2024-06-12:
Commit https://github.com/genodelabs/genode/commit/a82a3e66910a2bdb3e506ab720c3039fe6d91ccf and fixup https://github.com/genodelabs/genode/commit/6af5a3e8871307025612440a54519d6bf45a18fb remove C++ exceptions from core's TRACE session interface and adjust the components throughout Genode's source tree. The new version of the interface is tighter in several respects. Hence, this change requires the adaptation of trace clients (@alex-ab thinking of top_view).

- The split between client.h and connection.h has been removed. Note that error types between the RPC functions and the corresponding `Connection` methods differ because the `Connection` handles the `OUT_OF_CAPS` and `OUT_OF_RAM` conditions.
- Basic types have been replaced by structs
- The notion of invalid policy IDs does not exist anymore. If a policy ID cannot be created, the `alloc_policy` operation reflects this condition explicitly via an `Alloc_policy_error`. At the clients, the former instances of `Policy_id` (which could be invalid) are now mapped to `Trace::Connection::Alloc_policy_result` now.

@jschlatow I have adapted the trace recorder to this change but admittedly haven't tested it. I'd appreciate you giving it a spin.

nfeske, 2024-06-14:
Commit https://github.com/genodelabs/genode/commit/7cac494dc2e05ab9f4b14790b75f80be12808b96 takes another step by revisiting the PD session interface.

jschlatow, 2024-06-14:
> @jschlatow I have adapted the trace recorder to this change but admittedly haven't tested it. I'd appreciate you giving it a spin.

It's spinning happily.

nfeske, 2024-06-24:
Commit https://github.com/genodelabs/genode/commit/2d77d43f89f184426b5a5efdf9bf9fd8974c6037 covers the CPU session interface.
Commit https://github.com/genodelabs/genode/commit/6a044c72042a9f941874adb7b8ad4d6f0fee6e7c is concerned with the `Region_map` interface. This one is by far the most elaborate step so far because of the many users of the interface. While being at it, I took the opportunity to modernize several further aspects of it, in particular addressing the requirement #5070 and passing attach arguments via an attribute structure, which makes the intent of client-side code more obvious.

While adapting the API users, I spotted a number of places where we could live with read-only mappings but currently establish read-writeable mappings, for the mundane reason that `rm.attach`used rw by default when called with only the dataspace capability as argument. For now, I kept the existing semantics. But as subsequent step, we may consider grepping the code for `"\.writeable *= *true"` and individually reviewing the places.

nfeske, 2024-06-25:
@cproc may you take care of adjusting qt5 to the latest `Region_map` change?

cproc, 2024-06-25:
Commit f89482c adapts qt5.

nfeske, 2024-06-25:
Fantastic! Thank you @cproc.

nfeske, 2024-07-01:
Commit https://github.com/genodelabs/genode/commit/3c00a1b8a0e24de3417397f568f6b879767d884e removes exceptions from the parent interface.
Commit https://github.com/genodelabs/genode/commit/87952dcaee22cf1740e9ddc562184dd64d9218ff revisits the child creation (ELF loading, process starting) in this respect.

nfeske, 2025-03-31:
Commit https://github.com/genodelabs/genode/commit/b444f0849c3085fb0452fc47a020e1712d70ef44 takes a preparatory step towards the removal of exceptions from base/thread.h.

nfeske, 2025-05-16:
@alex-ab my merge_to_staging branch contains the final batch of commits to clear the base framework and core from exceptions. Commit https://github.com/nfeske/genode/commit/507139ed4e7e0b10052a4e70d73a0d223f38ea04 (`Bit_allocator`, `Bit_array`) is expected have ripple effects throughout the code base. Your help is very welcome.

alex-ab, 2025-05-16:
@nfeske: going to look into base-sel4 now

cnuke, 2025-05-16:
Commit [67f6d6b](https://github.com/cnuke/genode/commit/67f6d6b51999b3b6fcfe21596ce4e35d57d1e903) adapts `driver/nvme` but is only compile-tested for now as the base of the platforms where it can be used cannot be compiled successfully yet.

nfeske, 2025-05-16:
@cnuke super! Thank you.

nfeske, 2025-05-16:
https://github.com/nfeske/genode/commit/7bfd95609c3259c68d59b32803ea0b8c1ccae451 adjusts base-foc.

nfeske, 2025-05-16:
base-hw has been reanimated just now.

nfeske, 2025-05-16:
With https://github.com/nfeske/genode/commit/288d994785ff2532c2f5b73ef2486caf0da61473, base-nova works again.

alex-ab, 2025-05-16:
With 8211bb2090fae91ddbdf8e0bf456b6468aa65613 base-sel4 works again, (but still a lot of exception in the base-sel4 part left)

nfeske, 2025-05-16:
@alex-ab excellent - thank you very much!

nfeske, 2025-05-16:
Commit https://github.com/nfeske/genode/commit/13497a2ad723808b9a31f48da86402f5d1c01b8e fixes the build of the platform driver. @skalk I'm not 100% sure about the shared-IRQ logic. May you give this change a close look?

Commit https://github.com/nfeske/genode/commit/f5508cef9f6b60588384b85899e68754ca19c6b7 fixes the build if the Intel GPU driver. @ssumpf could you please review those changes?

cnuke, 2025-05-16:
@nfeske 

* [11fbadb](https://github.com/cnuke/genode/commit/11fbadb95986cfd2739609ef76a2e3b9566146bc) fixes the fixup of `driver/nvme` (now that I could take it for a test-drive…)
* [f3cdf07](https://github.com/cnuke/genode/commit/f3cdf07ff67851debe12bd4c65b48e585b793b31) addresses an uncaught exception in `driver/ahci`
* [2a35c6e](https://github.com/cnuke/genode/commit/2a35c6eab594edc7a88884cd20af6d1f6a959e39) addresses the `sub_node` usage in `driver/acpi` (where you already beat me to the punch)
* [6617de1](https://github.com/cnuke/genode/commit/6617de1d0289f3bc595906eb4643411c3a8bcdf4) addresses the `sub_node` usage in `driver/nvme`

While testing `driver/nvme` as well as `driver/ahci` I noticed that massively had to increase the `platform` driver's cap quota.

nfeske, 2025-05-16:
Thanks @cnuke. I also happened to adapt to the platform driver. Tested so far as running sculpt_test with NOVA on Qemu.

A few known remaining issues notwithstanding (e.g., test-trace is currently not working), I merged the merge_to_staging branch to staging now so we can jointly stabilize the branch in the coming week.

ssumpf, 2025-05-16:
> Commit [nfeske@f5508ce](https://github.com/nfeske/genode/commit/f5508cef9f6b60588384b85899e68754ca19c6b7) fixes the build if the Intel GPU driver. [@ssumpf](https://github.com/ssumpf) could you please review those changes?

@nfeske: With [fe39e7c](https://github.com/ssumpf/genode/commit/fe39e7ca2b7532317383b15cfde80b8663a68a66) applied I was able to run glmark2 on modular Sculpt :+1:

nfeske, 2025-05-18:
@ssumpf thanks for for fixup and the fantastic news! :)

skalk, 2025-05-19:
> Commit [nfeske@13497a2](https://github.com/nfeske/genode/commit/13497a2ad723808b9a31f48da86402f5d1c01b8e) fixes the build of the platform driver. [@skalk](https://github.com/skalk) I'm not 100% sure about the shared-IRQ logic. May you give this change a close look?
> 

Yes, I did. It is semantically equivalent. I think there is some room left to simplify those loops a bit, but it is correct so.

chelmuth, 2025-05-20:
Commit 003acfa98d0693455ec8f9b952a9c2c0d462bf05 incl. fixup 2db9c07cb52f4a4e047a29d518bb1cbb866df923 produce a strange error on `make -C build/x86_32 driver/gpu/intel`.

```
In file included from /data/depot/checker/api/base/2025-05-19/include/base/allocator_avl.h:17,
                 from /data/depot/checker/api/base/2025-05-19/include/base/heap.h:21,
                 from /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:20:
In function âvoid operator delete(void*, Genode::Allocator*)â,
    inlined from âvoid Genode::destroy(auto:80&&, T*) [with T = Igd::Ggtt::Mapping; auto:80 = Allocator*]â at /data/depot/checker/api/base/2025-05-19/include/base/allocator.h:279:18,
    inlined from âIgd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)> [with auto:156 = Igd::Ggtt; auto:157 = Igd::Mmio]â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:352:19,
    inlined from âIgd::Device::with_ggtt<Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>, Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()> >(const Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>&, const Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()>&)::<lambda(auto:151&)> [with auto:151 = Igd::Mmio]â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:267:6,
    inlined from âvoid Platform::Resources::with_mmio(const auto:121&, const auto:122&) [with auto:121 = Igd::Device::with_ggtt<Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>, Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()> >(const Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>&, const Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()>&)::<lambda(auto:151&)>; auto:122 = Igd::Device::with_ggtt<Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>, Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()> >(const Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>&, const Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()>&)::<lambda()>]â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/platform_session.h:617:7,
    inlined from âvoid Igd::Device::with_ggtt(const auto:141&, const auto:142&) [with auto:141 = Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>; auto:142 = Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()>]â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:266:23,
    inlined from âvoid Igd::Device::with_ggtt(const auto:141&, const auto:142&) [with auto:141 = Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda(auto:156&, auto:157&)>; auto:142 = Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)::<lambda()>]â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:259:7,
    inlined from âvoid Igd::Device::unmap_dataspace_ggtt(Genode::Allocator&, Igd::Ggtt::Mapping*)â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:349:12,
    inlined from âGpu::Session_component::~Session_component()::<lambda(Gpu::Session_component::Vram_local&)>::<lambda(Gpu::Session_component::Vram&)>â at /data/depot/checker/src/intel_gpu/2025-05-19/src/driver/gpu/intel/main.cc:2083:35:
/data/depot/checker/api/base/2025-05-19/include/base/allocator.h:230:25: error: âvoid operator delete(void*, Genode::Deallocator*)â called on pointer âvramâ with nonzero offset 92 [-Werror=free-nonheap-object]
  230 |         operator delete (ptr, static_cast<Genode::Deallocator *>(a)); }
      |         ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

chelmuth, 2025-05-21:
> Commit [003acfa](https://github.com/genodelabs/genode/commit/003acfa98d0693455ec8f9b952a9c2c0d462bf05) incl. fixup [2db9c07](https://github.com/genodelabs/genode/commit/2db9c07cb52f4a4e047a29d518bb1cbb866df923) produce a strange error on `make -C build/x86_32 driver/gpu/intel`.

This may be a compiler bug (false-positive) in GCC 13/14/15 that is fixed in current trunk https://gcc.gnu.org/bugzilla/show_bug.cgi?id=115016. I still have no idea how to fix this.

ssumpf, 2025-05-21:
> This may be a compiler bug (false-positive) in GCC 13/14/15 that is fixed in current trunk https://gcc.gnu.org/bugzilla/show_bug.cgi?id=115016. I still have no idea how to fix this.

@chelmuth: I had a quick look, this looks more like a bug in the driver. I will let you know when I know more.

ssumpf, 2025-05-21:
@chelmuth:  Okay, 
```
_device.unmap_dataspace_ggtt(_heap, &vram.map);
```
`vram.map` is actually a non-allocated, non-pointer member of `Vram` and cannot be freed. This certainly is wrong and the compiler tells us the right thing. The whole `map` thing seems to be a relic which is never even properly initialized and only used by `set_tiling` which is not used by the current Iris-Mesa driver. I will try to get rid of it.

nfeske, 2025-05-21:
Thanks a lot @ssumpf. Merged to staging.