genodelabs/genode #5248: tool/run: configuration syntax check fails with libxml2 >= 2.13

jschlatow, 2024-06-14:
After a system update on my Archlinux, I got an error from xmllint not able to find the base_types.xsd. Although I haven't found anything suspicious in the release notes it turned out that the `-path` argument seems to have been rendered ineffective by some change. I'm not even sure that it was supposed to work in the way we made use of it. I therefore looked for alternative ways to tell xmllint about where to look for xsd files and found the environment variable XML_CATALOG_FILES.

jschlatow, 2024-06-14:
Commit 1681ca9 is my take on fixing this. Since I'm unable to downgrade my libxml2, I'd appreciate if someone could test this with an older libxml2 version.

m-stein, 2024-06-17:
Cool that you fixed it! Unfortunately, I have version 2.9.13 .

m-stein, 2024-06-17:
Sorry, I just realized that 2.9.13 should be older than 2.13 XD will test it.

m-stein, 2024-06-17:
Seems to work for me.

commit 3969128c (2024-06-17): tool/run: use xmlcatalog for xsd file paths

With libxml2 >= 2.13, the `-path` argument can no longer be used for
setting search paths for xsd files. Instead, we use an XML catalog to
replace genode:// URIs with absolute paths.

Fixes #5248
closed 2024-06-20 by nfeske